zephyrproject-rtos/zephyr #85270: esp32c3 mcuboot not use CONFIG_MCUBOOT_SIGNATURE_KEY_FILE

zhang-wenchao, 2025-02-06:
nrf52840 normal.

esp32c3 without using this configuration, I don't know how to set up the signature.

zhang-wenchao, 2025-02-06:
`cd /bootloader/mcuboot`

`rm *.pem`

`cd samples/sysbuild/with_mcuboot`

`west build -p always -b esp32c3_devkitm --sysbuild`

After I removed all the test pems of mcuboot, I was able to compile normally and generate zephyr.signed.bin

zhang-wenchao, 2025-02-06:
I also tried bootloader/mcuboot/samples/zephyr/hello-world and got the same result, no configuration was followed.

zhang-wenchao, 2025-02-06:
I hope the issue can be assigned and tracked to be resolved as soon as possible, and I will provide necessary assistance.

nordicjm, 2025-02-06:
Documented on https://docs.zephyrproject.org/latest/build/signing/index.html

closed 2025-02-06 by nordicjm

zhang-wenchao, 2025-02-06:
@nordicjm 

@sylvioalves 

```
SB_CONFIG_BOOTLOADER_MCUBOOT=y
SB_CONFIG_BOOT_SIGNATURE_KEY_FILE="bootloader/mcuboot/root-rsa-2048.pem"
```

**nrf52840** is normal.

**esp32c3**:

```
warning: BOOT_SIGNATURE_KEY_FILE (defined at share/sysbuild/images/bootloader/Kconfig:142) was
assigned the value 'bootloader/mcuboot/root-rsa-2048.pem' but got the value ''. See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_BOOT_SIGNATURE_KEY_FILE and/or look up
BOOT_SIGNATURE_KEY_FILE in the menuconfig/guiconfig interface. The Application Development Primer,
Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be
helpful too.
```

I recommend opening this issue to track esp32c3 specific issues.

nordicjm, 2025-02-06:
The default signature for that is none, you would need to select one of:
```
SB_CONFIG_BOOT_SIGNATURE_TYPE_RSA
SB_CONFIG_BOOT_SIGNATURE_TYPE_ECDSA_P256
SB_CONFIG_BOOT_SIGNATURE_TYPE_ED25519
```

zhang-wenchao, 2025-02-06:
esp32c3 still has problems:

```
# Enable the bootloader when building with sysbuild.
SB_CONFIG_BOOTLOADER_MCUBOOT=y
SB_CONFIG_BOOT_SIGNATURE_KEY_FILE="/home/life/Develop/platform/zephyr-project/bootloader/mcuboot/root-rsa-2048.pem"
SB_CONFIG_BOOT_SIGNATURE_TYPE_RSA=y
```

```
$ west build -p always -b esp32c3_devkitm --sysbuild

-- west build: making build dir /home/life/Develop/platform/zephyr-project/bootloader/mcuboot/samples/zephyr/hello-world/build pristine
-- west build: generating a build system
Loading Zephyr module(s) (Zephyr base): sysbuild_default
-- Found Python3: /home/life/Develop/platform/zephyr-project/.venv/bin/python3 (found suitable version "3.11.2", minimum required is "3.10") found components: Interpreter
-- Cache files will be written to: /home/life/.cache/zephyr
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: esp32c3_devkitm, qualifiers: esp32c3
Parsing /home/life/Develop/platform/zephyr-project/zephyr/share/sysbuild/Kconfig
Loaded configuration '/home/life/Develop/platform/zephyr-project/bootloader/mcuboot/samples/zephyr/hello-world/build/_sysbuild/empty.conf'
Merged configuration '/home/life/Develop/platform/zephyr-project/bootloader/mcuboot/samples/zephyr/hello-world/sysbuild.conf'
Configuration saved to '/home/life/Develop/platform/zephyr-project/bootloader/mcuboot/samples/zephyr/hello-world/build/zephyr/.config'
Kconfig header saved to '/home/life/Develop/platform/zephyr-project/bootloader/mcuboot/samples/zephyr/hello-world/build/_sysbuild/autoconf.h'
-- 
   *****************************
   * Running CMake for mcuboot *
   *****************************

Loading Zephyr default modules (Zephyr base).
-- Application: /home/life/Develop/platform/zephyr-project/bootloader/mcuboot/boot/zephyr
-- CMake version: 3.30.5
-- Found Python3: /home/life/Develop/platform/zephyr-project/.venv/bin/python (found suitable version "3.11.2", minimum required is "3.10") found components: Interpreter
-- Cache files will be written to: /home/life/.cache/zephyr
-- Zephyr version: 4.0.0 (/home/life/Develop/platform/zephyr-project/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: esp32c3_devkitm, qualifiers: esp32c3
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.8 (/home/life/Develop/platform/zephyr-project/sdk/zephyr-sdk-0.16.8)
-- Found toolchain: zephyr 0.16.8 (/home/life/Develop/platform/zephyr-project/sdk/zephyr-sdk-0.16.8)
-- Found Dtc: /home/life/Develop/platform/zephyr-project/sdk/zephyr-sdk-0.16.8/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/life/Develop/platform/zephyr-project/zephyr/boards/espressif/esp32c3_devkitm/esp32c3_devkitm.dts
-- Found devicetree overlay: /home/life/Develop/platform/zephyr-project/bootloader/mcuboot/boot/zephyr/app.overlay
-- Generated zephyr.dts: /home/life/Develop/platform/zephyr-project/bootloader/mcuboot/samples/zephyr/hello-world/build/mcuboot/zephyr/zephyr.dts
-- Generated pickled edt: /home/life/Develop/platform/zephyr-project/bootloader/mcuboot/samples/zephyr/hello-world/build/mcuboot/zephyr/edt.pickle
-- Generated zephyr.dts: /home/life/Develop/platform/zephyr-project/bootloader/mcuboot/samples/zephyr/hello-world/build/mcuboot/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/life/Develop/platform/zephyr-project/bootloader/mcuboot/samples/zephyr/hello-world/build/mcuboot/zephyr/include/generated/zephyr/devicetree_generated.h
-- Including generated dts.cmake file: /home/life/Develop/platform/zephyr-project/bootloader/mcuboot/samples/zephyr/hello-world/build/mcuboot/zephyr/dts.cmake

warning: BOOT_MAX_IMG_SECTORS (defined at /home/life/Develop/platform/zephyr-
project/bootloader/mcuboot/boot/zephyr/Kconfig:449) was assigned the value '512' but got the value
''. Check these unsatisfied dependencies: (!BOOT_MAX_IMG_SECTORS_AUTO) (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_BOOT_MAX_IMG_SECTORS and/or look up
BOOT_MAX_IMG_SECTORS in the menuconfig/guiconfig interface. The Application Development Primer,
Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be
helpful too.

Parsing /home/life/Develop/platform/zephyr-project/bootloader/mcuboot/boot/zephyr/Kconfig
Loaded configuration '/home/life/Develop/platform/zephyr-project/zephyr/boards/espressif/esp32c3_devkitm/esp32c3_devkitm_defconfig'
Merged configuration '/home/life/Develop/platform/zephyr-project/bootloader/mcuboot/boot/zephyr/prj.conf'
Merged configuration '/home/life/Develop/platform/zephyr-project/bootloader/mcuboot/boot/zephyr/socs/esp32c3.conf'
Merged configuration '/home/life/Develop/platform/zephyr-project/bootloader/mcuboot/samples/zephyr/hello-world/build/mcuboot/zephyr/.config.sysbuild'
Configuration saved to '/home/life/Develop/platform/zephyr-project/bootloader/mcuboot/samples/zephyr/hello-world/build/mcuboot/zephyr/.config'
Kconfig header saved to '/home/life/Develop/platform/zephyr-project/bootloader/mcuboot/samples/zephyr/hello-world/build/mcuboot/zephyr/include/generated/zephyr/autoconf.h'
-- Found GnuLd: /home/life/Develop/platform/zephyr-project/sdk/zephyr-sdk-0.16.8/riscv64-zephyr-elf/riscv64-zephyr-elf/bin/ld.bfd (found version "2.38")
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
[... cut ...]
FAILED: _sysbuild/sysbuild/images/bootloader/mcuboot-prefix/src/mcuboot-stamp/mcuboot-build /home/life/Develop/platform/zephyr-project/bootloader/mcuboot/samples/zephyr/hello-world/build/_sysbuild/sysbuild/images/bootloader/mcuboot-prefix/src/mcuboot-stamp/mcuboot-build 
cd /home/life/Develop/platform/zephyr-project/bootloader/mcuboot/samples/zephyr/hello-world/build/mcuboot && /home/linuxbrew/.linuxbrew/Cellar/cmake/3.30.5/bin/cmake --build .
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /home/linuxbrew/.linuxbrew/bin/cmake --build /home/life/Develop/platform/zephyr-project/bootloader/mcuboot/samples/zephyr/hello-world/build
```

zhang-wenchao, 2025-02-06:
@sylvioalves Is this an esp32c3 specific issue?

zhang-wenchao, 2025-02-06:
```
SB_CONFIG_BOOTLOADER_MCUBOOT=y
SB_CONFIG_BOOT_SIGNATURE_KEY_FILE="/home/life/Develop/platform/zephyr-project/bootloader/mcuboot/root-ec-p256.pem"
SB_CONFIG_BOOT_SIGNATURE_TYPE_ECDSA_P256=y

```

```
diff --git a/soc/espressif/esp32c3/memory.h b/soc/espressif/esp32c3/memory.h
index 97b2068bb45..784b0a3355e 100644
--- a/soc/espressif/esp32c3/memory.h
+++ b/soc/espressif/esp32c3/memory.h
@@ -46,8 +46,8 @@
 /* For safety margin between bootloader data section and startup stacks */
 #define BOOTLOADER_STACK_OVERHEAD      0x0
 /* These lengths can be adjusted, if necessary: */
-#define BOOTLOADER_DRAM_SEG_LEN        0x9800
-#define BOOTLOADER_IRAM_SEG_LEN        0x9800
+#define BOOTLOADER_DRAM_SEG_LEN        0xa800
+#define BOOTLOADER_IRAM_SEG_LEN        0xb400
 #define BOOTLOADER_IRAM_LOADER_SEG_LEN 0x1400
 
 /* Start of the lower region is determined by region size and the end of the higher region */
```

This configuration works, and I'll use it this way for now.