zephyrproject-rtos/zephyr #85290: check_compliance.py: Running Kconfig module fails when hal_espressif is not present

ydamigos, 2025-02-06:
**Describe the bug**
Running `scripts/ci/check_compliance.py --annotate -e KconfigBasic -e SysbuildKconfigBasic -e ClangFormat -m Kconfig -c origin/main..` fails when `hal_espressif` is not present

```
ERROR   : Test Kconfig failed: 
Found references to undefined Kconfig symbols. If any of these are false
positives, then add them to UNDEF_KCONFIG_ALLOWLIST in /home/coder/workspace/zephyrproject/zephyr/scripts/ci/check_compliance.py.

If the reference is for a comment like /* CONFIG_FOO_* */ (or
/* CONFIG_FOO_*_... */), then please use exactly that form (with the '*'). The
CI check knows not to flag it.

More generally, a reference followed by $, @, {, (, ., *, or ## will never be
flagged.

CONFIG_ESP_CONSOLE                         soc/espressif/esp32/hw_init.c:54, soc/espressif/esp32/hw_init.c:58, soc/espressif/esp32c2/hw_init.c:54, soc/espressif/esp32c2/hw_init.c:58, soc/espressif/esp32c3/hw_init.c:56, soc/espressif/esp32c3/hw_init.c:60, soc/espressif/esp32c6/hw_init.c:72, soc/espressif/esp32c6/hw_init.c:76, soc/espressif/esp32s2/hw_init.c:54, soc/espressif/esp32s2/hw_init.c:58, soc/espressif/esp32s3/hw_init.c:60, soc/espressif/esp32s3/hw_init.c:64
CONFIG_ESP_CONSOLE_UART_BAUDRATE           drivers/clock_control/clock_control_esp32.c:713
CONFIG_ESP_CONSOLE_UART_NONE               drivers/clock_control/clock_control_esp32.c:705
CONFIG_ESP_CONSOLE_UART_NUM                drivers/clock_control/clock_control_esp32.c:196, drivers/clock_control/clock_control_esp32.c:199, drivers/clock_control/clock_control_esp32.c:208, drivers/clock_control/clock_control_esp32.c:211, drivers/clock_control/clock_control_esp32.c:215, drivers/clock_control/clock_control_esp32.c:316, drivers/clock_control/clock_control_esp32.c:319, drivers/clock_control/clock_control_esp32.c:325, drivers/clock_control/clock_control_esp32.c:328, drivers/clock_control/clock_control_esp32.c:332, drivers/clock_control/clock_control_esp32.c:357, drivers/clock_control/clock_control_esp32.c:360, drivers/clock_control/clock_control_esp32.c:365, drivers/clock_control/clock_control_esp32.c:640, drivers/clock_control/clock_control_esp32.c:712
CONFIG_RESERVE_RTC_MEM                     soc/espressif/esp32/default.ld:72, soc/espressif/esp32/default.ld:75, soc/espressif/esp32/default.ld:79, soc/espressif/esp32/default.ld:81, soc/espressif/esp32/default.ld:82, soc/espressif/esp32/default.ld:83, soc/espressif/esp32/default.ld:262, soc/espressif/esp32c3/default.ld:76, soc/espressif/esp32c3/default.ld:78, soc/espressif/esp32c3/default.ld:84, soc/espressif/esp32c3/default.ld:85, soc/espressif/esp32c3/default.ld:86, soc/espressif/esp32c3/default.ld:261, soc/espressif/esp32c6/default.ld:70, soc/espressif/esp32c6/default.ld:72, soc/espressif/esp32c6/default.ld:79, soc/espressif/esp32c6/default.ld:80, soc/espressif/esp32c6/default.ld:81, soc/espressif/esp32c6/default.ld:262, soc/espressif/esp32s2/default.ld:76, soc/espressif/esp32s2/default.ld:78, soc/espressif/esp32s2/default.ld:80, soc/espressif/esp32s2/default.ld:87, soc/espressif/esp32s2/default.ld:88, soc/espressif/esp32s2/default.ld:89, soc/espressif/esp32s2/default.ld:270, soc/espressif/esp32s3/default.ld:79, soc/espressif/esp32s3/default.ld:81, soc/espressif/esp32s3/default.ld:88, soc/espressif/esp32s3/default.ld:89, soc/espressif/esp32s3/default.ld:90, soc/espressif/esp32s3/default.ld:273
```

The issue was introduced in https://github.com/zephyrproject-rtos/zephyr/pull/84532

**To Reproduce**
1. Remove hal_espressif from modules folder
2. `scripts/ci/check_compliance.py --annotate -e KconfigBasic -e SysbuildKconfigBasic -e ClangFormat -m Kconfig -c origin/main..`

**Impact**
Internal CI `check_compliance` job fails.

sylvioalves, 2025-02-06:
@ydamigos thanks for reporting. I addressed it.
It is also worth mentioning that compliance check also fails when hal_nordic is no present. Have you tested?

ydamigos, 2025-02-06:
@sylvioalves Are you talking about the dependency of `nrf_hw_models` module to `hal_nordic` module? 

If yes, we came across it and we added both modules in our `manifest.project-filter` :

`west config --local 'manifest.project-filter' -- '-nrf_hw_models,-hal_.*,+hal_renesas'`

PR #85293 (2025-02-10): espressif: add console and RTC kconfig entries
Add hidden console and RTC configurations used in hal
to common SoC folder.

Fixes #85290.
closed 2025-02-10 by kartben