zephyrproject-rtos/zephyr #85305: intel_ish_5_8_0: libraries.libc.c11_threads.* - Build failure

majunkier, 2025-02-06:
**Describe the bug**
Below tests are failing during building on platform **intel_ish_5_8_0**
libraries.libc.c11_threads.minimal
libraries.libc.c11_threads.picolibc
libraries.libc.c11_threads.picolibc.module
libraries.libc.c11_threads.picolibc.notls
libraries.libc.c11_threads.newlib
libraries.libc.c11_threads.newlib_nano 

This commit causes a regression [0ac931b](https://github.com/zephyrproject-rtos/zephyr/commit/0ac931b38647d875693a8b881fae0319692562a6)

**To Reproduce**
_example of one test from list:_
west twister -p intel_ish_5_8_0/intel_ish_5_8_0 -s libraries.libc.c11_threads.minimal

**Logs and console output**
```
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /zephyrproject/zephyr/twister-out/intel_ish_5_8_0_intel_ish_5_8_0/zephyr/tests/lib/c_lib/thrd/libraries.libc.c11_threads.minimal/zephyr/zephyr_pre0.map
:   <trimmed>
/zephyr-sdk-0.17.0/x86_64-zephyr-elf/bin/../lib/gcc/x86_64-zephyr-elf/12.2.0/../../../../x86_64-zephyr-elf/bin/ld.bfd:
 zephyr/arch/arch/x86/core/libarch__x86__core.a(prep_c.c.obj): in function `z_prep_c':
/zephyrproject/zephyr/arch/x86/core/prep_c.c:40: undefined reference to `z_x86_mmu_init'

/zephyr-sdk-0.17.0/x86_64-zephyr-elf/bin/../lib/gcc/x86_64-zephyr-elf/12.2.0/../../../../x86_64-zephyr-elf/bin/ld.bfd: zephyr/kernel/libkernel.a(mmu.c.obj):
in function `k_mem_map_phys_bare':
/zephyrproject/zephyr/kernel/mmu.c:961: undefined reference to `arch_mem_map'

collect2: error: ld returned 1 exit status
```

**Environment (please complete the following information):**

 - OS: Ubuntu
 - Toolchain:  zephyr-sdk-0.17.0
 -  Version used : v4.0.0-4669-g20516b78348d

nashif, 2025-02-06:
@cfriedt bisection leads to 0ac931b38647d875693a8b881fae0319692562a6, any idea what is going on here?

cfriedt, 2025-02-06:
Yes. I'll submit a PR

cfriedt, 2025-02-06:
@nashif - is there a way to automatically run this test suite when there are changes under `lib/posix`?

cfriedt, 2025-02-10:
Found a better way than what is in #85327 currently. Will re-use that PR to fix the regression.

nashif, 2025-02-14:
as already noted in the proposed fix, posix layer should not be selecting MMU and should not be doing this based on HW capability check (CPU_HAS_MMU), this is all reserved to the platform/soc and hardwre support layer.

MMU is not selectable by the OS layer, it is only selectable by the platform if the architecture level feature is enabled. In the case of X86, X86_MMU should be selecting MMU. This applies to all architectures and also to MPU.

For example:

`west build -p -b qemu_x86/atom/nommu  tests/benchmarks/posix/threads -T benchmark.posix.threads` fails the same way, because posix is forcing MMU, although the platform is not enabling MMU and despite the fact that the CPU is capable.

Taking a random board with no mpu/mmu, i.e.

`west build -p -b bbc_microbit  tests/benchmarks/posix/threads -T benchmark.posix.threads`

builds fine, which should be the same behavior in the case of `qemu_x86/atom/nommu `.

Posix in this case should be depending on the capability, not selecting it.

cfriedt, 2025-02-16:
> as already noted in the proposed fix, posix layer should not be selecting MMU and should not be doing this based on HW capability check (CPU_HAS_MMU), this is all reserved to the platform/soc and hardwre support layer.

The posix layer does not select the mmu. It implies it, but only when the architecture has an MMU and when the application enables the memory mapping option group, as that particular hardware feature is required for functionality. And of course, this is how posix options are intended to work.

However, the main issue here is that the x86 architecture has an inconsistent implementation of this particular aspect of the arch interface (also mentioned in the proposed fix), which is least invasive.

A slightly easier fix could be simply disabling the feature for x86 until the architecture interface is consistent.

nashif, 2025-02-16:
> The posix layer does not select the mmu. It implies it, 

imply or select, it makes no difference if you should not be doing this in the first place. MMU shall only be selected by the implementation of MMU, i.e. X86_MMU or ARM_MMU, it should not be selected/implied by anyone else.

> but only when the architecture has an MMU  and when the application enables the memory mapping option group, as that particular hardware feature is required for functionality. 

Architecture might support MMU, but a platform might not be enabling that for various valid reasons. POSIX should not enable MMU in this case, it needs to just depend on it and if MMU is enabled on the platform, those posix features needing that would not be enabled. 

And of course, this is how MMU options are intended to work.

> And of course, this is how posix options are intended to work.

No, POSIX features requiring MMU shall not enable MMU, it shall depend on it instead.

> However, the main issue here is that the x86 architecture has an inconsistent implementation of this particular aspect of the arch interface (also mentioned in the proposed fix), which is least invasive.

What is the inconsistency here exactly? I tried to explain above how this is supposed to work and this is not an X86 issue, this is how MMU on all architectures is working now , this is how MPU works and this is how it was since this was introduced.

try:

`west build -p -b qemu_cortex_a53  tests/benchmarks/posix/threads -T benchmark.posix.threads  -- -DCONFIG_ARM_MMU=n`

Which represents an arch that has MMU support but does not want to enable MMU, this could be a platform configuration, similar to the x86 nommu variant, this exposes exactly the same issue seen on ish:

POSIX enables MMU, although the platform does not enable or want that, we end up with:

```
-- Zephyr version: 4.0.99 (/home/anashif/zephyrproject/zephyr), build: v4.0.0-5519-gd339b961321f
[101/166] Building C object zephyr/subsys/demand_paging/eviction/CMakeFiles/subsys__demand_paging__eviction.dir/lru.c.obj
FAILED: zephyr/subsys/demand_paging/eviction/CMakeFiles/subsys__demand_paging__eviction.dir/lru.c.obj
ccache /home/anashif/zephyr-sdk-0.17.0/aarch64-zephyr-elf/bin/aarch64-zephyr-elf-gcc -DKERNEL -DK_HEAP_MEM_POOL_SIZE=1024 -DPICOLIBC_DOUBLE_PRINTF_SCANF -D__LINUX_ERRNO_EXTENSIONS__ -D__ZEPHYR_SUPERVISOR__ -D__ZEPHYR__=1 -I/home/anashif/zephyrproject/zephyr/kernel/include -I/home/anashif/zephyrproject/zephyr/arch/arm64/include -I/home/anashif/zephyrproject/zephyr/build/zephyr/include/generated/zephyr -I/home/anashif/zephyrproject/zephyr/include -I/home/anashif/zephyrproject/zephyr/build/zephyr/include/generated -I/home/anashif/zephyrproject/zephyr/soc/arm -I/home/anashif/zephyrproject/zephyr/include/zephyr/posix -I/home/anashif/zephyrproject/zephyr/soc/arm/qemu_cortex_a53/. -I/home/anashif/zephyrproject/zephyr/subsys/testsuite/include -isystem /home/anashif/zephyrproject/zephyr/lib/libc/common/include -Wshadow -fno-strict-aliasing -Os -imacros /home/anashif/zephyrproject/zephyr/build/zephyr/include/generated/zephyr/autoconf.h -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-a53 -mabi=lp64 --sysroot=/home/anashif/zephyr-sdk-0.17.0/aarch64-zephyr-elf/aarch64-zephyr-elf -imacros /home/anashif/zephyrproject/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wdouble-promotion -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -ftls-model=local-exec -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=/home/anashif/zephyrproject/zephyr/tests/benchmarks/posix/threads=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/anashif/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/anashif/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections -moverride=tune=no_ldp_stp_qregs -specs=picolibc.specs -D_POSIX_THREADS -std=c99 -MD -MT zephyr/subsys/demand_paging/eviction/CMakeFiles/subsys__demand_paging__eviction.dir/lru.c.obj -MF zephyr/subsys/demand_paging/eviction/CMakeFiles/subsys__demand_paging__eviction.dir/lru.c.obj.d -o zephyr/subsys/demand_paging/eviction/CMakeFiles/subsys__demand_paging__eviction.dir/lru.c.obj -c /home/anashif/zephyrproject/zephyr/subsys/demand_paging/eviction/lru.c
/home/anashif/zephyrproject/zephyr/subsys/demand_paging/eviction/lru.c: In function 'k_mem_paging_eviction_select':
/home/anashif/zephyrproject/zephyr/subsys/demand_paging/eviction/lru.c:174:32: error: 'ARCH_DATA_PAGE_DIRTY' undeclared (first use in this function)
  174 |         *dirty_ptr = ((flags & ARCH_DATA_PAGE_DIRTY) != 0);
      |                                ^~~~~~~~~~~~~~~~~~~~
/home/anashif/zephyrproject/zephyr/subsys/demand_paging/eviction/lru.c:174:32: note: each undeclared identifier is reported only once for each function it appears in
[110/166] Building C object zephyr/subsys/demand_paging/backing_store/CMakeFiles/subsys__demand_paging__backing_store.dir/ram.c.obj
FAILED: zephyr/subsys/demand_paging/backing_store/CMakeFiles/subsys__demand_paging__backing_store.dir/ram.c.obj
ccache /home/anashif/zephyr-sdk-0.17.0/aarch64-zephyr-elf/bin/aarch64-zephyr-elf-gcc -DKERNEL -DK_HEAP_MEM_POOL_SIZE=1024 -DPICOLIBC_DOUBLE_PRINTF_SCANF -D__LINUX_ERRNO_EXTENSIONS__ -D__ZEPHYR_SUPERVISOR__ -D__ZEPHYR__=1 -I/home/anashif/zephyrproject/zephyr/kernel/include -I/home/anashif/zephyrproject/zephyr/arch/arm64/include - [... cut ...]
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/anashif/zephyrproject/zephyr/build
```

not enabling MMU is a valid configuration/option, i.e. hello world works fine on this plaform without MMU:

`west build -t run  -p -b qemu_cortex_a53  samples/hello_world/ -- -DCONFIG_ARM_MMU=n`

```
*** Booting Zephyr OS build v4.0.0-5519-gd339b961321f ***
Hello World! qemu_cortex_a53/qemu_cortex_a53
qemu-system-aarch64: terminating on signal 2
```

cfriedt, 2025-02-16:
> > The posix layer does not select the mmu. It implies it, 
> 
> imply or select, it makes no difference if you should not be doing this in the first place. MMU shall only be selected by the implementation of MMU, i.e. X86_MMU or ARM_MMU, it should not be selected/implied by anyone else.

That doesn't make any sense. Why is it user-configurable?

> Architecture might support MMU, but a platform might not be enabling that for various valid reasons.

Sure - then those valid reasons should prevent MMU from being enabled. Why is it possible to enable MMU even when it isn't supported by the platform?

Perhaps in addition to CPU_HAS_MMU, there needs to be a CPU_USE_MMU as well such that MMU depends on CPU_USE_MMU instead of CPU_HAS_MMU.

> POSIX should not enable MMU in this case, it needs to just depend on it and if MMU is enabled on the platform, those posix features needing that would not be enabled.

I would love to rely on MMU support being accurately reported by the build system. However, it really hasn't been accurately reported at all.

Hence all of the workarounds needed at the library and application level.

I'll make an alternate PR.

cfriedt, 2025-02-16:
Alternative PR
https://github.com/zephyrproject-rtos/zephyr/pull/85837

PR #85837 (2025-02-19): posix: mem: remove imply MMU from POSIX options
With MMU being converted to a non-user-configurable option in #85834, we can rely it exclusively to know whether `arch_mem_map()` and `arch_mem_unmap()` are available and we can remove the workarounds in POSIX at the Kconfig level.

Fixes #85305
closed 2025-02-19 by kartben