zephyrproject-rtos/zephyr #85308: missing dependency nxp frdm_mcxn947 board

davidealessandrini88, 2025-02-06:
Trying to compile even a simple hello_world example for the frdm_mcxn947 board, the compilation fails.

C:/my_workspace/zephyr/boards/nxp/frdm_mcxn947/frdm_mcxn947-pinctrl.dtsi:7:10: fatal error: nxp/mcx/MCXN947VDF-pinctrl.h: No such file or directory
    7 | #include <nxp/mcx/MCXN947VDF-pinctrl.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

henrikbrixandersen, 2025-02-07:
You need to have the NXP HAL module present. Have you ran `west update`?

davidealessandrini88, 2025-02-07:
Hello, yes of course, I also see project
"name: hal_nxp
      revision: 49ff7e33f848e4b59da59369a77da63e346fb1a3
      path: modules/hal/nxp
      groups:
        - hal"
in west.yml but it doesn't seem to be enough. I compile without ptoblem for other boards (e.g. Nucleo boards).

DerekSnell, 2025-02-07:
Hi @davidealessandrini88 ,
Like Brix said, that file is included in the HAL_NXP repo.  It is located at [MCXN947VDF-pinctrl.h](https://github.com/zephyrproject-rtos/hal_nxp/blob/master/dts/nxp/mcx/MCXN947VDF-pinctrl.h).  It seems either your workspace did not clone the HAL_NXP repo, or it has an older revision before this file was added.

Since this is not a bug with the codebase, I am converting this Issue to a Discussion.

Let us know what you find.  Best regards