zephyrproject-rtos/zephyr #85314: samples: subsys: usb: dfu-next: sample.usbd.dfu{-flash} report timeout

katgiadla, 2025-02-06:
**Describe the bug**
Samples `

- `samples/subsys/usb/dfu-next/sample.usbd.dfu`
- `samples/subsys/usb/dfu-next/sample.usbd.dfu-flash`

Observed for:
-  `nrf52840dk/nrf52840`
- `nrf5340dk/nrf5340/cpuapp/ns`
- `nrf5340dk/nrf5340/cpuapp`

**To Reproduce**
Steps to reproduce the behavior:
1. have `nrf52840dk/nrf52840` connected
2. go to your zephyr dir
3. call `./scripts/twister -s sample.usbd.dfu -p nrf52840dk/nrf52840 --device-testing --device-serial /dev/ttyACM0 -v --inline-logs --west-flash="--erase"`
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
*** Booting Zephyr OS build v4.0.0-4571-g02b2d5563cf3 ***
[00:00:00.384,094] <inf> main: USB DFU sample is initialized
```

**Environment (please complete the following information):**
- OS: Ubuntu 20.04.1 LTS
- Toolchain Zephyr SDK 0.17.0
- Commit SHA or Version used: `v4.0.0-4571-g02b2d5`

**Additional information**
Probably introduced by 694c342885f7ab7c64ed119220a721db99c8d5b6 (result of bisection)

jfischer-no, 2025-02-06:
@katgiadla Please describe what is the problem you are facing or what is "Valid console output"?

katgiadla, 2025-02-28:
Fixes #86362

PR #86362 (2025-03-04): Add console harness to check log message in sample.usbd.dfu test
- Added harness configuration to the sample.usbd.dfu test to avoid failing results on the bench report.
- Configured the console harness with a one_line type and a regex pattern to match the "USB DFU sample is initialized" log message.
- Ensures the test passes if the expected log message is found in the console output.

Fixes: #85314
closed 2025-03-04 by fabiobaltieri