zephyrproject-rtos/zephyr #85318: ESP32 stuck in Interrupt Loop when both Ethernet (wired) and Bluetooth are used

Jannik-M91, 2025-02-06:
**Describe the bug**

I am trying to setup an [ESP32 Ethernet Kit](https://docs.zephyrproject.org/latest/boards/espressif/esp32_ethernet_kit/doc/index.html) with a MCUmgr server running on both the Ethernet (UDP) and Bluetooth interfaces.

Each of the interfaces on their own works, but as soon as I use both the systems gets stuck in an interrupt loop. Booting the board works fine if the Ethernet port is not connected or if I do not start the Bluetooth advertising (by commenting out start_smp_bluetooth_adverts() in main.c). As soon as I connect the Ethernet port to a network switch or start advertising, the board hangs a few seconds later.

The only code that seems to be executed after this IRQ 5 which is registered by the esp_bt_adapter:

[image]

[image]

The "interrupted_stack" pointer of the  points to "bt_stack+3784":

[image]

Unfortunately the thread which is running cannot be debugged as it is part of the binary blobs (it is started by task_create_wrapper() in esp_bt_adapter.c).

**To Reproduce**

```
python -m venv .venv
.\.venv\Scripts\Activate.ps1
pip install west
west init -m https://github.com/Proton-Camera/zephyr.git --mr esp32_eth_bl
pip install -r .\zephyr\scripts\requirements.txt
west update hal_espressif zcbor littlefs
west blobs fetch hal_espressif
west build -b esp32_ethernet_kit/esp32/procpu zephyr/samples/subsys/mgmt/mcumgr/smp_svr -- -DEXTRA_CONF_FILE='overlay-bt.conf;overlay-udp.conf'
west flash
```

**Expected behavior**

The system should not hang when both Ethernet and Bluetooth are used.

**Impact**

This is a showstopper for my application where I want to build a Ethernet/Bluetooth to UART bridge.

**Logs and console output**

```
rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3ffe8000,len:12452
load:0x40078000,len:6656
ho 0 tail 12 room 4
load:0x4007a400,len:59780
1150 mmu set 00010000, pos 00010000
entry 0x4007df48
I (48) soc_init: MCUboot 2nd stage bootloader
I (48) soc_init: compile time Feb  5 2025 17:36:32
W (48) soc_init: Unicore bootloader
I (50) spi_flash: detected chip: generic
I (53) spi_flash: flash io: dio
I (56) soc_init: chip revision: v3.1
I (59) flash_init: SPI Speed      : 40MHz
I (63) flash_init: SPI Mode       : DIO
I (67) flash_init: SPI Flash Size : 4MB
I (70) boot: Jumping to the main image...
RTConnected src 34 to int 2 (cpu 0)
I (106) spi_flash: flash io: dio
*** Booting MCUboot v2.1.0-rc1-184-ga2bc982b3379 ***
*** Using Zephyr OS build v4.0.0-4430-gee56c8dec32b ***
[esp32] [INF] Image index: 0, Swap type: none
[esp32] [INF] Loading image 0 - slot 0 from flash, area id: 0
[esp32] [INF] Application start=40090bd0h
[esp32] [INF] DRAM segment: paddr=000360ach, vaddr=3ffbdb60h, size=06408h ( 25608) load
[esp32] [INF] IRAM segment: paddr=00020080h, vaddr=40080000h, size=1602ch ( 90156) load
I (151) boot: DROM segment: paddr=00040000h, vaddr=3f400000h, size=092D0h ( 37584) map
I (151) boot: IROM segment: paddr=00050000h, vaddr=400d0000h, size=4B918h (309528) map
I (168) boot: libc heap size 5 kB.
I (195) spi_flash: detected chip: generic
I (195) spi_flash: flash io: dio

[00:00:00.246,000] <inf> phy_mii: PHY (1) ID 2430C54
*** Booting Zephyr OS build 29c21a9dc290 ***
[00:00:05.463,000] <inf> phy_mii: PHY (1) Link speed 100 Mb, full duplex
[00:00:05.467,000] <inf> smp_udp: Started (IPv4)
[00:00:05.467,000] <inf> smp_udp: Started (IPv6)
[00:00:05.468,000] <inf> littlefs: LittleFS version 2.10, disk version 2.1
[00:00:05.468,000] <inf> littlefs: FS at flash-controller@3ff42000:0x3b0000 is 48 0x1000-byte blocks with 512 cycle
[00:00:05.468,000] <inf> littlefs: partition sizes: rd 16 ; pr 16 ; ca 64 ; la 32
[00:00:05.469,000] <inf> esp32_bt_adapter: BT controller compile version [241c96c]
[00:00:05.686,000] <inf> bt_hci_core: Identity: 34:5F:45:3E:1A:44 (public)
[00:00:05.686,000] <inf> bt_hci_core: HCI: version 4.2 (0x08) revision 0x030e, manufacturer 0x0060
[00:00:05.686,000] <inf> bt_hci_core: LMP: version 4.2 (0x08) subver 0x030e
[00:00:05.687,000] <inf> smp_bt_sample: Advertising successfully started
[00:00:05.687,000] <inf> smp_sample: build time: Feb  6 2025 13:59:27
uart:~$ 
```

**Environment (please complete the following information):**

 - OS: Windows 11
 - Toolchain: Zephyr SDK 0.17.0
 - Branch (with only minimal changes to add the board to the sample): https://github.com/Proton-Camera/zephyr/tree/esp32_eth_bl

Jannik-M91, 2025-02-06:
This issue might be related: https://github.com/zephyrproject-rtos/zephyr/issues/83949

rftafas, 2025-02-06:
Nope. Not related. We have pending updates to support ESP32 rev 3.1.

Jannik-M91, 2025-02-10:
> [edit] This is more like a 'request' than a bug.

So using both Ethernet and Bluetooth at the same time is not a use case for this chip?

rftafas, 2025-04-28:
There was a bug with the level of the BT IRQ, yet to be more understood that affected Wi-Fi. We believe it affects Ethernet and/or anything touching network. It is related to Xtensas and IRQs (nested) but the current workaround it to lower the BT IRQ priority. There is a fix on the go: https://github.com/zephyrproject-rtos/hal_espressif/pull/435

It can be tested in this scenario.

arifbalik, 2025-04-28:
Could this be related?

https://github.com/zephyrproject-rtos/zephyr/discussions/86398

rftafas, 2025-04-28:
> Could this be related?
> 
> [#86398](https://github.com/zephyrproject-rtos/zephyr/discussions/86398)

Albeit the symptom was a crash, not a penalty in performance like this case, it won't hurt testing.

PR #92270 (2025-06-30): west.yml: hal_espressif: reserve BLE IRQ in interrupt descriptor
Ensure that the BLE interrupt remains marked as reserved in the interrupt table descriptor when it’s in use. This prevents accidental reassignment of the BLE IRQ in downstream configurations.

Depends on #92144

Fix #90336
Fix #85318
closed 2025-06-30 by danieldegrasse