zephyrproject-rtos/zephyr #85320: Blobs: add support to pull in NXP MCXW72 NBU BLE firmware

axelnxp, 2025-02-06:
## Origin

The following binary blob is being added:
`mcxw72/mcxw72_nbu_ble_all_hosted.bin`

The blob origins from [NXP MCUXpresso SDK](https://github.com/nxp-mcuxpresso/mcuxsdk-middleware-bluetooth-controller/blob/36fec038f52171d46a0804434144c600c4233e9f/bin/mcxw72_nbu_ble_all_hosted.bin)

## Type

- [ ] Precompiled library
- [x] Firmware image

## Module

hal_nxp: https://github.com/zephyrproject-rtos/hal_nxp

## Purpose

Precompiled firmware (binary blob) supporting BLE on NXP MCXW72 SoC.

## Pull Request

HAL: https://github.com/zephyrproject-rtos/hal_nxp/pull/505
manifest: https://github.com/zephyrproject-rtos/zephyr/pull/85317

## Dependencies

None

## License

[LA_OPT_NXP_Online_Code_Hosting](https://github.com/zephyrproject-rtos/hal_nxp/pull/505/files#diff-3b581d800562378446fb52d96f6241cbdff6c96d01070adbfb70beb915d61d7c)

dleach02, 2025-02-12:
@axelnxp @mmahadevan108 Approved by TSC 02-12-2025

nashif, 2025-02-12:
approved!

closed 2025-02-12 by nashif