zephyrproject-rtos/zephyr #85322: Reattaching the LVGL repository fork

fabiobaltieri, 2025-02-06:
## Introduction

This is a proposal to reattach the local lvgl repository to the upstream mirror so it can be synced up with the GitHub UI.

### Problem description

The LVGL (and others) repository was created from scratch and used to loosely track upstream, but recently it started tracking upstream by the commit. In this setup updating the repository is somewhat clunky and error prone as the GitHub UI can't be used since merging the update would end up messing up the commit by either rebasing them or creating a merge.

Ideally the repository would be attached to the upstream, which should give us a convenient "sync fork" button, as it's already the case for https://github.com/zephyrproject-rtos/sof, but since the repository has not been created as a fork, it's not possible to attach it to the upstream repo, even by the GitHub admins:

[image]

### Proposed change

Delete the repository, recreate it as a fork, push all branches, done. We'd lose the PRs but that's fine they should go upstream anyway.

### Plan A

- do a local checkout of the current repository, make sure to grab the current branches (`v4.0-branch` and `zephyr`, which points at the same thing)
- delete the repository
- recreate it by forking https://github.com/lvgl/lvgl
- push the old branches on the new repository
- re-set the acl and other repo parameters

This should be pretty simple and should push back all objects and references used by old Zephyr releases, there's no specific tags that have been created and the two branches above points to the same commit so really just `v4.0-branch` would suffice.

The con is that it would leave the repo/objects non existing during the transient, like for a minute or something, probably less than the length of your average monthly github outage anyway, doubt anyone would notice.

### Plan B

Github allows repository renames and seems to maintain the old reference until it's overwritten, a possible sequence that has no transient of unavailable data could be:

- fork https://github.com/lvgl/lvgl into `lvgl_new` or something
- push the old branch to `lvgl_new`
- rename the current repo to `lvgl_old` or something, objects should still be accessible to the old name
- rename `lvgl_new` to `lvgl`
- delete `lvgl_old` out of existance
- re-set the acl and other repo parameters

Though I haven't tested the whole process myself I could try on a test repo on my own.

Would it work? Probably.

Is it worth it? Probably not.

## Alternatives

Keep suffering pushing stuff manually for eternity.

cc @faxe1008 @pdgendt @MaureenHelm (tried to do this with `hal_adi`) @nashif @stephanosio (what do you think? needs to be done by an admin obviously)

hfakkiz, 2025-02-07:
It would be nice to have this change for "hal_adi" repository too. "hal_adi" repository was also created from scratch and has similar problems. We have to move internally tested and approved changes to the "zephyrproject-rtos/hal_adi" repository manually and wait for another approval process like most other repositories. Having a "sync fork" button in "hal_adi" repository would make life easier for us. @MaureenHelm @gkhnclkmxm @ttmut

MaureenHelm, 2025-02-07:
I think there are other modules that suffer from the same problem (@dleach02 @erwango @carlescufi ?). Let's test the options, pick one, and fix all the affected modules the same way.

@keith-zephyr can we discuss in next week's process WG meeting?

aescolar, 2025-02-12:
@fabiobaltieri be sure to mirror **all** branches and tags (specially if people starts taking about other repositories). (In general we do not know what commit may be referenced from some manifest, either zephyr branch, or PR, or any of its downstreams)

>  Let's test the options, pick one, and fix **all the affected modules** the same way.

Please be careful here. Specially those of you like @MaureenHelm who have organization admin roles.
I'm fine with you changing the hal_adi repo. But please be really careful with other modules. Please **do not go changing other modules unless the maintainers of those modules ask for it**, or they are very much on board.
There is much more to it than just the module keeping the main branch: There is a ton of other data and metadata associated with GH repos, incl. issues, PRs (past and present), projects, wiki pages, permissions, protections, automations, and lots of other configuration; there is all current forks of those modules, there can be tones of downstream jobs against those modules.
Pressing the delete button for a repo cannot be undone, and can lead to absolute mayhem both for Zephyr users, developers and downstreams. So again, please, do not do it unless the maintainer of that modules asked for it.

fabiobaltieri, 2025-02-12:
Reworked the proposal to only cover the case where the old repository is renamed and then archived so any development history is maintained. Also not renaming and just switching to a new repository would be possible though we'd start ending up with confusing repository names which I'd rather avoid.

nashif, 2025-02-19:
> Also not renaming and just switching to a new repository would be possible though we'd start ending up with confusing repository names which I'd rather avoid.

This would be plan C, which is the easiest and most compatible IMO and does not require us to push old branches and commits into the newly created fork, there is no room for confusion or error and no old commits and history to deal with in the newly created repo, which remains clean.
The only "issue" is that we need to come up with a new name and a schema for new revision of repos that are less confusing.
We did this in the past for a few repos.

fabiobaltieri, 2025-02-19:
> This would be plan C, which is the easiest and most compatible IMO and does not require us to push old branches and commits into the newly created fork, there is no room for confusion or error and no old commits and history to deal with in the newly created repo, which remains clean.

Sure let's try to do the rename on lvgl and if there are major issues we can do the "plan easy" instead.

fabiobaltieri, 2025-02-19:
> [@fabiobaltieri](https://github.com/fabiobaltieri) why don't you do another thing: automate the replica from the upstream lvgl master branch to this repository master branch? It's not like "pressing the sync button" is easier than letting a 1 line script do a pull & push every 5 minutes. No need to mess around with new or renamed repositories.

Interesting. I don't hate this actually, if the other folks are fine with this and @stephanosio is happy to have such a thing running somewhere I think it may do the trick, then we'd basically have the main/master branch mirror upstream 1:1. It'd still be nice to have the repository metadata setup properly but this would certainly work as well.

nashif, 2025-02-19:
> Interesting. I don't hate this actually, if the other folks are fine with this and @stephanosio is happy to have such a thing running somewhere I think it may do the trick, then we'd basically have [...]

only makes sense if we can do this using github actions, in something we can manage without having to rely on some host somewhere, i.e. if something breaks or if we need to deply this on more repos, we do not want someone or some resource (where the actual work is done) to be the bottleneck.

fabiobaltieri, 2025-02-19:
> only makes sense if we can do this using github actions, in something we can manage without having to rely on some host somewhere, i.e. if something breaks or if we need to deply this on more repos, we do not want someone or some resource (where the actual work is done) to be the bottleneck.

Yeah that'd be disaster to manage, I'll dig a bit on that front and propose something.

nashif, 2025-02-19:
thinking about it a bit more, the synchronization process is not a blocker for moving forward with tracking upstream. We should be able to change the tree and reset the default branch to be following upstream already now and do the updates and syncing manually until we have a job that does all of this. Once we have automation, this can be enabled.

faxe1008, 2025-02-20:
@fabiobaltieri did a quick PoC and indeed the idea of having a github action which syncs the repository seems promising. Here is the repository:
https://github.com/faxe1008/gh-sync/blob/main/.github/workflows/sync-lvgl.yml

fabiobaltieri, 2025-02-20:
@nashif yeah it'd be nice to have both

@faxe1008 wow that was quick, yeah that's pretty much what I had in mind too, we kicked around the idea of having the workflow as part of the repository but github only supports scheduled workflow on the default branch so that would force us to add commits on top of it, which is missing the point of tracking upstream commit by commit in the first place.

So yeah I'm thinking that thing you put together should go in a repository in the org, get the zephyrbot secret, run every 10 minutes or something and then it'd be something that can be used by all modules that wants to track an upstream automatically. Good job putting that together.

fabiobaltieri, 2025-03-18:
Hey I think https://github.com/faxe1008/gh-sync/tree/main is in good shape, @keith-zephyr can we put this back on the agenda?

keith-zephyr, 2025-04-30:
Process WG:
* Scripting for automatic syncing is possible
* Anas working on automation of generation of OATH tokens
* Next steps - apply manual steps to the LVGL project specifically
* Use LVGL trial run to help define a common process
* @nashif  and @fabiobaltieri  to continue working on this

stephanosio, 2025-08-13:
I agree that the module upstream sync process should be scripted and automated in GitHub Actions so that the sync process is well defined and therefore can be performed consistently.

https://github.com/zephyrproject-rtos/infrastructure would be a good place to add such workflows.

fabiobaltieri, 2025-08-13:
@stephanosio so you think https://github.com/faxe1008/gh-sync/tree/main/.github/workflows could land in https://github.com/zephyrproject-rtos/infrastructure/tree/main/.github/workflows? Do you forsee any permissions issue?

stephanosio, 2025-08-13:
> so you think https://github.com/faxe1008/gh-sync/tree/main/.github/workflows could land in https://github.com/zephyrproject-rtos/infrastructure/tree/main/.github/workflows? 

Yes.

> Do you forsee any permissions issue?

I will need to create a new zephyrbot token for module sync jobs (i.e. `secrets.ZB_REPO_SYNC_GITHUB_TOKEN`); but, other than that, none.