zephyrproject-rtos/zephyr #85355: NXP SPI driver (spi_nxp_lpspi) is broken

ubieda, 2025-02-07:
**Describe the bug**
NXP's LPSPI driver is not functional (spi_nxp_lpspi.c), as it fails to perform a SPI read transfer upon initialization.

Git Bisect points to this patch 62b911feea9b7c55f80a871137b12ee07ad1aa0f (#82877).

**To Reproduce**
- Bisect good: `v4.0.0`
- Bisect bad: `upstream/main`
- Build command used while bisecting:
``` console
west build -b vmu_rt1170/mimxrt1176/cm7 samples/sensor/sensor_shell/ -- -DCONFIG_SPI_RTIO=n
```
- SPI Transfer failing:
https://github.com/zephyrproject-rtos/zephyr/blob/50a044a33d4d7a447a5f2a9553ff399a52a59394/drivers/sensor/tdk/icm42688/icm42688_common.c#L41

**Expected behavior**
SPI Initialization succeeds.

**Impact**

NXP SPI driver is not functional

ubieda, 2025-02-07:
CC @decsny

decsny, 2025-02-07:
Well, obviously I would not have let the PR merge if I had found that the driver was "not functional". Can you give some more information about what specifically is failing in the initialization of this sensor.

ubieda, 2025-02-07:
> Well, obviously I would not have let the PR merge if I had found that the "driver was not functional". Can you give some more information about what specifically is failing in the initialization of this sensor.

Hey @decsny - Sure! I'm using the ICM42688 driver, which upon initialization reads a register and compares the contents (https://github.com/zephyrproject-rtos/zephyr/blob/50a044a33d4d7a447a5f2a9553ff399a52a59394/drivers/sensor/tdk/icm42688/icm42688_common.c#L41) and it returns 0x00 (empty buffer?).

ubieda, 2025-02-07:
I should mention that I do not think this is an IMU issue, as this driver works on other platforms, and the patch pointed by git bisect only has NXP SPI driver changes.

decsny, 2025-02-07:
Do you have a signal trace of before and after? I don't know if I have access to this board right away, I'll have to check

decsny, 2025-02-07:
@ubieda One thing I'm noticing is that the board DTS for vmurt1170 has not been defined correctly for the LPSPI node to meet the sensor timing requirements.

Can you try adding these properties to the board DTS to the lpspi nodes with the sensors (according to icm42688 datasheet) and test again:
- `pcs-sck-delay = <39>`
- `sck-pcs-delay = <18>`
- `transfer-delay = <42>`

ubieda, 2025-02-09:
Updating this issue:

I tried the suggested changes and it still fails.

I was able to replicate the issue on an RT1170 EVK, which allowed me to probe the SPI bus:
- Here's a capture before the regression (SPI Write/Read command completed):
[image]
- Here's the same SPI transfer after the regression (it appears the second byte isn't clocked out):
[image]

Here's the overlay I used:
- mimxrt1170_evk_mimxrt1176_cm7.overlay
``` dts
#include <zephyr/dt-bindings/sensor/icm42688.h>

/* Omit the CS line as it's driven by GPIO */
&pinmux_lpspi1 {
	group0 {
		pinmux =
			<&iomuxc_gpio_ad_28_lpspi1_sck>,
			<&iomuxc_gpio_ad_31_lpspi1_sdi>,
			<&iomuxc_gpio_ad_30_lpspi1_sdo>;
		drive-strength = "high";
		slew-rate = "fast";
	};
};

&lpspi1 {
	status = "okay";
	cs-gpios =<&gpio9 28 GPIO_ACTIVE_LOW>;

	/delete-property/ dmas;
	/delete-property/ dma-names;

	icm42688_1: icm42688p1@0 {
		compatible = "invensense,icm42688";
		reg = <0>;
		spi-max-frequency = <24000000>;
		accel-pwr-mode = <ICM42688_DT_ACCEL_LN>;
		accel-odr = <ICM42688_DT_ACCEL_ODR_1000>;
		accel-fs = <ICM42688_DT_ACCEL_FS_16>;
		gyro-pwr-mode = <ICM42688_DT_GYRO_LN>;
		gyro-odr = <ICM42688_DT_GYRO_ODR_1000>;
		gyro-fs = <ICM42688_DT_GYRO_FS_2000>;
	};
};

```

azmat-bilal, 2025-02-11:
i can confirm this as well. getting the same error where the second byte is not clocked out. here is the timing output for ADI sensor adxl372 interfaced to MIMXRT1170-EVKB board.

[image]

interestingly, if i use a use a IMU from st e.g. LSM6DSV16 then i can see all the 16 clock cycles for a read operation ( the physical sensor is not attached since i dont have it at the moment but the read operation does the full 16 bit transaction instead of just 8 cycles in the first case). i modified the SPI Mode to mode 0 (CPOL=0, CPHA=0) to match the spi operating mode for both the st and adi sensors, but the results are the same even with spi mode 3.

[image]

azmat-bilal, 2025-02-11:
testing the PR https://github.com/zephyrproject-rtos/zephyr/pull/85529 changes solves the issue. the spi communication is happening as expect on MIMXRT1170-EVKB using ADI adxl372 sensor.

PR #85529 (2025-02-20): spi_nxp_lpspi: RX bigger than TX read all RX
Change the driver behavior so that if the provided RX buffer set is bigger than the TX buffer, we will read all the RX buffer and fill the TX with NOPs. The SPI driver API does not say to do this, and my original interpretation of the API was that the TX length controls the entire transfer length, but this behavior might fit better with some de facto expectations of in tree consumers.

Alternative to #85526 to Fixes #85355

There is clearly some ambiguity in the SPI API, I have on my todo list to improve the documentation of SPI API, and probably th [... cut]
closed 2025-02-20 by kartben