zephyrproject-rtos/zephyr #85368: Remove setting stack sizes in defconfig files

nordicjm, 2025-02-07:
Stack sizes should have defaults that can be set by applications, boards should not set values for these symbols in *_defconfig files, the following is a list of where this is currently done which should be moved to Kconfig.defconfig files:

@erwango
- [ ] boards/st/stm32g0316_disco/stm32g0316_disco_defconfig
- [ ] boards/st/stm32f0_disco/stm32f0_disco_defconfig
- [ ] boards/st/nucleo_l053r8/nucleo_l053r8_defconfig
- [ ] boards/st/nucleo_l031k6/nucleo_l031k6_defconfig
- [ ] boards/st/nucleo_l011k4/nucleo_l011k4_defconfig
- [ ] boards/st/nucleo_g031k8/nucleo_g031k8_defconfig
- [ ] boards/st/nucleo_f031k6/nucleo_f031k6_defconfig
- [ ] boards/st/nucleo_f030r8/nucleo_f030r8_defconfig
- [ ] boards/others/stm32f030_demo/stm32f030_demo_defconfig
- [ ] boards/seagate/faze/faze_defconfig
---
@dleach02 @mmahadevan108
- [ ] boards/nxp/s32z2xxdc2/s32z2xxdc2_s32z270_rtu1_defconfig
- [ ] boards/nxp/s32z2xxdc2/s32z2xxdc2_s32z270_rtu0_defconfig
- [ ] boards/nxp/lpcxpresso11u68/lpcxpresso11u68_defconfig
- [ ] boards/nxp/imx8qxp_mek/imx8qxp_mek_mimx8qx6_adsp_defconfig
- [ ] boards/nxp/imx8qm_mek/imx8qm_mek_mimx8qm6_adsp_defconfig
- [ ] boards/nxp/imx8mp_evk/imx8mp_evk_mimx8ml8_adsp_defconfig
---
@dcpleung
- [ ] boards/intel/adsp/intel_adsp_cavs25_tgph_defconfig
- [ ] boards/intel/adsp/intel_adsp_cavs25_defconfig
- [ ] boards/intel/adsp/intel_adsp_ace30_ptl_sim_defconfig
- [ ] boards/intel/adsp/intel_adsp_ace30_ptl_defconfig
- [ ] boards/intel/adsp/intel_adsp_ace30_defconfig
- [ ] boards/intel/adsp/intel_adsp_ace20_lnl_sim_defconfig
- [ ] boards/intel/adsp/intel_adsp_ace20_lnl_defconfig
- [ ] boards/intel/adsp/intel_adsp_ace15_mtpm_sim_defconfig
- [ ] boards/intel/adsp/intel_adsp_ace15_mtpm_defconfig
---
@ifyall
- [ ] boards/infineon/cyw920829m2evk_02/cyw920829m2evk_02_defconfig
- [ ] boards/infineon/cy8cproto_063_ble/cy8cproto_063_ble_defconfig
- [ ] boards/infineon/cy8cproto_062_4343w/cy8cproto_062_4343w_defconfig
---
@nordicjm 
- [ ] boards/silabs/starter_kits/slstk3400a/slstk3400a_defconfig
- [ ] boards/renode/cortex_r8_virtual/cortex_r8_virtual_defconfig
- [ ] boards/qemu/cortex_r5/qemu_cortex_r5_defconfig
- [ ] boards/qemu/cortex_a9/qemu_cortex_a9_defconfig
- [ ] boards/qemu/xtensa/qemu_xtensa_sample_controller32_mpu_defconfig
- [ ] boards/qemu/xtensa/qemu_xtensa_dc233c_mmu_defconfig
- [ ] boards/qemu/xtensa/qemu_xtensa_dc233c_defconfig
- [ ] boards/qemu/riscv64/qemu_riscv64_qemu_virt_riscv64_smp_defconfig
- [ ] boards/qemu/riscv64/qemu_riscv64_defconfig
- [ ] boards/qemu/riscv32/qemu_riscv32_qemu_virt_riscv32_smp_defconfig
- [ ] boards/cdns/xt-sim/xt-sim_defconfig
- [ ] boards/aspeed/ast1030_evb/ast1030_evb_defconfig
- [ ] boards/arm/fvp_baser_aemv8r/fvp_baser_aemv8r_fvp_aemv8r_aarch32_smp_defconfig
- [ ] boards/arm/fvp_baser_aemv8r/fvp_baser_aemv8r_fvp_aemv8r_aarch32_defconfig
- [ ] boards/amd/kv260_r5/kv260_r5_defconfig
---

For the above, if this is something that should be defaulted for all boards based on those SoCs then please move to `soc/<soc>/*/Kconfig.default` file e.g.
```
config MAIN_STACK_SIZE
        default 2048

config IDLE_STACK_SIZE
        default 512 if SOC_SERIES_blah
```
If it is a board requirements then move to `boards/<board>/Kconfig.defconfig` file e.g.
```
config MAIN_STACK_SIZE
        default 2048

config IDLE_STACK_SIZE
        default 512 if BOARD_blah_cpucluster_blah
```

- [ ] Additionally check compliance needs to be updated to specify if disallowed symbols apply to board defconfig files only or also SoC Kconfig.defconfig files too

nashif, 2025-02-07:
Agree with this effort. is this documented somewhere as a guideline? Would be great to have this somewhere.
Would also be nice to start catching this in CI somewhere, so we do not end up with new ones being added.

Some simulators and special variants of those probably need that, but can be done in the SoC as suggested above.

nordicjm, 2025-02-07:
> Agree with this effort. is this documented somewhere as a guideline? Would be great to have this somewhere. Would also be nice to start catching this in CI somewhere, so we do not end up with new [...]

Not documented, once cleaned up will add to the disallowed defconfig compliance check though so will fail CI (like currently with PINCTRL) if present in a PR

yawningwinner, 2025-03-26:
I would like to contribute to this issue, could you assign me some of the above tasks?

moonlight83340, 2025-07-07:
boards/intel/adsp/* already fixes here : https://github.com/zephyrproject-rtos/zephyr/pull/85810/files
boards/infineon/* already fixes here : https://github.com/zephyrproject-rtos/zephyr/pull/85576

closed 2026-09-18 by fabiobaltieri