zephyrproject-rtos/zephyr #85372: stm32: usb: Drivers clean up

erwango, 2025-02-07:
**Is your enhancement proposal related to a problem? Please describe.**
STM32 USB drivers have been implemented and maintained w/o using dts information sufficiently.
They have became hard to read and maintain.

**Describe the solution you'd like**
Review driver implementation to maximize usage of dt information.  Ultimately, driver should be fully instantiable (at least for usb next version).
There are multiple steps from low hanging fruits from bindings overhaul.
https://github.com/zephyrproject-rtos/zephyr/issues/71248 should be part of this effort.

Example of low hanging fruit:
```
#if defined(USB)
	priv->pcd.Instance = DT_INST_REG_ADDR(0);
#elif defined(USB_DRD_FS)
	priv->pcd.Instance = USB_DRD_FS;
#elif defined(USB_OTG_FS) || defined(USB_OTG_HS)
	priv->pcd.Init.speed = usb_dc_stm32_get_maximum_speed();
#if DT_HAS_COMPAT_STATUS_OKAY(st_stm32_otghs)
	priv->pcd.Instance = USB_OTG_HS;
#else
	priv->pcd.Instance = USB_OTG_FS;
#endif
#endif /* USB */
```
could be replaced by 
```
	priv->pcd.Instance = (USB_OTG_GlobalTypeDef *)DT_INST_REG_ADDR(0);

#if defined(USB_OTG_FS) || defined(USB_OTG_HS)
	priv->pcd.Init.speed = usb_dc_stm32_get_maximum_speed();
#endif
```

Side notes:
Implementation should get rid of STM32Cube defines, such as:
`USB`, `USB_DRD_FS`

kbidani, 2025-03-12:
PR link: https://github.com/zephyrproject-rtos/zephyr/pull/86100

mathieuchopstm, 2025-10-01:
Outstanding issues: 
- [ ] testusb failures
  - See #93924
- [ ] Max EP size is not correct
  - Caused by bad HAL defines used here: https://github.com/zephyrproject-rtos/zephyr/blob/48a9b04885cb9485c1beaefd728e1b02c26b381e/drivers/usb/udc/udc_stm32.c#L1070-L1084 
  - Requires support for endpoints MPS that is not word-size-aligned here: https://github.com/zephyrproject-rtos/zephyr/blob/48a9b04885cb9485c1beaefd728e1b02c26b381e/drivers/usb/udc/udc_stm32.c#L644-L650
- [ ] ...

Identified clean-up/rework:
- [ ] Extract PHY configuration from main driver + clean up: https://github.com/zephyrproject-rtos/zephyr/blob/48a9b04885cb9485c1beaefd728e1b02c26b381e/drivers/usb/udc/udc_stm32.c#L1224-L1286
  - A separate file `udc_stm32_phy.c` in the same folder should hold this code (or maybe in `usb/common` since shared with an hypothetical UHC driver?) 
  - PWR could also be moved to a different place (SoC-specific code?)
- [ ] Rework all single-instance-only code
  - e.g., remove patterns such as https://github.com/zephyrproject-rtos/zephyr/blob/48a9b04885cb9485c1beaefd728e1b02c26b381e/drivers/usb/udc/udc_stm32.c#L1086-L1091 
- [ ] ...

Possible future evolutions:
- [ ] Use `udc_dwc2.c` driver (`snps,dwc2` compatible) on applicable hardware
  - `st,stm32-otgfs` / `st,stm32-otghs`