zephyrproject-rtos/zephyr #85378: Error when building for DSPIC33EV256GM106.

Harish-00774, 2025-02-07:
Describe the bug
Building the hello world sample for DSPIC33EV256GM106 with XC-16 Toolchain gives an error. See logs below.

I'm on Windows V.Scode on a Lenovo.

To Reproduce
Steps to reproduce the behavior:

Follow the steps for Windows in [Getting Started] and related Architecture and Toolchain docs.
Activate Virtual Env.
Go to ~/<zephyr_repository
Run west build -p  -b dspic32ev256gm106 samples/hello_world

See the error in the logs output of this report.
Expected behavior
I would expect it to build like it does when I build for SAML21_XPRO.

[image]

Impact
Blocking learnings about Zephyr OS

ks-bobban, 2025-02-07:
Hi,

This our exact issue, pls see the below log
CMake Error at C:/Users/emira/zephyrproject/zephyr/cmake/modules/extensions.cmake:2451 (message):
  **No such file or directory: LIBGCC_FILE_NAME: 'libgcc.a**'
Call Stack (most recent call first):
  C:/Users/emira/zephyrproject/zephyr/cmake/compiler/gcc/target.cmake:109 (assert_exists)

I am adding the full compilations error log in text format:

C:\Users\emira\zephyrproject\zephyr\samples\basic\blinky>west build -p -b dspic33ev256gm106
-- west build: making build dir C:\Users\emira\zephyrproject\zephyr\samples\basic\blinky\build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: C:/Users/emira/zephyrproject/zephyr/samples/basic/blinky
-- CMake version: 3.28.3
-- Found Python3: C:/Python311/python.exe (found suitable version "3.11.8", minimum required is "3.8") found components: Interpreter
-- Cache files will be written to: C:/Users/emira/zephyrproject/zephyr/.cache
-- Zephyr version: 3.6.99 (C:/Users/emira/zephyrproject/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: dspic33ev256gm106, identifier: dspic33ev256gm106
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.5 (C:/Users/emira/zephyrproject/zephyr-sdk-0.16.5)
-- Found toolchain: zephyr 0.16.5 (C:/Users/emira/zephyrproject/zephyr-sdk-0.16.5)
-- Found Dtc: C:/ProgramData/chocolatey/bin/dtc.exe (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: C:/Users/emira/zephyrproject/zephyr/boards/common/stub.dts
-- Generated zephyr.dts: C:/Users/emira/zephyrproject/zephyr/samples/basic/blinky/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: C:/Users/emira/zephyrproject/zephyr/samples/basic/blinky/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: C:/Users/emira/zephyrproject/zephyr/samples/basic/blinky/build/zephyr/dts.cmake
Parsing C:/Users/emira/zephyrproject/zephyr/Kconfig
Loaded configuration 'C:/Users/emira/zephyrproject/zephyr/samples/basic/blinky/prj.conf'
Configuration saved to 'C:/Users/emira/zephyrproject/zephyr/samples/basic/blinky/build/zephyr/.config'
Kconfig header saved to 'C:/Users/emira/zephyrproject/zephyr/samples/basic/blinky/build/zephyr/include/generated/autoconf.h'
-- Using Microchip XC16 toolchain
CMake Error at C:/Users/emira/zephyrproject/zephyr/cmake/modules/extensions.cmake:2451 (message):
  No such file or directory: LIBGCC_FILE_NAME: 'libgcc.a'
Call Stack (most recent call first):
  C:/Users/emira/zephyrproject/zephyr/cmake/compiler/gcc/target.cmake:109 (assert_exists)
  C:/Users/emira/zephyrproject/zephyr/cmake/modules/FindTargetTools.cmake:102 (include)
  C:/Users/emira/zephyrproject/zephyr/cmake/modules/kernel.cmake:25 (find_package)
  C:/Users/emira/zephyrproject/zephyr/cmake/modules/zephyr_default.cmake:141 (include)
  C:/Users/emira/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  C:/Users/emira/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
←[91mFATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' -DWEST_PYTHON=C:/Python311/python.exe '-BC:\Users\emira\zephyrproject\zephyr\samples\basic\blinky\build' -GNinja -DBOARD=dspic33ev256gm106 '-SC:\Users\emira\zephyrproject\zephyr\samples\basic\blinky'
←[0m
C:\Users\emira\zephyrproject\zephyr\samples\basic\blinky>

henrikbrixandersen, 2025-02-07:
Where did you obtain this version of Zephyr? The "dspic33ev256gm106" is not present in upstream `main` branch. Also, what is the "XC-16" toolchain, you mention?

ks-bobban, 2025-02-07:
1. We are developing the Zephyr RTOS for dspic mcu (dspic33ev256gm106). We got the Zephyr RTOS SDK from git hub and the version is zephyr-sdk-0.16.5-1.
2.  The "dspic33ev256gm106" is not present in upstream main branch- Yes (Developed from our side)
3. Microchip XC16 Compiler version 2.10 . It's free compiler and we can use in MPLABX IDE.
https://www.microchip.com/en-us/tools-resources/develop/mplab-xc-compilers/xc16
our MCU- dspic33ev256gm106 (https://www.microchip.com/en-us/product/dsPIC33EV256GM106)