zephyrproject-rtos/zephyr #85380: DHCP address disappears when TCP socket is closed.

hjuul, 2025-02-07:
**Describe the bug**

If the network interface temporarily goes down while a TCP connection is active, it will continue to work when the network interface comes up again - the TCP connection is operational, and the application is visible on the network.

But when the TCP connection is closed, the newly acquired DHCP address disappears and I can no longer connect to the application.

**To Reproduce**

Steps to reproduce the behavior:
1. Run a Zephyr application that gets its IP address from DHCP and accepts TCP connections
2. Create a TCP connection to the application
3. Unplug the network cable and plug it in again immediately
4. The TCP connection should still work
5. You can probably ping the application at its DHCP address, and it should respond
6. Close the TCP connection
7. Now if you try to ping or reconnect, you will not find the application

If you now log the content of `ipv4->unicast[0].ipv4` you will see that the correct DHCP info is there, but `is_used == 1`.

**Impact**

Application must be rebooted if the user disconnects network cable while a connection is active.

But I also see (during robustness testing) that I sometimes get `NET_EVENT_IF_DOWN` immediately followed by `NET_EVENT_IF_UP`, for apparently no reason. When these glitches occur, the application remains offline until I reboot or cycle power.

So this bug is a showstopper for releasing my product.

**Environment**

 - OS: Linux
 - Toolchain: zephyr-sdk-0.16.9
 - Branch used: `v3.7-branch`
 - Hardware: Custom board with stm32f777

**Other**

Possible explanation:  A reference counter was not updated by DHCP after network glitch because the address was already present. Then when the TCP connection is closed, the reference count is zero and the address is removed.

jukkar, 2025-02-10:
@hjuul I have a fix proposal at #85485, please try it and report if it solves your issue.

hjuul, 2025-02-12:
I've verified that the fix in #85485 solves this issue

PR #85485 (2025-02-12): net: Update IP address refcount properly when address already exists
If an IP address already exists when it is tried to be added to the network interface, then just return it but update ref count if it was not updated. This could happen if the address was added and then removed, but for example an active connection was still using it and keeping the ref count > 0. In this case we must update the ref count so that the IP address is not removed if the connection is closed.

Fixes #85380
commit ae052217 (2025-02-12): net: Update IP address refcount properly when address already exists

If an IP address already exists when it is tried to be added to the
network interface, then just return it but update ref count if it was
not updated. This could happen if the address was added and then removed,
but for example an active connection was still using it and keeping the
ref count > 0. In this case we must update the ref count so that the IP
address is not removed if the connection is closed.

Fixes #85380
commit b40cacb6 (2025-02-20): net: Update IP address refcount properly when address already exists

If an IP address already exists when it is tried to be added to the
network interface, then just return it but update ref count if it was
not updated. This could happen if the address was added and then removed,
but for example an active connection was still using it and keeping the
ref count > 0. In this case we must update the ref count so that the IP
address is not removed if the connection is closed.

Fixes #85380

(cherry picked from commit ae052217625881a581e7585f3603e1dbbc298571)
PR #85659 (2025-02-21): [Backport v3.7-branch] net: Update IP address refcount properly when address already exists
This is a backport of #85485 to v3.7. It also contains required commit from #78393 so that the fix can be applied cleanly.

If an IP address already exists when it is tried to be added to the network interface, then just return it but update ref count if it was not updated. This could happen if the address was added and then removed, but for example an active connection was still using it and keeping the ref count > 0. In this case we must update the ref count so that the IP address is not removed if th [... cut]
closed 2025-02-12 by kartben