zephyrproject-rtos/zephyr #85396: Snippets fail to find app/snippet when using --sysbuild

raintamer, 2025-02-07:
**Describe the bug**

When building an application with --sysbuild,  snippets in the application directory are not found.

**To Reproduce**

building without --sysbuild works as expected either foo of fu is printed depending on whether the snippet is enabled.
building with --sybuild  with the foo snippet enabled causes the build to fail with the following

west build -p always -S foo --sysbuild -b esp32_devkitc_wrover/esp32/procpu test_snip

**Logs and console output**
CMake Error at /home/brad/zephyrproject/zephyr/cmake/modules/snippets.cmake:97 (message):
    snippets: error: snippets not found: foo
    Please choose from among the following snippets:
      bt-ll-sw-split
      cdc-acm-console
      nordic-flpr
      nordic-flpr-xip
      nordic-log-stm
      nordic-log-stm-dict
      nordic-ppr
      nordic-ppr-xip
      nus-console
      ram-console
      rtt-console
      rtt-tracing
      serial-console
      wifi-ipv4
      xen_dom0

Call Stack (most recent call first):
  /home/brad/zephyrproject/zephyr/cmake/modules/snippets.cmake:126 (zephyr_process_snippets)
  /home/brad/zephyrproject/zephyr/cmake/modules/zephyr_default.cmake:133 (include)
  /home/brad/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /home/brad/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:12 (find_package)

-- Configuring incomplete, errors occurred!
CMake Error at cmake/modules/sysbuild_extensions.cmake:514 (message):
  CMake configure failed for Zephyr project: mcuboot

  Location: /home/brad/zephyrproject/bootloader/mcuboot/boot/zephyr
Call Stack (most recent call first):
  cmake/modules/sysbuild_images.cmake:20 (ExternalZephyrProject_Cmake)
  cmake/modules/sysbuild_default.cmake:21 (include)
  /home/brad/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:75 (include)
  /home/brad/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  /home/brad/zephyrproject/zephyr/share/sysbuild-package/cmake/SysbuildConfig.cmake:8 (include)
  template/CMakeLists.txt:10 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/bin/cmake -DWEST_PYTHON=/home/brad/.pyenv/versions/3.11.2/bin/python3.11 -B/home/brad/zephyrproject/build -GNinja -DBOARD=esp32_devkitc_wrover/esp32/procpu -DSNIPPET=foo -S/home/brad/zephyrproject/zephyr/share/sysbuild -DAPP_DIR:PATH=/home/brad/zephyrproject/test_snip

** Sources used **
***This is  main.c*** 
#include <stdio.h>

int main()
{

#ifdef CONFIG_FOO
  printf("foo\n");
#else
  printf("fu\n");
#endif
}

***The local Kconfig file***

config FOO
    bool "foo"
    default n

source "Kconfig.zephyr"

The snippets in the snippet/foo diretroy

***snippet.yml***

name: foo
append:
    EXTRA_CONF_FILE: foo.conf

***foo.conf***
CONFIG_FOO=y

**Expected behavior**
The build should find the local snippet file and succeed

**Impact**
showstopper the inability for snippets to set local Kconfig options during sysbuild makes snippets useless. 

**Environment (please complete the following information):**
This is on an Ubunut 24.04 workstation
zephyr-sdk.0.17.0
the hash for zephyr is 337d27141fc0704afc222ab2a2692b37f886e238

robertinant, 2025-02-09:
Having the same issue.
Project uses T2 topology. Snippets are in the snippets directory. A non sysbuild succeeds with the following
`west build  -b frdm_mcxa156 -p auto -Sble` a `west build  -b frdm_mcxa156 -p auto -Sble --sysbuild` fails in the same way as the OP reported.
```
.
├── CMakeLists.txt
├── Kconfig
├── README.md
├── app
│   ├── ble
│   └── main.c
├── boards
│   ├── frdm_mcxa156.overlay
├── manifest
│   └── west.yml
├── snippets
│   └── ble
│       ├── ble.conf
|       └── snippet.yml
└── zephyr
    ├── bootloader
    ├── modules
    └── zephyr

nordicjm, 2025-02-10:
Based on the output there it looks like you are building multiple images, snippets will be use the application config dear for **each** image, meaning `<repo>/snippets/ble` is usable from your **application image only**, I assume you are also building MCUboot, when you do ``-S`` you are applying a snippet to every image, apply it to that image only `-- -Dapp_SNIPPET=ble`

Not a bug.

nordicjm, 2025-02-10:
Try with https://github.com/zephyrproject-rtos/zephyr/pull/85489 and see what error you get

robertinant, 2025-02-10:
Thanks you @nordicjm for the swift reply.

Bingo! That works perfectly. I had a feeling that sysbuild indeed was attempting to apply the snippet to both bootloader and app. Given the location of the snippets folder, mcuboot was not able to find it.
I scoured the documentation to find anything mentioning ${image}_SNIPPET define. Now that I know what it is, see it in the sysbuild_snippets/sysbuild CMakeLists.txt. Would be great to add this to either the snippet or sysbuild documentation.

closed 2025-04-28 by github-actions[bot]