zephyrproject-rtos/zephyr #85398: ESP32-S3 Blinky example fails to build

LTVA1, 2025-02-07:
**Describe the bug**
Blinky example does not compile for ESP32-S3-DevKitC target, fails because of undeclared define.

**To Reproduce**
1. Configure the envoronment as in [getting started guide](https://docs.zephyrproject.org/latest/develop/getting_started/index.html).
2. Try to compile and flash Blinky example as stated in this guide.

**Expected behavior**
I expected the compilation to finish and the resulting binaries to be flashed on the device.

**Impact**
I can't work with the Zephyr toolchain because it does not compile the project from examples directory.

**Logs and console output**
Console output: [log.txt](https://github.com/user-attachments/files/18708860/log.txt)

**Environment (please complete the following information):**

 - OS: Windows 10
 - Toolchain: Zephyr SDK
 - West version: v1.3.0

**Additional context**
It's funny how I just tried to test it on AMP IPM example and it managed to build it for some reason, doesn't start properly on the device though: [log_amp.txt](https://github.com/user-attachments/files/18708967/log_amp.txt)

sylvioalves, 2025-02-07:
This board has no embedded LED that can be controlled over GPIO as done in blinky sample. If you manually attached an LED externally, please read:
https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/basic/blinky/README.rst#adding-board-support

The board has a RGB led though. You can try the sample below:
`samples/drivers/led/led_strip`.

closed 2025-02-07 by sylvioalves

LTVA1, 2025-02-08:
So that's the reason why it doesn't compile?

LTVA1, 2025-02-08:
@sylvioalves I tried and it still fails.

[log_strip.txt](https://github.com/user-attachments/files/18717067/log_strip.txt)

sylvioalves, 2025-02-08:
`No SOURCES given to Zephyr library: drivers__led_strip`
This message in the log file indicates LED strip source isn't added into the build.
1) Can you confirm you run `west update` prior `west build..`?
2) Share the Zephyr branch revision.

I've tested here and it builds properly.

LTVA1, 2025-02-08:
I just ran `west update` to make sure, same error after that (and all stuff was up to date).
How can I check Zephyr branch revision?

sylvioalves, 2025-02-08:
try `git describe --tags`

LTVA1, 2025-02-08:
`v4.0.0-4696-g96a235499bc`

sylvioalves, 2025-02-08:
Ok, I see the issue now. You have to build the sample for the`procpu`. `appcpu` is intended to be use in AMP scenario only.
Please, run: 
`west build -p -b esp32s3_devkitc/esp32s3/procpu samples/drivers/led/led_strip`

LTVA1, 2025-02-08:
Thanks! I completely forgot to test it. I tested blinky for both cores...
However, now it does not start properly... Should I open another issue for that?

[log_failed_start.txt](https://github.com/user-attachments/files/18721984/log_failed_start.txt)

sylvioalves, 2025-02-08:
Board is in bootloader mode. Can you try pressing the BOOT or Reset button?