zephyrproject-rtos/zephyr #85403: DMA behaves different for i.MXRT1064 and i.MXRT685

janmft, 2025-02-07:
**Describe the bug**

By testing DMA, I observed different behavior between  i.MXRT1064 and i.MXRT685.

After adding mimxrt1064_evk.[overlay|conf] to boards, the test case tests/drivers/dma/scatter_gather/ pass for   i.MXRT1064 .

The  test case tests/drivers/dma/scatter_gather/ fails for   i.MXRT685 .

By analyzing the rx_data buffer, its clear to see that the first 4 byte form tx_buffer are copied over and over.

The test pass for   i.MXRT685 if you insert:
dma_block_cfgs[i].source_gather_interval = 4U ;
after line 104 in zephyr/tests/drivers/dma/scatter_gather/src/test_dma_sg.c

I have no knowledge if this test have passed earlier. 

The mimxrt685_evk is not in the platform_allow, but as there is a board overlay for that board in the test, I expect the test would pass without modifications.

**To Reproduce**

```
west build -d build_evk_dma_test -b mimxrt685_evk/mimxrt685s/cm33 zephyr/tests/drivers/dma/scatter_gather/ -p always
west flash  -d build_evk_dma_test -r jlink
```

**Expected behavior**

Test shall pass without modifications for both boards.

**Impact**

Ugly board/MCU dependent if/else Statement in code. 

**Environment (please complete the following information):**

 - OS:  Ubuntu
 - Toolchain zephyr 0.16.1
 - Commit SHA or Version used
commit 7254a615307d732cf5b4535baf5c6360261c03fa (HEAD -> main, origin/main, origin/HEAD)
Author: Carles Cufi <carles.cufi@nordicsemi.no>
Date:   Thu Feb 6 11:08:31 2025 +0100

    scripts: runners: nrf: Fix flashing of FLPR/PPR on nRF54H20
    
    After commit aaefaad, flashing of the FLPR and PPR cores in the nRF54H20
    was broken due to those cores missing from the _get_core() function that
    retrieves the core to program, triggering an exception. Fix it by
    referencing those cores and mapping them to the application core.
    
    Signed-off-by: Carles Cufi <carles.cufi@nordicsemi.no>

dleach02, 2025-02-19:
@hakehuang can you run this test against the RT685?

I'm not quite understanding why this would fail and why setting source_gather_interval=4 in the test code would fix this for the author.

hakehuang, 2025-02-19:
> I'm not quite understanding why this would fail and why setting source_gather_interval=4 in the test code would fix this for the author.

@dleach02 the scatter_gather only enables below platforms in testcase.yaml.
```
      - frdm_k64f
      - mimxrt1010_evk
      - mimxrt1060_evk/mimxrt1062/qspi
      - lpcxpresso55s36
```

dleach02, 2025-02-19:
I'm not convinced that the test application is the problem. I actually think the lpc dma driver may have a problem with scatter/gather

janmft, 2025-02-19:
Thanks, investigations. I go with @dleach02.

My expectation is that struct [dma_block_config](https://docs.zephyrproject.org/latest/doxygen/html/structdma__block__config.html) behaves the same for every dma driver.
As source_gather_interval  is not marked as HW specific.

dleach02, 2025-02-20:
@janmft I've posted a PR that addresses the problem you reported. Thankyou for the report.

PR #86033 (2025-02-20): dma: dma_mcux_lpc: fix scatter/gather src/dst increment
The logic for setting the src_inc and dst_inc would improperly set
these to zero when gatter/scatter was enabled but the respective
gatther_interval/scatter_interval was zero (which would imply
continuous operation).

Fixes: #85403
commit 3fb8a94a (2025-02-20): dma: dma_mcux_lpc: fix scatter/gather src/dst increment

The logic for setting the src_inc and dst_inc would improperly set
these to zero when gatter/scatter was enabled but the respective
gatther_interval/scatter_interval was zero (which would imply
continuous operation).

Fixes: #85403
closed 2025-02-20 by kartben