zephyrproject-rtos/zephyr #85440: ESP32S3 not support CONFIG_SPIRAM_FETCH_INSTRUCTIONS and CONFIG_SPIRAM_RODATA

zhang-wenchao, 2025-02-08:
**Describe the bug**

ESP32S3 chip

```
CONFIG_SPIRAM_FETCH_INSTRUCTIONS=y
CONFIG_SPIRAM_RODATA=y
```

Not available, compilation error occurred.

**To Reproduce**

Use `samples/hello_world` :

Add `prj.conf`

```
# SPIRAM
CONFIG_ESP_SPIRAM=y
CONFIG_SPIRAM_MODE_OCT=y
CONFIG_SPIRAM_FETCH_INSTRUCTIONS=y
CONFIG_SPIRAM_RODATA=y

```

```
west build -p always -b esp32s3_devkitm/esp32s3/procpu
```

**Expected behavior**

Compiles fine, and built-in ram usage is reduced.

**Impact**

Cannot compile and cannot use external psram

**Logs and console output**

```
$ west build -p always -b esp32s3_devkitm/esp32s3/procpu                                                          
-- west build: making build dir /home/life/Develop/platform/zephyr-project/zephyr/samples/hello_world/build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/life/Develop/platform/zephyr-project/zephyr/samples/hello_world
-- CMake version: 3.30.5
-- Found Python3: /home/life/Develop/platform/zephyr-project/.venv/bin/python3 (found suitable version "3.11.2", minimum required is "3.10") found components: Interpreter
-- Cache files will be written to: /home/life/.cache/zephyr
-- Zephyr version: 4.0.0 (/home/life/Develop/platform/zephyr-project/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: esp32s3_devkitm, qualifiers: esp32s3/procpu
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.8 (/home/life/Develop/platform/zephyr-project/sdk/zephyr-sdk-0.16.8)
-- Found toolchain: zephyr 0.16.8 (/home/life/Develop/platform/zephyr-project/sdk/zephyr-sdk-0.16.8)
-- Found Dtc: /home/life/Develop/platform/zephyr-project/sdk/zephyr-sdk-0.16.8/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/life/Develop/platform/zephyr-project/zephyr/boards/espressif/esp32s3_devkitm/esp32s3_devkitm_procpu.dts
-- Generated zephyr.dts: /home/life/Develop/platform/zephyr-project/zephyr/samples/hello_world/build/zephyr/zephyr.dts
-- Generated pickled edt: /home/life/Develop/platform/zephyr-project/zephyr/samples/hello_world/build/zephyr/edt.pickle
-- Generated zephyr.dts: /home/life/Develop/platform/zephyr-project/zephyr/samples/hello_world/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/life/Develop/platform/zephyr-project/zephyr/samples/hello_world/build/zephyr/include/generated/zephyr/devicetree_generated.h
-- Including generated dts.cmake file: /home/life/Develop/platform/zephyr-project/zephyr/samples/hello_world/build/zephyr/dts.cmake
Parsing /home/life/Develop/platform/zephyr-project/zephyr/Kconfig
Loaded configuration '/home/life/Develop/platform/zephyr-project/zephyr/boards/espressif/esp32s3_devkitm/esp32s3_devkitm_procpu_defconfig'
Merged configuration '/home/life/Develop/platform/zephyr-project/zephyr/samples/hello_world/prj.conf'
Configuration saved to '/home/life/Develop/platform/zephyr-project/zephyr/samples/hello_world/build/zephyr/.config'
Kconfig header saved to '/home/life/Develop/platform/zephyr-project/zephyr/samples/hello_world/build/zephyr/include/generated/zephyr/autoconf.h'
-- Found GnuLd: /home/life/Develop/platform/zephyr-project/sdk/zephyr-sdk-0.16.8/xtensa-espressif_esp32s3_zephyr-elf/xtensa-espressif_esp32s3_zephyr-elf/bin/ld.bfd (found version "2.38")
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
[... 8 lines cut ...]
-- Build files have been written to: /home/life/Develop/platform/zephyr-project/zephyr/samples/hello_world/build
-- west build: building application
[1/211] Preparing syscall dependency handling

[4/211] Generating include/generated/zephyr/version.h
-- Zephyr version: 4.0.0 (/home/life/Develop/platform/zephyr-project/zephyr), build: v4.0.0-65-gd2e04b99b7b2
[206/211] Linking C executable zephyr/zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /home/life/Develop/platform/zephyr-project/zephyr/samples/hello_world/build/zephyr/zephyr_pre0.map 
: && ccache /home/life/Develop/platform/zephyr-project/sdk/zephyr-sdk-0.16.8/xtensa-espressif_esp32s3_zephyr-elf/bin/xtensa-espressif_esp32s3_zephyr-elf-gcc  -gdwarf-4 -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/xtensa/core/offsets/offsets.c.obj  -T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/home/life/Develop/platform/zephyr-project/zephyr/samples/hello_world/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/xtensa/core/libarch__xtensa__core.a  zephyr/lib/libc/picolibc/liblib__libc__picolibc.a  zephyr/lib/libc/common/liblib__libc__common.a  zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L/home/life/Develop/platform/zephyr-project/zephyr/samples/hello_world/build/zephyr  zephyr/arch/common/libisr_tables.a  -fuse-ld=bfd  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -Wl,-no-pie  -specs=picolibc.specs  -DPICOLIBC_LONG_LONG_PRINTF_SCANF  -lgcc  -T/home/life/Develop/platform/zephyr-project/modules/hal/espressif/zephyr/esp32s3/src/linker/esp32s3.rom.alias.ld  -T/home/life/Develop/platform/zephyr-project/modules/hal/espressif/zephyr/esp32s3/../../components/esp_rom/esp32s3/ld/esp32s3.rom.ld  -T/home/life/Develop/platform/zephyr-project/modules/hal/espressif/zephyr/esp32s3/../../components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld  -T/home/life/Develop/platform/zephyr-project/modules/hal/espressif/zephyr/esp32s3/../../components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld  -T/home/life/Develop/platform/zephyr-project/modules/hal/espressif/zephyr/esp32s3/../../components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld  -T/home/life/Develop/platform/zephyr-project/modules/hal/espressif/zephyr/esp32s3/../../components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld  -T/home/life/Develop/platform/zephyr-project/modules/hal/espressif/zephyr/esp32s3/../../components/soc/esp32s3/ld/esp32s3.peripherals.ld  -Wl,--wrap=longjmp -L"/home/life/Develop/platform/zephyr-project/sdk/zephyr-sdk-0.16.8/xtensa-espressif_esp32s3_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32s3_zephyr-elf/12.2.0" -lc -lgcc && cd /home/life/Develop/platform/zephyr-project/zephyr/samples/hello_world/build/zephyr && /home/linuxbrew/.linuxbrew/Cellar/cmake/3.30.5/bin/cmake -E true
/home/life/Develop/platform/zephyr-project/sdk/zephyr-sdk-0.16.8/xtensa-espressif_esp32s3_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32s3_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32s3_zephyr-elf/bin/ld.bfd: zephyr/libzephyr.a(esp_psram.c.obj):(.literal.esp_psram_init+0x3c): undefined reference to `mmu_config_psram_text_segment'
/home/life/Develop/platform/zephyr-project/sdk/zephyr-sdk-0.16.8/xtensa-espressif_esp32s3_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32s3_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32s3_zephyr-elf/bin/ld.bfd: zephyr/libzephyr.a(esp_psram.c.obj):(.literal.esp_psram_init+0x40): undefined reference to `mmu_config_psram_rodata_segment'
/home/life/Develop/platform/zephyr-project/sdk/zephyr-sdk-0.16.8/xtensa-espressif_esp32s3_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32s3_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32s3_zephyr-elf/bin/ld.bfd: zephyr/libzephyr.a(esp_psram.c.obj): in function `esp_psram_init':
/home/life/Develop/platform/zephyr-project/modules/hal/espressif/components/esp_psram/esp_psram.c:165: undefined reference to `mmu_config_psram_text_segment'
/home/life/Develop/platform/zephyr-project/sdk/zephyr-sdk-0.16.8/xtensa-espressif_esp32s3_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32s3_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32s3_zephyr-elf/bin/ld.bfd: /home/life/Develop/platform/zephyr-project/modules/hal/espressif/components/esp_psram/esp_psram.c:177: undefined reference to `mmu_config_psram_rodata_segment'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /home/linuxbrew/.linuxbrew/bin/cmake --build /home/life/Develop/platform/zephyr-project/zephyr/samples/hello_world/build
(.venv) 
```

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain Zephyr SDK
 - v4.0.0-65-gd2e04b99b7b2

zhang-wenchao, 2025-02-08:
I'm not sure if this is due to zephyr or the esp toolchain. If it's not suitable, then transfer it.

sylvioalves, 2025-02-12:
There is indeed an issue. Will be fixed.

zhang-wenchao, 2025-06-12:
Sorry, I don't have the time and test environment recently, so you can close this issue first.

marekmatej, 2025-06-25:
@mari-rv thank you for your help. The issue was addressed on hal side. Please take a look and test. zephyrproject-rtos/hal_espressif#455

zhang-wenchao, 2025-07-11:
> [@zhang-wenchao](https://github.com/zhang-wenchao) sorry, did you delete your previous message? I got the notification email but not sure what happened.

Sorry, I tested it and it works fine, I tested it wrong at first.