zephyrproject-rtos/zephyr #85462: bma4xx driver doesn't work for bma400

Charlie-83, 2025-02-09:
The bma400 has completely different register setup to the bma422 which was used as the reference.

bma4xx should be renamed or there should at least be a warning

henrikbrixandersen, 2025-02-10:
Please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

Charlie-83, 2025-02-10:
Thanks for the response. This isn't exactly a bug so none of the sections in the template really apply. Is there a better tag I should use?

Charlie-83, 2025-02-10:
To clarify, my issue is not that there is a bug in the driver that means it doesn't work with BMA400. My issue is that the driver is completely incompatible with BMA400. This incompatibility is surprising given the driver's name.

In this issue I am just suggesting renaming or clarifying this somewhere so that the next person doesn't spend time wondering why it doesn't work like I did.

Adding an entirely new driver for BMA400 would be an enhancement outside the scope of this issue.

fabiobaltieri, 2025-02-11:
hi @Charlie-83, these drivers are usually contributed, so there's not really a team to take feature reqeusts, would you be able to troubleshoot the issue and send a PR fixing it yourself? Maybe you can ask for help on chat.zephyrproject.org

chriseui, 2025-08-22:
Hi I agree with @Charlie-83, Please, Please  add a Warning I have wasted many hours trying to get this driver to work with a BMA400!!!

chriseui, 2025-08-22:
Hi, 
I'm sorry I'm not the person for that job, I'm a hardware designer just trying to write some simple test code to check the hardware. I think the issue as has been pointed out, there is no mention anywhere that this driver in not compatible with BMA400 despite it being called BMA4xx. 

The whole driver was re-written/re-packaged recently, yet still no comment in any of the source files about the incompatibility!

I fully understand open-source projects and the need for contributions, sorry.

chriseui, 2025-12-29:
You cannot use the BMA4XX driver for the BMA400, as I posted a while back the driver should be renamed BMA422. I spent many wasted hours with the BMA4xx driver.

You need to use the Bosch driver https://github.com/boschsensortec/BMA400_SensorAPI
It is written for use with Bosch's own development hardware "COINES", but it's quite easy to modify for use with nRF.