zephyrproject-rtos/zephyr #85467: Nominate myself (ajf58) as Collaborator

ajf58, 2025-02-09:
# Background

The [TSC Project Roles] defines the main roles for the Zephyr Project, including
Maintainer, Collaborator, and Contributor.

By default anyone that contributes code or documentation is a Contributor, but
with the lowest [GitHub Permission Level] of Read. For example, Contributors
with Read permission do not have the permission to add reviewers to a pull
request.

Use this template to nominate a GitHub user for the Contributor role with
Triage permission level, which allows the user to add reviewers to a pull
request and be added as a reviewer by other users.

# Nomination

## GitHub User

Provide the following information about the GitHub user:

1. Full Name: Andrew Featherstone
1. GitHub username: @ajf58 
1. Organization (optional)

## Supporting Documents

1. https://github.com/zephyrproject-rtos/zephyr/pull/84974#issuecomment-2627408580
2. #84548 
3. #77368
4. #80781

Add links to 3-5 GitHub pull requests, in the Zephyr project, authored or
reviewed by the GitHub user that demonstrate the user's dedication to the
Zephyr project.

[TSC Project Roles]: <https://docs.zephyrproject.org/latest/project/project_roles.html>
[GitHub Permission Level]: <https://docs.github.com/en/organizations/managing-access-to-your-organizations-repositories/repository-roles-for-an-organization>

nashif, 2025-02-09:
Thank you for your contribution to the Zephyr project. 

An invite was sent to you to join the project. You can find the invite here: https://github.com/zephyrproject-rtos/zephyr/invitations.

Note that further access and permissoins to the project will be provided following the project [guidelines](https://docs.zephyrproject.org/latest/project/project_roles.html#tsc-project-roles).

kartben, 2025-02-22:
@ajf58 did you accept that invitation? I've noticed your reviews still don't have a green check mark?

ajf58, 2025-02-22:
> [@ajf58](https://github.com/ajf58) did you accept that invitation? I've noticed your reviews still don't have a green check mark?

Yes, I think so, I am listed as an "outside collaborator" on the relevant projects. I don't think there's a green tick on #86041 because it doesn't match the listing at https://github.com/zephyrproject-rtos/zephyr/blob/main/MAINTAINERS.yml#L3600 .

PR #85469 (2025-02-12): MAINTAINERS: rpi_pico: Add @ajf58 as collaborator
Following a nudge at https://github.com/zephyrproject-rtos/zephyr/pull/84974#issuecomment-2627408580 , append myself to the relevant list of collaborators.

Closes #85467