zephyrproject-rtos/zephyr #85482: Drivers: NXP :  dependency for CONFIG_NXP_LP_FLEXCOMM missing hal include driver_lpflexcomm

hakehuang, 2025-02-10:
**Describe the bug**

in https://github.com/zephyrproject-rtos/zephyr/pull/85446 
there is one issue revealed for driver dependency

for mimxrt700_evk/mimxrt798s/cm33_cpu0
```
/home/shared/disk/zephyr_project/zephyr_test/zephyr/include/zephyr/drivers/mfd/nxp_lp_flexcomm.h:9:10: fatal error: fsl_lpflexcomm.h: No such file or directory
    9 | #include "fsl_lpflexcomm.h"
      |  
```

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 - mimxrt700_evk/mimxrt798s/cm33_cpu0
 - What have you tried to diagnose or workaround this issue?
 - build analysis
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
 - No

**To Reproduce**

```
west twister -p mimxrt700_evk/mimxrt798s/cm33_cpu0 -T samples/basic/minimal/
```

**Expected behavior**

build pass

**Impact**

minimal checking and driver dependency

**Logs and console output**

```
In file included from /home/shared/disk/zephyr_project/zephyr_test/zephyr/drivers/mfd/mfd_nxp_lp_flexcomm.c:14:
/home/shared/disk/zephyr_project/zephyr_test/zephyr/include/zephyr/drivers/mfd/nxp_lp_flexcomm.h:9:10: fatal error: fsl_lpflexcomm.h: No such file or directory
    9 | #include "fsl_lpflexcomm.h"
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.

```

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: b10587f2d6190dbc8897702540bce63b9d448cc4

mmahadevan108, 2025-02-18:
@hakehuang , I am unable to reproduce this build failure. Below is the command I used
`west build -p -b frdm_mcxn947/mcxn947/cpu0 zephyr/samples/basic/minimal`

nashif, 2025-02-18:
You do not need the minimal sample for this....

```west build -p -b frdm_mcxn947/mcxn947/cpu0   samples/hello_world/ -- -DCONFIG_SERIAL=n```

will get you the same build failure.

hakehuang, 2025-02-19:
> west build -p -b frdm_mcxn947/mcxn947/cpu0 zephyr/samples/basic/minimal

Thanks Anas's help, if using west command, there are some extra _config need add for build parameter.

closed 2025-02-27 by fabiobaltieri