zephyrproject-rtos/zephyr #85501: esp32: use SRAM1 as DRAM

arifbalik, 2025-02-10:
**Is your feature request related to a problem? Please describe.**

Enabling WiFi, BLE and mbedTLS which is a pretty common use case, drains all the memory in SRAM2 (192kB) which is used for DRAM. Currently SRAM1 is entirely dedicated for IRAM (SRAM0 + SRAM1 = 224kB)

**Describe the solution you'd like**

It would be nice to be able to allocate some of the SRAM1 for DRAM

**Describe alternatives you've considered**

SPIRAM is an alternative to this but unfortunately some versions of ESP32 does not have this. I am using ESP32-D0WD-V3, which does not have an SPIRAM.

**Additional context**

Here is my image only with WiFi and mbedTLS enabled

[image]

rftafas, 2025-02-10:
Espressif sells ESP32 in chip format and module format. This is the chip PN. Make sure you are using a down to chip design first, because PSRAM is indeed not on the chip for this device.

arifbalik, 2025-02-10:
> Espressif sells ESP32 in chip format and module format. This is the chip PN. Make sure you are using a down to chip design first, because PSRAM is indeed not on the chip for this device.

Forgot to mention that, my bad. Module is ESP32-WROOM-32E. So there is no external RAM on the module as well.

closed 2025-04-07 by kartben