zephyrproject-rtos/zephyr #85522: Zephyr 4.1 security related modules update overview

fabiobaltieri, 2025-02-10:
Umbrella issue for tracking security related modules which may require an update before Zephyr 4.0.

- [x] https://github.com/zephyrproject-rtos/mbedtls (@ceolin , @d3zd3z , @tomi-font)
  - updated recently
- [x] https://github.com/zephyrproject-rtos/tinycrypt (@ceolin)
  - this is being phased out
- [x] https://github.com/zephyrproject-rtos/trusted-firmware-m (@d3zd3z, @tomi-font)
  - updated recently, probably fine
- [ ] https://github.com/zephyrproject-rtos/trusted-firmware-a (@povergoing, @sgrrzhf)
  - last updated 8 months ago, may need an uprev?
- [ ] https://github.com/zephyrproject-rtos/uoscore-uedhoc (@rlubos , @StefanHri)
  - last updated 8 months ago, may need an uprev?

fabiobaltieri, 2025-02-10:
Hey I've marked few items that I think are already up to date as done but feel free to double check if you know better, thanks

rlubos, 2025-02-11:
I've opened a PR updating the `uoscore-uedhoc` module revision to the latest: https://github.com/zephyrproject-rtos/zephyr/pull/85574

ceolin, 2025-02-12:
@fabiobaltieri FYI mbedTLS we are using is the latest release 3.6.2. We are good there.

closed 2025-03-05 by fabiobaltieri