zephyrproject-rtos/zephyr #85523: Zephyr 4.1 module update overview

fabiobaltieri, 2025-02-10:
Umbrella issue for modules which may require an update before Zephyr 4.1.

- [ ] https://github.com/zephyrproject-rtos/acpica (@najumon1980)
- [ ] https://github.com/zephyrproject-rtos/babblesim-manifest (@aescolar)
- [ ] https://github.com/zephyrproject-rtos/canopennode (@henrikbrixandersen)
- [ ] https://github.com/zephyrproject-rtos/chre (@yperess)
- [ ] https://github.com/zephyrproject-rtos/cmsis (@stephanosio, @microbuilder, @povergoing)
- [ ] https://github.com/zephyrproject-rtos/cmsis-dsp (@XenuIsWatching, @stephanosio)
- [ ] https://github.com/zephyrproject-rtos/cmsis-nn (@XenuIsWatching, @stephanosio)
- [ ] https://github.com/zephyrproject-rtos/edtt (@aescolar, @wopu-ot, @thoh-ot)
- [ ] https://github.com/zephyrproject-rtos/fatfs (@de-nordic)
- [ ] https://github.com/zephyrproject-rtos/hostap (@krish2718, @jukkar)
- [ ] https://github.com/zephyrproject-rtos/liblc3 (@Casper-Bonde-Bose, @MariuszSkamra)
- [ ] https://github.com/zephyrproject-rtos/libmetal (@carlocaione, @arnopo)
- [ ] https://github.com/zephyrproject-rtos/littlefs (???)
- [ ] https://github.com/zephyrproject-rtos/loramac-node (@Mani-Sadhasivam)
- [ ] https://github.com/zephyrproject-rtos/lvgl (@faxe1008, @brgl, @pdgendt)
- [ ] https://github.com/zephyrproject-rtos/lz4 (@parthitce)
- [ ] https://github.com/zephyrproject-rtos/mcuboot (@d3zd3z, @de-nordic, @nordicjm)
- [ ] https://github.com/zephyrproject-rtos/nanopb (@pdgendt)
- [ ] https://github.com/zephyrproject-rtos/open-amp (@carlocaione @uLipe @iuliana-prodan)
- [ ] https://github.com/zephyrproject-rtos/openthread (@rlubos)
- [ ] https://github.com/zephyrproject-rtos/percepio (@eriktamlin, @aronlander-pe)
- [ ] https://github.com/zephyrproject-rtos/picolibc (@keith-packard, @stephanosio)
- [ ] https://github.com/zephyrproject-rtos/sof (@kv2019i)
- [ ] https://github.com/zephyrproject-rtos/tflite-micro (@laurenmurphyx64)
- [ ] https://github.com/zephyrproject-rtos/thrift (@cfriedt)
- [ ] https://github.com/zephyrproject-rtos/zcbor (@de-nordic)
- [ ] https://github.com/zephyrproject-rtos/zephyr-lang-rust (@d3zd3z)
- [ ] https://github.com/zephyrproject-rtos/zscilib (@microbuilder)

Folks that are tagged please comment and confirm the status of your modules with respect to readiness for the 4.1 release. Thanks.

cfriedt, 2025-02-10:
No update to thrift yet.

Basically, this patch needs to be reworked to address a number of corner cases.

https://github.com/apache/thrift/pull/2689/files

henrikbrixandersen, 2025-02-10:
My update to CANopenNode unfortunately will not be ready in time for the v4.1 feature freeze.

jukkar, 2025-02-11:
hostap updated to upstream v2.11

rlubos, 2025-02-11:
OpenThread is up-to-date (was updated last month).

pdgendt, 2025-02-11:
No changes needed for `nanopb`

de-nordic, 2025-02-11:
@oyvindronningstad Can you take a look into updating zcbor?

de-nordic, 2025-02-11:
@nashif @carlescufi The FAT FS update is here:
https://github.com/zephyrproject-rtos/zephyr/pull/85230
https://github.com/zephyrproject-rtos/fatfs/pull/19
please take a look.

de-nordic, 2025-02-11:
@fabiobaltieri LittleFS has been updated to the latest available version from upstream (https://github.com/littlefs-project/littlefs/commits/master/), on Jan 17, 2025 (https://github.com/zephyrproject-rtos/littlefs/commit/424a5e0d5ab273ab6833453288ccf64342c7a5cd), and there are no updates planned for this release.

XenuIsWatching, 2025-02-12:
`cmsis-nn` and `tflite-micro` is in a PR here https://github.com/zephyrproject-rtos/zephyr/pull/85547

There is no update for cmsis-dsp

faxe1008, 2025-02-12:
`lvgl` was updated last week-ish. No update for 4.1 pending.

oyvindronningstad, 2025-02-12:
zcbor is at 0.9.1 in Zephyr, which is the latest version.

eriktamlin, 2025-02-12:
`percepio` module should be good to go.

microbuilder, 2025-02-12:
No changes for zscilib

nordicjm, 2025-02-13:
MCUboot update: https://github.com/zephyrproject-rtos/zephyr/pull/85671

closed 2025-03-05 by fabiobaltieri