zephyrproject-rtos/zephyr #85542: LLVM toolchain no longer used in CI

thughes, 2025-02-11:
**Describe the bug**

With https://github.com/zephyrproject-rtos/zephyr/pull/85091, it appears that the LLVM toolchain is not used to verify changes in CI.

Here's an example that will fail to compile if `clang` is used, but now successfully passes in CI: https://github.com/zephyrproject-rtos/zephyr/pull/85540.

**Impact**
Downstream uses of Zephyr that build with `clang` are likely to be broken by upstream changes since they are no longer verified.

I have several pending PRs that are `clang`-specific, but will not be tested in CI with `clang`: 
* https://github.com/zephyrproject-rtos/zephyr/pull/84063
* https://github.com/zephyrproject-rtos/zephyr/pull/83922
* https://github.com/zephyrproject-rtos/zephyr/pull/83830

thughes, 2025-02-11:
@fabiobaltieri @nashif @kartben

nashif, 2025-02-11:
the goal is to expand toolchain coverage in the main workflow and not duplicate building everything in a secondary workflow like we did until now and more workflows when we get proper llvm support in the toolchain.

thughes, 2025-02-11:
> the goal is to expand toolchain coverage in the main workflow and not duplicate building everything in a secondary workflow like we did until now and more workflows when we get proper llvm support in the toolchain.

I understand the goal, but this is a regression and will allow changes that do not compile with `clang` to be submitted, which will cause problems for downstream users of `clang`. For example, there are many warnings that `clang` warns about in the exact same code that `gcc` does not warn about. The reason is that `gcc` doesn't perform warning analysis until after optimizations. You can see in the PRs in https://github.com/zephyrproject-rtos/zephyr/issues/85542#issue-2843993112 how many files this affects. If `clang` does not run on these files then changes will introduce new instances of warnings, which will break the `clang` builds.

fabiobaltieri, 2025-02-11:
Retagging as enhancement as that's not strictly a bug and it is somewhat working as intended, but sicne this is causing issues that used to be detected to go undetected now I think it's fair to say that this is a regression from the CI perspective.

thughes, 2025-03-17:
Since the LLVM CI has been disabled Zephyr, changes have been introduced that cause Zephyr to fail to build. One of the bugs was a memory corruption issue caught by `clang`:

* https://github.com/zephyrproject-rtos/zephyr/pull/87243
* https://github.com/zephyrproject-rtos/zephyr/pull/87244

Can we please re-enable so that we can prevent regressions and bugs?

thughes, 2025-03-17:
cc @tejlmand

fabiobaltieri, 2025-03-18:
@keith-zephyr can we discuss this in the process wg?

fabiobaltieri, 2025-03-18:
Yes I'm [well aware](https://github.com/zephyrproject-rtos/zephyr/issues/63738) of it but at least llvm is available without a license and we had a minimum of CI coverage for it in the past, so people built applications and downstream processes on top of it, and now that the toolchain coverage has been drastically reduced, people are seeing downstream breakages that would have been previously discovered upstream, which in turns challenges the sustainability of downstream processes that have been acceptable in the past.

Would it be possible to setup the current system to have a comparable coverage than what we had before without reintroducing the whole workflow? It was a single host job going fairly fast, it'd normally finish before the first twister run even.

marc-hb, 2025-03-18:
> We need a proper toolchain test strategy that does not involve building every sample/test on each toolchain and on each PR.

By definition, nightly runs do not run on each PR. Could this have caught the bugs above post-merge? Catching bugs post-merge is not as good as pre-merge but better than none at all.

Combining multiple PRs pre-merge is another, popular option:
https://github.blog/engineering/engineering-principles/how-github-uses-merge-queue-to-ship-hundreds-of-changes-every-day/
https://about.gitlab.com/blog/2020/01/30/all-aboard-merge-trains/

Main drawback: failures from unrelated PRs confuse and upset submitters. This can be mitigated thanks to multiple levels and running quicker checks in isolation first before combining: https://chromium.googlesource.com/chromium/src/+/HEAD/docs/infra/cq.md#Modes

Just sharing some "prior art". No silver bullet.

nashif, 2025-03-18:
proposed pr https://github.com/zephyrproject-rtos/zephyr/pull/87306

fabiobaltieri, 2025-03-18:
> By definition, nightly runs do not run on each PR. Could this have caught the bugs above post-merge? Catching bugs post-merge is not as good as pre-merge but better than not at all.

The current setup is of a per-PR workflow and a "push" workflow which runs after each merge batch, there's no "nightly as such", though there's a weekly but that's hard to keep up with.

For the merge, queue, it's been suggested few times, as far as I understand (have not looked into that personally) it really wants to run CI on each PR at a time so that doesn't really scale with the amount of traffic the project has. Unfortunately.

thughes, 2025-03-18:
One idea: would passing `--build-only` to twister make the LLVM job take a shorter amount of time? I don't expect there to be many cases where the code will compile with both `gcc` and `clang` and pass at runtime with `gcc`, but not `clang` (though it's definitely possible). Just trying to see if there are compromises to help with CI load.

marc-hb, 2025-03-19:
> For the merge, queue, it's been suggested few times, as far as I understand (have not looked into that personally) it really wants to run CI on each PR at a time so that doesn't really scale with the amount of traffic the project has. Unfortunately.

The main idea behind "trains" and "queues" is the exact opposite: *not* testing each PR at a time in order to scale to a large amount of traffic.

According to their blog, GitHub uses a queue internally to merge one Pull Request every 17 min in average. In the last 30 days, this repo had 2 commits/hour in average.

fabiobaltieri, 2025-03-19:
> The main idea behind "trains" and "queues" is the exact opposite: _not_ testing each PR at a time in order to scale to a large amount of traffic.
>
> According to their blog, GitHub uses a queue [...]

^^ @stephanosio

carlescufi, 2025-04-09:
@nashif @thughes should we close this issue? I see that https://github.com/zephyrproject-rtos/zephyr/actions/workflows/clang.yaml is enabled now and building on CI.

carlescufi, 2025-04-09:
@nashif why aren't we buildilng with clang on Pull Request, and instead only on push? I suspect it's because we were using `pull_request_target` but now failures are not seen on PR.

thughes, 2025-04-18:
> why aren't we buildilng with clang on Pull Request, https://github.com/zephyrproject-rtos/zephyr/pull/87306?

It seems to be a result of concerns around the resource usage if it's done on every PR.

It does look like @kartben is [paying attention to failures on push](https://github.com/zephyrproject-rtos/zephyr/pull/87088#discussion_r2049295801), so I'm ok with this setup. Some things will get merged that break clang, but as long as they're addressed quickly, it's ok with me. If we come across a situation where there's some type of change that breaks things in a way that's difficult to fix quickly, we can re-evaluate.

fabiobaltieri, 2025-04-20:
> It does look like [@kartben](https://github.com/kartben) is [paying attention to failures on push](https://github.com/zephyrproject-rtos/zephyr/pull/87088#discussion_r2049295801), so I'm ok with [...]

The result of that run is very visible on the top of the dashboard that everyone in the release team uses to merge patches and if there's a breakage whoever is merging is responsible to at least triage and start the process of fixing the issue (directly or indirectly) or roollback the change if it's bad enough. Ideally it'd run on PRs as it used to but for now at least there is someone monitoring, my understanding is still that there are plans to spread more of these on PRs in the future anyway.