zephyrproject-rtos/zephyr #85551: DNS recv error (-22) in system log on MDNS activity

hjuul, 2025-02-11:
**Describe the bug**

When physically connected to the local network, my Zephyr application logs the following error every few hours:

```
<err> net_dns_dispatcher: DNS recv error (-22)
```

This is what I think happens:
1. Some host on the network sends an MDNS PTR query.
2. A printer responds with a long answer (so long that it doesn't fit into a single IP frame)
3. The host sends two MDNS queries to get more information about the printer service
     - likely because the "Answers" section is too large to fit into one query
4. Zephyr MDNS responder throws an error when parsing. See logs below.

I've found that the error code (EINVAL) comes from the responder callback [here](https://github.com/zephyrproject-rtos/zephyr/blob/30102936bcab8a5de4b5f156d6e4a546f8e332ba/subsys/net/lib/dns/dispatcher.c#L64C1-L66C31).

**To Reproduce**

Probably very hard to reproduce, as it depends on specific network activity.
In my case it seems to be caused by a very  verbose OKI printer.

**Impact**

Annoyance.

**Logs and console output**

This is the log when it happens. I have also attached a tcpdump of this exact activity: [wireshark_subset.zip](https://github.com/user-attachments/files/18746738/wireshark_subset.zip)

 ```
00> [00:36:17.650,000] <dbg> net_mdns_responder: (net_socket_service): Received 2 queries from 192.168.14.113
00> [00:36:17.650,000] <dbg> net_mdns_responder: (net_socket_service): [2] query PTR/IN label ._ipps._tcp.local (17 bytes)
00> [00:36:17.650,000] <dbg> net_mdns_responder: (net_socket_service): [1] query PTR/IN label ._ipp._tcp.local (16 bytes)
00> [00:36:17.851,000] <dbg> net_mdns_responder: (net_socket_service): Received 3 queries from 192.168.14.113
00> [00:36:17.851,000] <dbg> net_mdns_responder: (net_socket_service): [2] query ANY/IN label .fluffy-thinkpad.local (22 bytes
00> [00:36:17.914,000] <dbg> net_mdns_responder: (net_socket_service): Received 6 queries from 192.168.14.113
00> [00:36:17.914,000] <dbg> net_mdns_responder: (net_socket_service): [6] query SRV/IN label .OKI-C844-AE8632._ipps._tcp.loca
00> [00:36:17.914,000] <dbg> net_mdns_responder: (net_socket_service): [5] query TXT/IN label .OKI-C844-AE8632._ipp._tcp.local
00> [00:36:17.914,000] <dbg> net_mdns_responder: (net_socket_service): [4] query TXT/IN label .OKI-C844-AE8632._ipp._tcp.local
00> [00:36:17.914,000] <dbg> net_mdns_responder: (net_socket_service): [3] query TXT/IN label .OKI-C844-AE8632._ipp._tcp.local
00> [00:36:17.914,000] <dbg> net_mdns_responder: (net_socket_service): [2] query TXT/IN label .OKI-C844-AE8632._ipp._tcp.local
00> [00:36:17.914,000] <dbg> net_mdns_responder: (net_socket_service): [1] query TXT/IN label .OKI-C844-AE8632._ipp._tcp.local
00> [00:36:17.914,000] <err> net_dns_dispatcher: DNS recv error (-22)
```

**Environment:**

 - OS: Linux
 - Toolchain: zephyr-sdk-0.16.9
 - Commit SHA or Version used: `v3.7-branch` at `30102936bca`
 - Hardware: Custom board based on stm32f777

**Additional context**

Just in case it can provide some clues: When this happens it usually happens multiple times, and the timestamps indicate that there is some retry mechanism at play here (1s after first error, then 2s, 4s, 8s and 16s)

```
[16:08:13.111,000] <err> net_dns_dispatcher: dns recv error (-22)
[16:08:14.112,000] <err> net_dns_dispatcher: dns recv error (-22)
[16:08:16.112,000] <err> net_dns_dispatcher: dns recv error (-22)
[16:08:20.052,000] <err> net_dns_dispatcher: dns recv error (-22)
[16:08:27.897,000] <err> net_dns_dispatcher: dns recv error (-22)
[16:08:43.828,000] <err> net_dns_dispatcher: dns recv error (-22)
```

jukkar, 2025-02-11:
@hjuul would you be able to capture the network traffic and provide a pcap file, it would help to replicate the issue.

jukkar, 2025-02-11:
> In my case it seems to be caused by a very verbose OKI printer.

The OKI printer is sending large (>512 bytes) DNS message but does not use EDNS0 extension mechanism https://datatracker.ietf.org/doc/html/rfc6891.
Anyway, zephyr DNS max record length is currently 512 bytes as dictated by RFC 1035 https://www.rfc-editor.org/rfc/rfc1035 and also atm EDNS0 is not supported.
If you want to experiment to receive larger frames you could try this (not tested)
```
diff --git a/include/zephyr/net/dns_resolve.h b/include/zephyr/net/dns_resolve.h
index 14938031717..be58c38b148 100644
--- a/include/zephyr/net/dns_resolve.h
+++ b/include/zephyr/net/dns_resolve.h
@@ -51,7 +51,7 @@ enum dns_query_type {
 #define DNS_BUF_TIMEOUT K_MSEC(500) /* ms */
 
 /* This value is recommended by RFC 1035 */
-#define DNS_RESOLVER_MAX_BUF_SIZE      512
+#define DNS_RESOLVER_MAX_BUF_SIZE      1232
 
 /* Make sure that we can compile things even if CONFIG_DNS_RESOLVER
  * is not enabled.
```

hjuul, 2025-02-11:
Ok, thanks. This is not a big issue for me, just something I noticed and wanted to report.
But maybe Zephyr should ignore this condition instead of reporting an error?

PR #85654 (2025-02-14): net: dns: Log with debug level instead of error level when recv fails
This condition can happen if there is MDNS activity on the network that is either not according to specifications or not supported by Zephyr.

Lowering the log level from ERR to DBG, since this does not indicate an error in the Zephyr application.

Fixes #85551
closed 2025-02-14 by kartben