zephyrproject-rtos/zephyr #85559: samples/drivers/flash_shell/sample.drivers.flash.shell fails at nrf54h20

nordic-piks, 2025-02-11:
**Describe the bug**
The tests:
- samples/drivers/flash_shell/sample.drivers.flash.shell

Observed for
- nrf54h20dk@0.9.0/nrf54h20/cpuapp

**To Reproduce**
Steps to reproduce the behavior:
1. have nrf54h20dk@0.9.0/nrf54h20/cpuapp connected
2. go to your zephyr dir
3. call ./scripts/twister -T  samples/drivers/flash_shell -p nrf54h20dk@0.9.0/nrf54h20/cpuapp --device-testing --device-serial /dev/ttyACM0 -v --inline-logs
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
INFO: send command: flash page_info 0
DEBUG: #: [1;32muart:~$ [mflash page_info 0
DEBUG: #: [1;31mNo flash device specified (required)
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 0.16.8
 - Commit SHA or Version used: v4.0.0-4840-g67e07b7a4116

nordic-piks, 2025-02-11:
Found after test was enabled for execution at HW: https://github.com/zephyrproject-rtos/zephyr/pull/85033

de-nordic, 2025-02-11:
Provided command does not work:

[image]
"FILTERED", I have been running the command on the nrf54l15 so changed to /dev/ttyACM1

Anyway, I have built the shell sample:
```
west build -b nrf54l15dk/nrf54l15/cpuapp -d builds/flash_shell_54 zephyr/samples/drivers/flash_shell/
```
and uploaded to to the board, and got:

[image]

nordic-piks, 2025-02-12:
@de-nordic the issue was found originally at nrf54h20, please reproduce at this board.
Regarding nrf54LXX, the console output there is a little bit different than in the description, thus I removed it from and I will add comment with actual results at nrf54LXX.

nordic-piks, 2025-02-12:
For:
- nrf54l15dk/nrf54l05/cpuapp
- nrf54l15dk/nrf54l15/cpuapp
I have output:
INFO: send command: flash page_info 0
ERROR: Read from device timeout occurred

de-nordic, 2025-02-12:
The nrf54h20 is missing chosen `zephyr,flash-controller`, so the flash shell sample does not know what to default to.

de-nordic, 2025-02-12:
> 
> updated filter to catch this:
> 
> [#85675](https://github.com/zephyrproject-rtos/zephyr/pull/85675)

Thanks for fixing @nashif . I will still annoy some people to do some dts fixing.

closed 2025-04-29 by github-actions[bot]