zephyrproject-rtos/zephyr #85595: samples: samples/basic/minimal: it82xx2_evb/it82202ax: error: Build failure

hakehuang, 2025-02-11:
**Describe the bug**

config minimal case device tree error

build is error on v4.0.0-4817-gad6fefe6afc1 on it82xx2_evb/it82202ax
path is samples/basic/minimal

see logs for details

**To Reproduce**

```
# west twister -p it82xx2_evb/it82202ax -T samples/basic/minimal -- -DCONF_FILE='common-runtime.conf;no-timers.conf;no-mt.conf;riscv.conf'
```
2. See error

```
Build failure
```

**Expected behavior**
test pass

**Impact**

**Logs and console output**

```
    
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /home/shared/disk/zephyr_project/zephyr_test/zephyr/twister-out/it82xx2_evb_it82202ax/zephyr/samples/basic/minimal/sample.minimal.riscv.runtime/zephyr/zephyr_pre0.map 
: && ccache /home/ubuntu/zephyr-sdk/zephyr-sdk-0.17.0-rc1/riscv64-zephyr-elf/bin/riscv64-zephyr-elf-gcc  -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/riscv/core/offsets/offsets.c.obj  -T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/home/shared/disk/zephyr_project/zephyr_test/zephyr/twister-out/it82xx2_evb_it82202ax/zephyr/samples/basic/minimal/sample.minimal.riscv.runtime/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/riscv/core/libarch__riscv__core.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/lib/libc/common/liblib__libc__common.a  zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/flash/libdrivers__flash.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  zephyr/drivers/serial/libdrivers__serial.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L/home/shared/disk/zephyr_project/zephyr_test/zephyr/twister-out/it82xx2_evb_it82202ax/zephyr/samples/basic/minimal/sample.minimal.riscv.runtime/zephyr  zephyr/arch/common/libisr_tables.a  -Wl,--fatal-warnings  -mabi=ilp32  -march=rv32imac_zicsr_zifencei  -fuse-ld=bfd  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -Wl,-no-pie -L\"/home/ubuntu/zephyr-sdk/zephyr-sdk-0.17.0-rc1/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/12.2.0/rv32imac_zicsr_zifencei/ilp32\"  -lgcc && cd /home/shared/disk/zephyr_project/zephyr_test/zephyr/twister-out/it82xx2_evb_it82202ax/zephyr/samples/basic/minimal/sample.minimal.riscv.runtime/zephyr && /snap/cmake/1441/bin/cmake -E true
/home/ubuntu/zephyr-sdk/zephyr-sdk-0.17.0-rc1/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/12.2.0/../../../../riscv64-zephyr-elf/bin/ld.bfd: zephyr/libzephyr.a(soc.c.obj): in function `chip_run_pll_sequence':
/home/shared/disk/zephyr_project/zephyr_test/zephyr/soc/ite/ec/it8xxx2/soc.c:181: undefined reference to `timer_5ms_one_shot'
/home/ubuntu/zephyr-sdk/zephyr-sdk-0.17.0-rc1/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/12.2.0/../../../../riscv64-zephyr-elf/bin/ld.bfd: zephyr/drivers/flash/libdrivers__flash.a(flash_ite_it8xxx2.c.obj): in function `ramcode_flash_cmd_write':
/home/shared/disk/zephyr_project/zephyr_test/zephyr/drivers/flash/flash_ite_it8xxx2.c:297: undefined reference to `z_impl_k_sem_take'
/home/ubuntu/zephyr-sdk/zephyr-sdk-0.17.0-rc1/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/12.2.0/../../../../riscv64-zephyr-elf/bin/ld.bfd: /home/shared/disk/zephyr_project/zephyr_test/zephyr/drivers/flash/flash_ite_it8xxx2.c:280: undefined reference to `z_impl_k_sem_give'
/home/ubuntu/zephyr-sdk/zephyr-sdk-0.17.0-rc1/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/12.2.0/../../../../riscv64-zephyr-elf/bin/ld.bfd: zephyr/drivers/flash/libdrivers__flash.a(flash_ite_it8xxx2.c.obj): in function `arch_irq_unlock':
/home/shared/disk/zephyr_project/zephyr_test/zephyr/include/zephyr/arch/riscv/arch.h:259: undefined reference to `z_impl_k_sem_take'
/home/ubuntu/zephyr-sdk/zephyr-sdk-0.17.0-rc1/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/12.2.0/../../../../riscv64-zephyr-elf/bin/ld.bfd: zephyr/drivers/flash/libdrivers__flash.a(flash_ite_it8xxx2.c.obj): in function `ramcode_flash_cmd_erase':
/home/shared/disk/zephyr_project/zephyr_test/zephyr/drivers/flash/flash_ite_it8xxx2.c:322: undefined reference to `z_impl_k_sem_give'
/home/ubuntu/zephyr-sdk/zephyr-sdk-0.17.0-rc1/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/12.2.0/../../../../riscv64-zephyr-elf/bin/ld.bfd: zephyr/drivers/flash/libdrivers__flash.a(flash_ite_it8xxx2.c.obj): in function `k_sem_init':
/home/shared/disk/zephyr_project/zephyr_test/zephyr/twister-out/it82xx2_evb_it82202ax/zephyr/samples/basic/minimal/sample.minimal.riscv.runtime/zephyr/include/generated/zephyr/syscalls/kernel.h:1085: undefined reference to `z_impl_k_sem_init'
/home/ubuntu/zephyr-sdk/zephyr-sdk-0.17.0-rc1/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/12.2.0/../../../../riscv64-zephyr-elf/bin/ld.bfd: zephyr/drivers/serial/libdrivers__serial.a(uart_ite_it8xxx2.c.obj):(.rodata.uart_it8xxx2_cfg_1+0x4): undefined reference to `__device_dts_ord_170'
/home/ubuntu/zephyr-sdk/zephyr-sdk-0.17.0-rc1/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/12.2.0/../../../../riscv64-zephyr-elf/bin/ld.bfd: zephyr/drivers/serial/libdrivers__serial.a(uart_ite_it8xxx2.c.obj):(.rodata.uart_it8xxx2_cfg_0+0x4): undefined reference to `__device_dts_ord_147'
collect2: error: ld returned 1 exit status

```

**Environment (please complete the following information):**
 - OS: (e.g. Linux ): posix
 - Toolchain (e.g Zephyr SDK): zephyr
 - Commit SHA or Version used: v4.0.0-4817-gad6fefe6afc1

closed 2025-02-11 by dleach02

dleach02, 2025-02-14:
@hakehuang the error in this one occurs because the target platform doesn't seem to allow you to set CONFIG_FLASH to 'n'. This then compiles in the driver that counts on some kernel primitives that don't get compiled in. So a dependency problem.

hakehuang, 2025-02-14:
> [@hakehuang](https://github.com/hakehuang) the error in this one occurs because the target platform doesn't seem to allow you to set CONFIG_FLASH to 'n'. This then compiles in the driver that counts on some kernel primitives that don't get compiled in. So a dependency problem.

@dleach02 that why I want to raise this issue to architecture team, because it is very clear that the kconfig and dts need some design mechanism to get locked, otherwise this mess was, is and will continues be there.