zephyrproject-rtos/zephyr #85608: subsys: logging: log_backend_fs: logging.backend.fs.* does not build

katgiadla, 2025-02-11:
**Describe the bug**
Tests

- `logging.backend.fs.automounted`
- `logging.backend.fs.manualmounted`

do not build.

Observed for `nrf52840dk/nrf52840`.

**To Reproduce**
Steps to reproduce the behavior:
1. have `nrf52840dk/nrf52840` connected
2. go to your zephyr dir
3. call `./scripts/twister -s logging.backend.fs.automounted -s logging.backend.fs.manualmounted -p nrf52840dk/nrf52840 --device-testing --device-serial /dev/ttyACM0 -v --inline-logs --west-flash="--erase"`
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
In file included from zephyr/tests/subsys/logging/log_backend_fs/src/log_fs_test.c:18:
zephyr/tests/subsys/logging/log_backend_fs/src/log_fs_test.c:35:17: error: unknown type name 'log_format_func_t'
   35 | FAKE_VALUE_FUNC(log_format_func_t, log_format_func_t_get, uint32_t);
      |                 ^~~~~~~~~~~~~~~~~
```

**Environment (please complete the following information):**
- OS: Ubuntu 20.04.1 LTS
- Toolchain Zephyr SDK 0.17.0
- Commit SHA or Version used: `v4.0.0-4796-ge4043`

nordic-krch, 2025-02-13:
It got broken by dc97cc0cd529702508e1ff0943dfee3e1f7fa7fb. @joerchan can you take a look?

joerchan, 2025-02-13:
This was already fixed: 

fix #https://github.com/zephyrproject-rtos/zephyr/pull/85460

closed 2025-02-13 by joerchan