zephyrproject-rtos/zephyr #85623: SEGGER Systemview does not show names for all events

HyperionCoding, 2025-02-11:
**Describe the bug**
SEGGER Systemview does not show all names defined in [SYSVIEW_Zephyr.txt](../blob/main/subsys/tracing/sysview/SYSVIEW_Zephyr.txt), in my case smallest event number that I can see not working is 95.

[image]

Definition in ~/.config/SEGGER/SYSVIEW_Zephyr.txt
`95  k_sleep                    msec=%u ms | Returns %u`

I'm using bit old Zephyr (3.3.99) but looking at the changes to relevant files there are no meaningful changes (that I can see).

**To Reproduce**
Use SEGGER Systemview, I assume.

**Expected behavior**
Systemview should be able to show name for all events.

**Impact**
Makes reading tracing output harder.

**Environment:**
 - OS: Ubuntu 24.04
 - Toolchain: Zephyr SDK
 - Version: 3.3.99

nashif, 2025-02-11:
curious about the platform you are using

nashif, 2025-02-23:
just tried, and I see no issues, i.e.

[image]

Did you provide a custom description file that matches your code base? the one in the segger application is outdated or wrong.

HyperionCoding, 2025-02-24:
> just tried, and I see no issues, i.e.
> 
> [image]
> 
> Did you provide a custom description file that matches your code base? the one in the segger application is outdated or wrong.

I did copy the description file.

Did you test on 3.3.99 or the latest? If on the latest, I'll have to try that too and see if it changes.

nashif, 2025-03-30:
sorry, can't reproduce

closed 2025-03-30 by nashif