zephyrproject-rtos/zephyr #85662: `tests/misc/llext-edk/misc.edk.pytest` build failure when running under load

ithinuel, 2025-02-12:
**Describe the bug**

While tests are batched run and the system's under heavy load, `tests/misc/llext-edk/misc.edk.pytest` fails with the following error:

```
-------------------------------- live log call ---------------------------------
INFO: -- west build: running target llext-edk
[1/1] Generating llext-edk.tar.xz
-- Found Python3: /usr/bin/python (found suitable version "3.10.12", minimum required is "3.10") found components: Interpreter
-- Cache files will be written to: /root/.cache/zephyr
DEBUG: Copying llext-edk.tar.xz to /tmp/tmps7cievp7
DEBUG: Extracting llext-edk.tar.xz to /tmp/tmps7cievp7
INFO:
DEBUG: Copying extension to /tmp/tmprpbpnrf2
DEBUG: Copying file2hex.py to /tmp/tmprpbpnrf2
DEBUG: Building extension in /tmp/tmprpbpnrf2 - cmake
INFO: -- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmprpbpnrf2/build
DEBUG: Building extension in /tmp/tmprpbpnrf2 - make
make: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
/tmp/tmprpbpnrf2/src/main.c:7:10: fatal error: zephyr/kernel.h: No such file or directory
7 | #include <zephyr/kernel.h>
|          ^~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/extension.dir/build.make:73: extension.llext] Error 1
make[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/extension.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
FAILED
```

but if we run this test alone (and on a fresh out directory)
`west twister -p qemu_cortex_r5/zynqmp_rpu -s tests/misc/llext-edk/misc.edk.pytest` it passes.

If the test has failed, trying to re-run whist preserving the output directory will keep on failing. The output directory (for that specific test is enough) needs to be removed in order to get the tests to build from fresh & pass.

 - What target platform are you using?
   Was noticed with `qemu_cortex_r5/zynqmp_rpu`. I don't know if there are other platforms running this specific test.
 - What have you tried to diagnose or workaround this issue?
   Clearing the tests output directory and running this test alone. It passed.
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
   It might be but considering how long it takes to run the suites and trigger the issue, a bisect hasn't be an option.

**To Reproduce**

Run `west twister -p qemu_cortex_r5/zynqmp_rpu`

**Expected behavior**

The test should build and pass on the first run.

**Impact**

It is pretty significant as it prevents local & CI run to pass.

**Logs and console output**

**Environment (please complete the following information):**

 - OS: Linux (the CI docker container 0.27.4)
 - Commit SHA: caa284e0355

pillo79, 2025-02-27:
I have been trying to reproduce the bug on my PC, and I wasn't able to replicate this exact issue. When running the full Twister suite or artificially using the CPU, I can at most get a `Pytest timeout` (because the test took more than 60s). 

Your error sounds like some files were not copied correctly, but I don't know if it's at EDK build or use time. If that happens again for you, would you mind saving the `zephyr/llext-edk.tar.xz` archive from the failing test and comparing it with a clean-run result? Their contents should match exactly. Thanks!

wearyzen, 2025-05-08:
I had split the twister tests to run this test separately only after all the other tests were done and haven't seen the issue in a while.
I'll remove the special treatment and see if the issue is back. I'll update in a few days with my observation.

wearyzen, 2025-06-05:
Just an update, tests/misc/llext-edk/misc.edk.pytest for platform qemu_cortex_r5/zynqmp_rpu still fails frequently but the failure is due to timeout now and not because of initially reported error.

PR #91176 (2025-06-11): tests: llext-edk: fix test timeout on some workloads
`tests/misc/llext-edk/misc.edk.pytest` is timing out on some workloads. Double the timeout to 120 seconds to accommodate longer execution times.

Fixes #85662.