zephyrproject-rtos/zephyr #85682: MCXA156 FLEX IO SPI crash

jerpa77, 2025-02-12:
**Bug description**
I'm trying to use the FLEXIO SPI peripheral of the MCXA156. As a first step, I tried to run the spi_loopback test.
I created a dt overlay file and it compiled correctly, but the execution runs into a "bus fault error".

I did verify the SPI bus signals with my logic analyzer and they look correct.

Board: FRDM-MCXA156

**To Reproduce**
Add the attached dt overlay file and run the test.

**Expected behavior**
-  Bus fault error displayed in the shell

**Logs and console output**

[image]
[frdm_mcxa156.zip](https://github.com/user-attachments/files/18770172/frdm_mcxa156.zip)

**Environment (please complete the following information):**

 - OS: Windows 11
 - zephyr-sdk-0.17.0

mmahadevan108, 2025-02-25:
@jerpa77 , can you try this PR https://github.com/zephyrproject-rtos/zephyr/pull/86166

mmahadevan108, 2025-02-28:
@jerpa77 , I have tested with the latest main and this test now passes on MCXA156 frdm board. 
I had to add `input-enable;` to the overlay you had attached to the `pinctrl` settings

closed 2025-02-28 by mmahadevan108