zephyrproject-rtos/zephyr #85683: kernel: k_work_submit_to_queue() doesn't reschedule when in ISR context on single core

eriktamlin, 2025-02-12:
**Describe the bug**
I'm not entirely sure here, but it appears that when calling on k_work_submit_to_queue() from an ISR it doesn't reschedule to do that work when running on a single core. It will just go to idle as if there is nothing to do.
It only does the work once something else finally calls for a reschedule.

Tried to look through the code quickly and describe what I think happens below under Additional Context.

**To Reproduce**
Call k_work_submit_to_queue() from ISR. It won't run after ISR ends.

**Expected behavior**
I guess it should do the work immediately, no?

**Impact**
Severe?

**Logs and console output**
wq_stepper is flagged as "Ready" but doesn't start executing until something else happens in the system that causes a proper reschedule.
[image]

**Environment (please complete the following information):**
-

**Additional context**
k_work_submit_to_queue() calls on z_reschedule_unlocked() which in turns calls z_reschedule_irqlock().
In z_reschedule_irqlock() it checks the boolean resched() which will always return 0 when in ISR context (return arch_irq_unlocked(key) && !arch_is_in_isr()) meaning it will never call on z_swap_irqlock() but instead go to the else which does nothing on a single core system.

Is it supposed to do it like this? Or did I get completely lost in the code when I poked around in there?

badevos, 2025-02-20:
I have exactly this issue. I already tried to use `k_reschedule()` in my ISR, yet it has no effect.
What are we missing here?

badevos, 2025-02-20:
Background hw information: this is observed on an NXP RW612, which is running a Cortex-M 33 core.

After reviewing this issue from the ISR point of view I came to this part of the code:
https://elixir.bootlin.com/zephyr/v4.1.0-rc1/source/include/zephyr/arch/arm/irq.h#L184
```
#define ARCH_ISR_DIRECT_DECLARE(name) \
	static inline int name##_body(void); \
	ARCH_ISR_DIAG_OFF \
	__attribute__ ((interrupt ("IRQ"))) void name(void) \
	{ \
		int check_reschedule; \
		ISR_DIRECT_HEADER(); \
		check_reschedule = name##_body(); \
		ISR_DIRECT_FOOTER(check_reschedule); \
	} \
	ARCH_ISR_DIAG_ON \
	static inline int name##_body(void)
```
According to some code-comments I realize that  `ISR_DIRECT_FOOTER()` should be given a non-zero value as argument in order to force a context switch after the return of the interrupt. However it's value is determined by `name##_body()` and I didn't understand that part of the code.

Should we declare our interrupts in a specific way?

This is how we use it:
```
static void Application_IRQHandler(void)
{
    APP_TRACE_BEG(isr_trace_line_sm);
  
    GPIO_PinClearInterruptFlag(GPIO, 0, GPIO_LINE_SM, 0);

    int ret = k_work_submit_to_queue(&k_stepper_work_q, &stepper_work);
    if (ret < 0)
    {
        LOG_ERR("M: k_work_submit_to_queue failed with error: %d", ret);
    }

    APP_TRACE_END(0);
}

void app_irq_init(void)
{
    IRQ_DIRECT_CONNECT(GPIO_INTA_IRQn, 2, Application_IRQHandler, 0);
    irq_enable(GPIO_INTA_IRQn);
}
```

mathieuchopstm, 2025-02-21:
> Should we declare our interrupts in a specific way?

Yes. The correct usage of direct ISR implies usage of `ISR_DIRECT_DECLARE` as documented here:
https://github.com/zephyrproject-rtos/zephyr/blob/a94acf3db236e4aaab7dd8b8a09dff59117d2810/include/zephyr/irq.h#L98-L137

Notably:
```
 * These ISRs are designed for performance-critical interrupt handling and do
 * not go through common interrupt handling code. They must be implemented in
 * such a way that it is safe to put them directly in the vector table.  For
 * ISRs written in C, The ISR_DIRECT_DECLARE() macro will do this
 * automatically. For ISRs written in assembly it is entirely up to the
 * developer to ensure that the right steps are taken.
[...]
 * - Scheduling decisions are now optional, controlled by the return value of
 *   ISRs implemented with the ISR_DIRECT_DECLARE() macro
```

Try the following instead:
```diff
- static void Application_IRQHandler(void)
+ ISR_DIRECT_DECLARE(Application_IRQHandler)
{
    APP_TRACE_BEG(isr_trace_line_sm);
  
     GPIO_PinClearInterruptFlag(GPIO, 0, GPIO_LINE_SM, 0);

    int ret = k_work_submit_to_queue(&k_stepper_work_q, &stepper_work);
    if (ret < 0)
    {
        LOG_ERR("M: k_work_submit_to_queue failed with error: %d", ret);
    }

    APP_TRACE_END(0);
+    return 1; /* always check for reschedule */
}
```

peter-mitsis, 2025-02-24:
@badevos - Did @mathieuchopstm's suggestion work for you?

@eriktamlin -  Your situation definitely sounds odd and I would like to dig into this a little more. Can you provide a little extra info such as Zephyr version and architecture/board? Are you using direct interrupts as @badevos was?

PR #86591 (2025-03-05): doc: Update direct ISR documentation
Updates the direct ISR documentation to make it more clear that direct ISR handlers should be declared using ISR_DIRECT_DECLARE().

Fixes #85683
commit c9a96477 (2025-03-05): doc: Update direct ISR documentation

Updates the direct ISR documentation to make it more clear that
direct ISR handlers should be declared using ISR_DIRECT_DECLARE().

Fixes #85683
closed 2025-03-05 by fabiobaltieri