zephyrproject-rtos/zephyr #85710: Collaborator nomination: Make Shi

jhedberg, 2025-02-13:
# Nomination

## GitHub User

Provide the following information about the GitHub user:

1. Full Name: Make Shi
1. GitHub username: @makeshi 
1. Organization: NXP

## Supporting Documents

https://github.com/zephyrproject-rtos/zephyr/pull/85582
https://github.com/zephyrproject-rtos/zephyr/pull/83256
https://github.com/zephyrproject-rtos/zephyr/pull/83250
https://github.com/zephyrproject-rtos/zephyr/pull/82895
https://github.com/zephyrproject-rtos/zephyr/pull/82894
https://github.com/zephyrproject-rtos/zephyr/pull/82833

[TSC Project Roles]: <https://docs.zephyrproject.org/latest/project/project_roles.html>
[GitHub Permission Level]: <https://docs.github.com/en/organizations/managing-access-to-your-organizations-repositories/repository-roles-for-an-organization>

RobinKastberg, 2025-02-13:
Duplicate of https://github.com/zephyrproject-rtos/zephyr/issues/85638

makeshi, 2025-02-13:
have been closed https://github.com/zephyrproject-rtos/zephyr/issues/85638

jhedberg, 2025-02-13:
> Duplicate of [#85638](https://github.com/zephyrproject-rtos/zephyr/issues/85638)

Thanks for pointing that out! I'll close this then.

jhedberg, 2025-02-13:
> have been closed [#85638](https://github.com/zephyrproject-rtos/zephyr/issues/85638)

That's excellent, thanks @makeshi. For some reason GitHub wasn't auto-completing your username in some cases, which made me think that you're not yet a member.

makeshi, 2025-02-13:
> That's excellent, thanks [@makeshi](https://github.com/makeshi). For some reason GitHub wasn't auto-completing your username in some cases, which made me think that you're not yet a member.

I close https://github.com/zephyrproject-rtos/zephyr/issues/85638 by myself because [RobinKastberg](https://github.com/RobinKastberg) comments the ticket is duplicated with https://github.com/zephyrproject-rtos/zephyr/issues/85710, now two tickets both are closed, so need reopen current ticket https://github.com/zephyrproject-rtos/zephyr/issues/85710?

jhedberg, 2025-02-13:
> I close [#85638](https://github.com/zephyrproject-rtos/zephyr/issues/85638) by myself because [RobinKastberg](https://github.com/RobinKastberg) comments the ticket is duplicated with [...]

I reopened this

closed 2025-02-20 by nashif