zephyrproject-rtos/zephyr #85718: Using hashed logs didn't reduce code size

hilamal, 2025-02-13:
**Describe the bug**
I'm trying to reduce my code size, one approach was to reduce it by changing the logging from using strings to binary hashing instead.
I expected the code size to reduce, since we don't need to store the full strings in the code RAM. 
I set the configs to get hash logs:
CONFIG_LOG_PRINTK=y
CONFIG_LOG_BACKEND_UART_OUTPUT_DICTIONARY_HEX=y or CONFIG_LOG_BACKEND_UART_OUTPUT_DICTIONARY_BIN
CONFIG_LOG_BACKEND_UART_OUTPUT_DICTIONARY=y
CONFIG_LOG_BACKEND_UART=y
After changing it I saw no difference in the code size.

**To Reproduce**
1. Build a version with logs - strings. Keep the code size.
2. Set those configs to get hash logs:
CONFIG_LOG_PRINTK=y
CONFIG_LOG_BACKEND_UART_OUTPUT_DICTIONARY_HEX=y or CONFIG_LOG_BACKEND_UART_OUTPUT_DICTIONARY_BIN
CONFIG_LOG_BACKEND_UART_OUTPUT_DICTIONARY=y
CONFIG_LOG_BACKEND_UART=y
3. Build the hashed logs version and get the code size.

**Expected behavior**
I'd expect the code size to reduce since we don't need to store the strings (we need only the hash codes for each string)

**Impact**
When we have code size issues, the logs take a lot of space.

**Logs and console output**
None

**Environment (please complete the following information):**

 - OS: Linux, Windows
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: v3.5.0

**Additional context**
None

nordic-krch, 2025-02-17:
Take a look at `samples/subsys/logging/dictionary/` this sample demonstrates logging with strings stripping. After compilation for `qemu_cortex_m3` you should see something like:
```
           FLASH:       15900 B       256 KB      6.07%
             RAM:        6216 B        64 KB      9.48%
     DEVNULL_ROM:        5647 B       256 KB      2.15%
        IDT_LIST:          0 GB        32 KB      0.00%
```
Where presence of `DEVNULL_ROM` section indicates that stripping was enabled. Note that it is supported only for cortex_m and riscv at the moment.

hilamal, 2025-02-17:
We use cortex_r, if we'll create a new section DEVNULL_ROM, can we use the string stripping feature (to move all the strings into this section)? Also, can we delete this section and run our FW without it? does the FW needs this DEVNULL_ROM section in runtime?

nordic-krch, 2025-02-17:
`DEVNULL_ROM` section is only in elf but it is stripped from binary (section has `COPY` type).

hilamal, 2025-02-18:
Thanks :)
Do you know if n when there's a roadmap plan to add the string stripping to Cortex-R?