zephyrproject-rtos/zephyr #85756: Bluetooth: Audio: AICS Registration Fails for Ambient Input Type

babrsn, 2025-02-13:
**Describe the bug**
When attempting to register the AICS with the input type set to `BT_AICS_INPUT_TYPE_AMBIENT`, the registration process fails.

This issue arises from a validation check in the `subsys/bluetooth/audio/aics.c` file at line `490` (https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/bluetooth/audio/aics.c#L490):

```
CHECKIF(param->type > BT_AICS_INPUT_TYPE_STREAMING) {
    LOG_DBG("Invalid AICS input type value: %u", param->type);
    return -EINVAL;
}
```
The condition incorrectly restricts valid input types to those less than or equal to `BT_AICS_INPUT_TYPE_STREAMING` (`0x06`), causing the `BT_AICS_INPUT_TYPE_AMBIENT` (`0x07`) type to be rejected.

To resolve this issue, the validation check should be updated by replacing `BT_AICS_INPUT_TYPE_STREAMING` with `BT_AICS_INPUT_TYPE_AMBIENT` to ensure that `BT_AICS_INPUT_TYPE_AMBIENT` is recognized as a valid input type during AICS registration. 

**To Reproduce**
N/A

**Expected behavior**
The AICS registration should accept `BT_AICS_INPUT_TYPE_AMBIENT` as a valid input type and complete successfully.

**Impact**
This issue prevents applications from utilizing the 'Ambient' input type for AICS, limiting functionality for devices that rely on ambient audio input.

**Logs and console output**
`[19:22:57.053,741] <dbg> bt_aics: bt_aics_register: Invalid AICS input type value: 7`

**Environment (please complete the following information):**
 - Commit SHA or Version used: `44c7a14`

**Additional context**
N/A

Thalley, 2025-02-14:
Seems like a simple fix. Will post a solution ASAP :) Thanks for reporting it

Thalley, 2025-02-14:
I've posted 2 solutions for this: One minimal and one proper. Given that the proper fix requires a minor API change, I don't think we can expect that to be merged for 4.1, so I've posted a minimal fix as well that can.

closed 2025-02-15 by kartben