zephyrproject-rtos/zephyr #85763: IPM ESP32 example does not launch on ESP32-S3

LTVA1, 2025-02-14:
**Describe the bug**
`samples/drivers/ipm/ipm_esp32/` example does not launch on ESP32-S3-DevKitC target, fails at `waiting for download`.

**To Reproduce**
1. Configure the environment as in [getting started guide](https://docs.zephyrproject.org/latest/develop/getting_started/index.html).
2. Try to compile and flash the AMP example as stated at the bottom of [this guide](https://developer.espressif.com/blog/esp32-bootstrapping/):
````
west build -b esp32s3_devkitm/esp32s3/procpu samples/drivers/ipm/ipm_esp32/ -p --sysbuild
west flash && west espressif monitor
````

**Expected behavior**
I expected the program to work as stated in the [guide](https://developer.espressif.com/blog/esp32-bootstrapping/) and send debug messages

**Impact**
I can't work with the Zephyr toolchain because it does not compile the example AMP project, and I want to work with ESP32-S3 in AMP mode.

**Logs and console output**
Console output: [log_amp_boot.txt](https://github.com/user-attachments/files/18795642/log_amp_boot.txt)

**Environment (please complete the following information):**

 - OS: Windows 10
 - Toolchain: Zephyr SDK
 - West version: v1.3.0
 - Zephyr branch revision: `v4.0.0-4696-g96a235499bc`

**Additional context**

In the log you can see that I first tried to press `RET` button several times. And I tried to flash two times. Third time I flashed after powering up the dev board with `BOOT` button pressed.

sylvioalves, 2025-02-14:
It does not seem Zephyr or sample code issue but HW/USB problem.

`ClearCommError failed (PermissionError(13..`.
After flashing, can you run `west espressif monitor` alone OR try using other serial monitor tool to check?

sylvioalves, 2025-02-14:
Please, you you pull latest `main`, run ` west update`and retry?

LTVA1, 2025-02-14:
As you can see, now it's always stuck at `waiting for download` and doesn't go further while it eventually did go further before

LTVA1, 2025-02-14:
Hm, after reopening console as admin and reentering venv it now goes to the same end point with `Application start=403a6a9ch`

[log_console_admin.txt](https://github.com/user-attachments/files/18799178/log_console_admin.txt)

sylvioalves, 2025-02-14:
I have the sample jump address but it continues to work. @marekmatej, any thoughts?

```c
I (189) boot: libc heap size 154 kB.
I (224) amp: APPCPU image, area id: 3, offset: 0x2c0000, hdr.off: 0x20, size: 448 kB
I (224) amp: IRAM segment: paddr=002c0080h, vaddr=403a6400h, size=035f0h ( 13808) load
I (230) amp: DRAM segment: paddr=002c3660h, vaddr=3fcb99f0h, size=0023ch (   572) load
I (236) amp: IROM segment: paddr=002d0000h, vaddr=43e00000h, size=022d4h (  8916) map
I (243) amp: DROM segment: paddr=002e0000h, vaddr=3de10000h, size=00850h (  2128) map
I (251) amp: Application start=403a6a9ch
*** Booting Zephyr OS build v4.0.0-5389-g5155cb673f02 ***
PRO_CPU is sending a request, waiting remote response...
PRO_CPU received a message from APP_CPU : APP_CPU uptime ticks 502

PRO_CPU is sending a request, waiting remote response...
PRO_CPU received a message from APP_CPU : APP_CPU uptime ticks 10505

PRO_CPU is sending a request, waiting remote response...
PRO_CPU received a message from APP_CPU : APP_CPU uptime ticks 20506
```

marekmatej, 2025-02-14:
hi @LTVA1 your issues seems to be related to `west espressif monitor` on the windows. In your first `log_amp_boot.txt` we can see all 3 images being build and flashed. Troubles begins at line `269` where connection to the serial console is about to start. To isolate this problem can you try flash the target `west flash` and just after then connect to the target serial port using terminal emulator?

LTVA1, 2025-02-14:
Terminal emulator? HTerm refused to work bc comport blocked by another app even though I ended the monitoring process in the console

LTVA1, 2025-02-14:
@marekmatej also see the latest log, it stops at `Application start=403a6a9ch`

LTVA1, 2025-02-14:
@marekmatej I tried connecting it via other type c port (which is usb-uart, the right one while usb-jtag is left one...) and now it prints all the debug! Why?

[log_another_usb_port.txt](https://github.com/user-attachments/files/18803316/log_another_usb_port.txt)

marekmatej, 2025-02-14:
> [@marekmatej](https://github.com/marekmatej) I tried connecting it via other type c port (which is usb-uart, the right one while usb-jtag is left one...) and now it prints all the debug! Why?
> 
> [log_another_usb_port.txt](https://github.com/user-attachments/files/18803316/log_another_usb_port.txt)

When flashing the target using the underlying esptool.py, it will find any USB port that is available - that could be USB/Jtag port or USB/Serial port. However, when connecting to the running target you need to connect to whatever uart/usb the application is using as a console. In this case console serial is set in `dts` under the `chosen` node as `zephyr,console = &uart0;` which is connected to ttl/usb converter available at the given USBC port of your board.

LTVA1, 2025-02-14:
Hm, in usual Espressif IDE examples debug & console are all through the USB/JTAG, is there some setting to make it work the same way? I don't want to connect two cables and configure serial port monitor separately...

closed 2025-02-14 by marekmatej