zephyrproject-rtos/zephyr #85766: CoAP Client (coap_client) poll error when closing socket.

povsel, 2025-02-14:
**Describe the bug**
In short -> Problem arises when I use CoAP Client (coap_client). Precisely when I close socket after finishing data transfer. I get socket poll error.

In full -> I use coap client via UDP. I open a socket, send some data, wait for response and then close the socket. But I get poll error "Error in poll: POLLNVAL - fd %d not open". I believe the cause for the error is `has_ongoing_exchange->exchange_lifetime_exceeded` indicating that the exchange is still ongoing and therefore it tries to poll socket that is already closed (note that it is different from request_ongoing parameter, which does indicate that request is finished). 

Seems like these changes were made 5 months ago. Commit explains quite extensively why these changes were made (18f355666bcd42f5afdfd8887a5b3e7ff388bcb3). 

**To Reproduce**
- Open a socket (I've used UDP socket);
- Create zephyr coap_client, send some requests;
- Wait for responses;
- Close socket;
- _get error_

**Expected behavior**
I am able to close socket when I finish communication.
Either I am able to configure / turn this ongoing_exchange off, if it is not really needed. Or I am able to get some kind of indication when I can close the socket safely without errors.

**Impact**
Showstopper ?

**Logs and console output**
[image]

**Environment:**
- zephyr (VERSION_MAJOR = 3, VERSION_MINOR = 7, PATCHLEVEL = 99);
- nRF9160 with ncs-v2.9.0;

**Additional context**
Seems like exchange lifetime is dependent on two settings:
COAP_EXCHANGE_LIFETIME_FACTOR - which is hardcoded in coap_client.c - not configurable;
ack_timeout / CONFIG_COAP_INIT_ACK_TIMEOUT_MS - which you can configure, but from my understanding it was originally used for the retransmission timeout, and later also reused for the exchange lifetime, so settings this to 0 does not seem like an option.

rlubos, 2025-02-14:
Can you try to cancel all active requests before closing the socket (`coap_client_cancel_requests()`)?
https://github.com/zephyrproject-rtos/zephyr/blob/ee37a94aa81f0f98879f615dd31c8cab4d322240/include/zephyr/net/coap_client.h#L163

povsel, 2025-02-14:
Forgot to mention, yes I do call `coap_client_cancel_requests`, in case there is something. 
It can also be seen in the screenshot that I have attached. handle_poll is polling socket, then cancel_requests_with_... is called. After `coap_client_cancel_requests` I call `close()` (not seen in the screenshot), but you can see that polling is still going and eventually I get the error.

boaks, 2025-02-17:
Just for curiosity: Why do you want to close the UDP socket? (I never do so.)

povsel, 2025-02-17:
> Just for curiosity: Why do you want to close the UDP socket? (I never do so.)

Just trying to fulfill the spec I got :D

closed 2025-02-18 by kartben

SeppoTakalo, 2025-02-26:
> Just for curiosity: Why do you want to close the UDP socket? (I never do so.)

Even though this is old topic:
There might be HW or FW limitations that you can keep only certain amount of sockets open.
For example, nRF91 modem might only have enough memory to hold 3 TLS/DTLS sessions open. So depending on the architecture, you might be forced to close the socket for idle period.

Example architecture could be: your main application running whatever protocol MQTT/CoAP/LwM2M and uses 1 socket for it. Then if there is FOTA, it might require 1 socket for downloading using HTTP. AGPS/PGPS services might use another for downloading the satellite data. Then if there is some background service that checks modem FOTA or provisioning data once in two weeks, it might be the one that should close unused sockets.

So usually number of sockets is not the limiting factor but number of active TLS sessions might easily be.

PR #85778 (2025-02-18): net: coap_client: Stop all socket activities when cancelling requests
Calling `coap_client_cancel_requests()` clears the internal request context only for active requests (i. e. not replied yet). However,
if there are any pending request context monitoring ACK duplicates, those would still make the corresponding client socket being monitored by poll(). In result, when application closes the socket, the polling thread will throw POLLNVAL error for the socket.

Fix this, by resetting all request contexts unconditionally. The request callback will only be called for the active requests.

Note for [... cut]