zephyrproject-rtos/zephyr #85774: Windows compilation error

tejlmand, 2025-02-14:
### Discussed in https://github.com/zephyrproject-rtos/zephyr/discussions/85747

<div type='discussions-op-text'>

<sup>Originally posted by **ofirshe** February 13, 2025</sup>
Hi, I've rebased onto the latest Zephyr commit (8b80c209446d8bbe9b454b91773ace14ac40d4c2) and I'm encountering compilation errors related to YAML and JSON escaping that I've never seen before.

Can someone help me debug this?
```
C:\Users\ofirs\bsp>west build -b mimxrt1060_evk
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: C:/Users/ofirs/bsp
-- CMake version: 3.29.1
-- Found Python3: C:/Python312/python.exe (found suitable version "3.12.3", minimum required is "3.10") found components: Interpreter
-- Cache files will be written to: C:/Users/ofirs/bsp/zephyr/.cache
-- Zephyr version: 4.0.99 (C:/Users/ofirs/bsp/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
CMake Warning at zephyr/cmake/modules/boards.cmake:110 (message):
  Deprecated BOARD=mimxrt1060_evk specified, board automatically changed to:
  mimxrt1060_evk/mimxrt1062/qspi.
Call Stack (most recent call first):
  zephyr/cmake/modules/zephyr_default.cmake:133 (include)
  zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:37 (find_package)

-- Board: mimxrt1060_evk, Revision: C, qualifiers: mimxrt1062/qspi
-- Found host-tools: zephyr 0.16.8 (C:/Users/ofirs/bsp/zephyr-sdk-0.16.8)
-- Found toolchain: zephyr 0.16.8 (C:/Users/ofirs/bsp/zephyr-sdk-0.16.8)
-- Found Dtc: C:/ProgramData/chocolatey/bin/dtc.exe (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: C:/Users/ofirs/bsp/zephyr/boards/nxp/mimxrt1060_evk/mimxrt1060_evk_mimxrt1062_qspi.dts
-- Found devicetree overlay: C:/Users/ofirs/bsp/zephyr/boards/nxp/mimxrt1060_evk/mimxrt1060_evk_mimxrt1062_qspi_C.overlay
-- Generated zephyr.dts: C:/Users/ofirs/bsp/build/zephyr/zephyr.dts
-- Generated pickled edt: C:/Users/ofirs/bsp/build/zephyr/edt.pickle
-- Generated devicetree_generated.h: C:/Users/ofirs/bsp/build/zephyr/include/generated/zephyr/devicetree_generated.h

-- Including generated dts.cmake file: C:/Users/ofirs/bsp/build/zephyr/dts.cmake
Parsing C:/Users/ofirs/bsp/zephyr/Kconfig
Loaded configuration 'C:/Users/ofirs/bsp/zephyr/boards/nxp/mimxrt1060_evk/mimxrt1060_evk_mimxrt1062_qspi_defconfig'
Merged configuration 'C:/Users/ofirs/bsp/project_configuration/prj.conf'
Merged configuration 'C:/Users/ofirs/bsp/project_configuration/ethernet-phy-overlay.conf'
Configuration saved to 'C:/Users/ofirs/bsp/build/zephyr/.config'
Kconfig header saved to 'C:/Users/ofirs/bsp/build/zephyr/include/generated/zephyr/autoconf.h'
-- Found GnuLd: c:/users/ofirs/bsp/zephyr-sdk-0.16.8/arm-zephyr-eabi/arm-zephyr-eabi/bin/ld.bfd.exe (found version "2.38")
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/ofirs/bsp/zephyr-sdk-0.16.8/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc.exe
Load components for MIMXRT1062:
driver_common component is included.
driver_reset component is included.
device_CMSIS component is included.
CMSIS_Include_core_cm component is included.
device_system component is included.
driver_trng component is included.
driver_enet component is included.
driver_lpi2c component is included.
driver_lpspi component is included.
driver_lpuart component is included.
driver_igpio component is included.
driver_flexspi component is included.
driver_cache_armv7_m7 component is included.
driver_ocotp component is included.
-- Configuring done (11.4s)
-- Generating done (0.7s)
-- Build files have been written to: C:/Users/ofirs/bsp/build
←[92m-- west build: building application
[1/293] Generating include/generated/zephyr/version.h
-- Zephyr version: 4.0.99 (C:/Users/ofirs/bsp/zephyr), build: v4.0.0-5273-gaea9db6e1d08
[2/293] Generating ../build_info.yml
FAILED: build_info.yml C:/Users/ofirs/bsp/build/build_info.yml
C:\WINDOWS\system32\cmd.exe /C "cd /D C:\Users\ofirs\bsp\build\zephyr && "C:\Program Files\CMake\bin\cmake.exe" -DJSON_FILE="build_info_1.json" -DYAML_FILE="C:/Users/ofirs/bsp/build/build_info.yml" -DTEMP_FILES="build_info_1.json" -P C:/Users/ofirs/bsp/zephyr/cmake/yaml-filter.cmake"
-- Found Python3: C:/Python312/python.exe (found suitable version "3.12.3", minimum required is "3.10") found components: Interpreter
-- Cache files will be written to: C:/Users/ofirs/bsp/zephyr/cmake/..//.cache
CMake Error at C:/Users/ofirs/bsp/zephyr/cmake/modules/yaml.cmake:548 (string):
  string sub-command JSON failed parsing json string: * Line 69, Column 24

    Bad escape sequence in string

  See Line 69, Column 1675 for detail.

  .
Call Stack (most recent call first):
  C:/Users/ofirs/bsp/zephyr/cmake/yaml-filter.cmake:27 (to_yaml)

[3/293] Generating misc/generated/syscalls.json, misc/generated/struct_tags.json
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' --build 'C:\Users\ofirs\bsp\build'
```</div>

tejlmand, 2025-02-14:
@ofirshe thanks for the CMake trace.

Both from the original issue error log and your trace I have traced down the issue to be related to the generator expressions used by the llext subsys here:
https://github.com/zephyrproject-rtos/zephyr/blob/ee37a94aa81f0f98879f615dd31c8cab4d322240/CMakeLists.txt#L2239

The genex expression is stored in a `build_info_1.json` which is processed at build time to create a final `build_info.yaml`.
The content of the `build_info_1.json` is:
```
  "cmake" : 
  {
    "application" : 
    {
      "configuration-dir" : "C:/Users/ofirs/bsp",
      "source-dir" : "C:/Users/ofirs/bsp"
    },
    "board" : 
    {
      "name" : "mimxrt1060_evk",
      "path" : [ "C:/Users/ofirs/bsp/zephyr/boards/nxp/mimxrt1060_evk" ],
      "qualifiers" : "mimxrt1062/qspi",
      "revision" : "B"
    },
    "devicetree" : 
    {
      "bindings-dirs" : 
      [
        "C:/Users/ofirs/bsp/dts/bindings",
        "C:/Users/ofirs/bsp/zephyr/dts/bindings"
      ],
      "files" : 
      [
        "C:/Users/ofirs/bsp/zephyr/boards/nxp/mimxrt1060_evk/mimxrt1060_evk_mimxrt1062_qspi.dts",
        "C:/Users/ofirs/bsp/zephyr/boards/nxp/mimxrt1060_evk/mimxrt1060_evk_mimxrt1062_qspi_B.overlay"
      ],
      "include-dirs" : 
      [
        "C:/Users/ofirs/bsp/modules/hal/atmel/include",
        "C:/Users/ofirs/bsp/modules/hal/gigadevice/include",
        "C:/Users/ofirs/bsp/modules/hal/microchip/include",
        "C:/Users/ofirs/bsp/modules/hal/microchip/dts",
        "C:/Users/ofirs/bsp/modules/hal/nuvoton/dts",
        "C:/Users/ofirs/bsp/modules/hal/nxp/dts",
        "C:/Users/ofirs/bsp/modules/hal/stm32/dts",
        "C:/Users/ofirs/bsp/include",
        "C:/Users/ofirs/bsp/dts",
        "C:/Users/ofirs/bsp/zephyr/include",
        "C:/Users/ofirs/bsp/zephyr/include/zephyr",
        "C:/Users/ofirs/bsp/zephyr/dts/common",
        "C:/Users/ofirs/bsp/zephyr/dts/x86",
        "C:/Users/ofirs/bsp/zephyr/dts/xtensa",
        "C:/Users/ofirs/bsp/zephyr/dts/sparc",
        "C:/Users/ofirs/bsp/zephyr/dts/riscv",
        "C:/Users/ofirs/bsp/zephyr/dts/posix",
        "C:/Users/ofirs/bsp/zephyr/dts/nios2",
        "C:/Users/ofirs/bsp/zephyr/dts/arm64",
        "C:/Users/ofirs/bsp/zephyr/dts/arm",
        "C:/Users/ofirs/bsp/zephyr/dts/arc",
        "C:/Users/ofirs/bsp/zephyr/dts"
      ]
    },
    "kconfig" : 
    {
      "extra-user-files" : [ "project_configuration/ethernet-phy-overlay.conf" ],
      "files" : 
      [
        "C:/Users/ofirs/bsp/zephyr/boards/nxp/mimxrt1060_evk/mimxrt1060_evk_mimxrt1062_qspi_defconfig",
        "C:/Users/ofirs/bsp/project_configuration/prj.conf",
        "C:/Users/ofirs/bsp/project_configuration/ethernet-phy-overlay.conf"
      ],
      "user-files" : [ "project_configuration/prj.conf" ]
[... 45 lines cut ...]
 build_info(llext-edk cflags VALUE ${llext_edk_cflags})
 build_info(llext-edk include-dirs VALUE "$<TARGET_PROPERTY:zephyr_interface,INTERFACE_INCLUDE_DIRECTORIES>")
+message("llext-cflags genex      : ${llext_edk_cflags}")
+add_custom_target(print_llext_cflags   COMMAND ${CMAKE_COMMAND} -E echo \"${llext_edk_cflags}\")
+add_custom_target(print_llext_includes COMMAND ${CMAKE_COMMAND} -E echo \"$<TARGET_PROPERTY:zephyr_interface,INTERFACE_INCLUDE_DIRECTORIES>\")
 
 add_custom_command(
     OUTPUT ${llext_edk_file}
```

This patch provides to custom targets which helps us understand exactly how the genex are expanded on your system.
I have unfortunately not been able to reproduce on my Windows machine.

To get the expanded values, then run:
```
$ cd <build-folder>
$ ninja print_llext_cflags
<output>
$ ninja print_llext_includes
<output>
```

and provide the output in this issue.

You can also do it using `west build`, like `west build -t print_llext_cflags` ( and `-t print_llext_includes`)after you have created the build directory.

tejlmand, 2025-02-14:
> Wondering how that happened...

@pillo79 Right, it was not the file written to disk, but the input that was going into `GENERATE`, but it is still the genex'es which seems to expand to some invalid JSON formatting, and thus breaking the build step.
Should have phrased it a bit different 😉

tejlmand, 2025-02-14:
> Note that I already have a related PR https://github.com/zephyrproject-rtos/zephyr/pull/85624 (pending review/approval...)

Which I guess will just downgrade this issue from medium to minor 😛

pillo79, 2025-02-14:
> > Note that I already have a related PR [#85624](https://github.com/zephyrproject-rtos/zephyr/pull/85624) (pending review/approval...)
> 
> Which I guess will just downgrade this issue from medium to minor 😛

Indeed. 😢
I have some ideas on how to fix this, and none looks good enough:
- use the `$<LIST REPLACE` to escape every value in the JSON output. Nice and clean but requires bumping CMake to 3.27 and I have no idea of other consequences.
- enforce the requirement that _all paths must be CMake style_. May be TheRightThing™ but very messy, treewide + modules... unfeasible.
- rewrite the JSON->YAML logic in Python and do some smarts. Works and is small but duplicates code and is hackish.

Do you think any of these is workable? Or maybe there's some magic `set(MAKE_THINGS_WORK 1)` I'm missing...

ofirshe, 2025-02-14:
Hey @tejlmand,

Regarding your [request](https://github.com/zephyrproject-rtos/zephyr/issues/85774#issuecomment-2658812331), could you clarify the exact steps you'd like me to run?

I'm working with Zephyr as a submodule in an out-of-tree setup:
- My root repository is at C:\Users\ofirs\bsp.
- Zephyr is located at C:\Users\ofirs\bsp\zephyr.
- My CMakeLists.txt is in the root project at C:\Users\ofirs\bsp\CMakeLists.txt.

Could you please provide precise instructions? I want to ensure I follow your request correctly.

tejlmand, 2025-02-17:
> Regarding your https://github.com/zephyrproject-rtos/zephyr/issues/85774#issuecomment-2658812331, could you clarify the exact steps you'd like me to run?

Let me try.

[debug_85774.patch](https://github.com/user-attachments/files/18822531/debug_85774.patch)

1. Download the patch file attached with this comment and place it in the Zephyr repo, `C:\Users\ofirs\bsp\zephyr`.
2. Apply the patch by running
    ```
    $ cd C:\Users\ofirs\bsp\zephyr
    $ git apply debug_85774.patch
    ```
3. Run the build as normal, which afaict would be:
    ```
    $ cd C:\Users\ofirs\bsp
    $ west build -b mimxrt1060_evk -- -DE1_STUB=1
    ````
4. Run the specific debug targets:
    ```
    $ west build -t print_llext_cflags
    $ west build -t print_llext_includes
    ```

and provide the output in this issue 🙂

ofirshe, 2025-02-17:
Build output:
```
C:\Users\ofirs\bsp>west build -b mimxrt1060_evkb -- -DE1_STUB=1
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: C:/Users/ofirs/bsp
-- CMake version: 3.29.1
-- Found Python3: C:/Python312/python.exe (found suitable version "3.12.3", minimum required is "3.10") found components: Interpreter
-- Cache files will be written to: C:/Users/ofirs/bsp/zephyr/.cache
-- Zephyr version: 4.0.99 (C:/Users/ofirs/bsp/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
CMake Warning at zephyr/cmake/modules/boards.cmake:110 (message):
  Deprecated BOARD=mimxrt1060_evkb specified, board automatically changed to:
  mimxrt1060_evk@B/mimxrt1062/qspi.
Call Stack (most recent call first):
  zephyr/cmake/modules/zephyr_default.cmake:133 (include)
  zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:37 (find_package)

-- Board: mimxrt1060_evk, Revision: B, qualifiers: mimxrt1062/qspi
-- Found host-tools: zephyr 0.16.8 (C:/Users/ofirs/bsp/zephyr-sdk-0.16.8)
-- Found toolchain: zephyr 0.16.8 (C:/Users/ofirs/bsp/zephyr-sdk-0.16.8)
-- Found Dtc: C:/ProgramData/chocolatey/bin/dtc.exe (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: C:/Users/ofirs/bsp/zephyr/boards/nxp/mimxrt1060_evk/mimxrt1060_evk_mimxrt1062_qspi.dts
-- Found devicetree overlay: C:/Users/ofirs/bsp/zephyr/boards/nxp/mimxrt1060_evk/mimxrt1060_evk_mimxrt1062_qspi_B.overlay
-- Generated zephyr.dts: C:/Users/ofirs/bsp/build/zephyr/zephyr.dts
-- Generated pickled edt: C:/Users/ofirs/bsp/build/zephyr/edt.pickle
-- Generated devicetree_generated.h: C:/Users/ofirs/bsp/build/zephyr/include/generated/zephyr/devicetree_generated.h

-- Including generated dts.cmake file: C:/Users/ofirs/bsp/build/zephyr/dts.cmake
Parsing C:/Users/ofirs/bsp/zephyr/Kconfig
Loaded configuration 'C:/Users/ofirs/bsp/zephyr/boards/nxp/mimxrt1060_evk/mimxrt1060_evk_mimxrt1062_qspi_defconfig'
Merged configuration 'C:/Users/ofirs/bsp/project_configuration/prj.conf'
Merged configuration 'C:/Users/ofirs/bsp/project_configuration/ethernet-phy-overlay.conf'
Configuration saved to 'C:/Users/ofirs/bsp/build/zephyr/.config'
Kconfig header saved to 'C:/Users/ofirs/bsp/build/zephyr/include/generated/zephyr/autoconf.h'
-- Found GnuLd: c:/users/ofirs/bsp/zephyr-sdk-0.16.8/arm-zephyr-eabi/arm-zephyr-eabi/bin/ld.bfd.exe (found version "2.38")
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/ofirs/bsp/zephyr-sdk-0.16.8/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc.exe
Load components for MIMXRT1062:
driver_common component is included.
driver_reset component is included.
device_CMSIS component is included.
CMSIS_Include_core_cm component is included.
device_system component is included.
driver_trng component is included.
driver_enet component is included.
driver_lpi2c component is included.
driver_lpspi component is included.
driver_lpuart component is included.
driver_igpio component is included.
driver_flexspi component is included.
driver_cache_armv7_m7 component is included.
driver_ocotp component is included.
llext-cflags genex      : -D$<JOIN:KERNEL$<SEMICOLON>__ZEPHYR__=1$<SEMICOLON>$<TARGET_PROPERTY:compiler,security_fortify_compile_time>$<SEMICOLON>__LINUX_ERRNO_EXTENSIONS__$<SEMICOLON>PICOLIBC_DOUBLE_PRINTF_SCANF$<SEMICOLON>XIP_EXTERNAL_FLASH$<SEMICOLON>XIP_BOOT_HEADER_ENABLE=1$<SEMICOLON>BOARD_FLASH_SIZE=CONFIG_FLASH_SIZE*1024$<SEMICOLON>XIP_BOOT_HEADER_DCD_ENABLE=1$<SEMICOLON>__PROGRAM_START$<SEMICOLON>CPU_MIMXRT1062DVL6A$<SEMICOLON>NDEBUG$<SEMICOLON>K_HEAP_MEM_POOL_SIZE=1280,$<SEMICOLON>-D>;-DLL_EXTENSION_BUILD;$<FILTER:$<FILTER:$<JOIN:$<1:$<TARGET_PROPERTY:compiler,no_strict_aliasing>>$<SEMICOLON>-Os$<SEMICOLON>$<TARGET_PROPERTY:compiler,imacros>$<SEMICOLON>C:/Users/ofirs/bsp/build/zephyr/include/generated/zephyr/autoconf.h$<SEMICOLON>$<TARGET_PROPERTY:compiler,no_common>$<SEMICOLON>$<TARGET_PROPERTY:compiler,debug>$<SEMICOLON>$<TARGET_PROPERTY:compiler,diagnostic>$<SEMICOLON>-mcpu=cortex-m7$<SEMICOLON>-mthumb$<SEMICOLON>-mabi=aapcs$<SEMICOLON>-mfp16-format=ieee$<SEMICOLON>-mtp=soft$<SEMICOLON>--sysroot=C:/Users/ofirs/bsp/zephyr-sdk-0.16.8/arm-zephyr-eabi/arm-zephyr-eabi$<SEMICOLON>$<TARGET_PROPERTY:compiler,imacros>$<SEMICOLON>C:/Users/ofirs/bsp/zephyr/include/zephyr/toolchain/zephyr_stdint.h$<SEMICOLON>$<1:$<TARGET_PROPERTY:compiler,warning_base>>$<SEMICOLON>$<TARGET_PROPERTY:compiler,warning_extended>$<SEMICOLON>$<1:$<TARGET_PROPERTY:compiler,warning_error_implicit_int>>$<SEMICOLON>$<1:$<TARGET_PROPERTY:compiler,no_position_independent>>$<SEMICOLON>-fno-asynchronous-unwind-tables$<SEMICOLON>-ftls-model=local-exec$<SEMICOLON>-fno-reorder-functions$<SEMICOLON>--param=min-pagesize=0$<SEMICOLON>-fno-defer-pop$<SEMICOLON>-fmacro-prefix-map=C:/Users/ofirs/bsp=CMAKE_SOURCE_DIR$<SEMICOLON>-fmacro-prefix-map=C:/Users/ofirs/bsp/zephyr=ZEPHYR_BASE$<SEMICOLON>-fmacro-prefix-map=C:/Users/ofirs/bsp=WEST_TOPDIR$<SEMICOLON>-ffunction-sections$<SEMICOLON>-fdata-sections$<SEMICOLON>$<$<BOOL:$<TARGET_PROPERTY:compiler,specs>>:$<TARGET_PROPERTY:compiler,specs>picolibc.specs>$<SEMICOLON>-D_POSIX_THREADS$<SEMICOLON>$<1:$<TARGET_PROPERTY:compiler,cstd>c99>,$<SEMICOLON>>,EXCLUDE,^-fno-pic$|^-fno-pie$|^-ffunction-sections$|^-fdata-sections$|^-Os$>,EXCLUDE,^--sysroot= [... cut ...] /mcux/mcux-sdk/drivers/flexspi;C:/Users/ofirs/bsp/modules/hal/nxp/mcux/mcux-sdk/drivers/cache/armv7-m7;C:/Users/ofirs/bsp/modules/hal/nxp/mcux/mcux-sdk/drivers/ocotp;C:/Users/ofirs/bsp/zephyr/modules/hal_nxp/.;C:\Users\ofirs\bsp/include/drivers/regulator;C:\Users\ofirs\bsp/include/drivers/pmbus;C:\Users\ofirs\bsp/include/utils;C:\Users\ofirs\bsp/include/drivers/e1_spi
```

@tejlmand On the latest Zephyr main branch, this issue is no longer reproducible (I assume recent changes have resolved it).

pillo79, 2025-02-17:
@ofirshe thanks for this output, very helpful!
I see some project-specific include paths (for example `bsp/include/utils` or `bsp/include/drivers/e1_spi`) that have backslashes in their path. Were those added via `zephyr_include_directories`?

> On the latest Zephyr main branch, this issue is no longer reproducible (I assume recent changes have resolved it).

Yes, the PR I mentioned above was just merged and avoids the bug unless you specifically opt in for the EDK functionality. So the bug is still there, but not visible to most users.

ofirshe, 2025-02-17:
@pillo79 I include the entire `include` directory by adding `target_include_directories(app PRIVATE include)` in my `CMakeLists.txt`.

pillo79, 2025-02-17:
Thank you... that effectively rules out the "easy fix" I had prepared (#85884). 😢
@tejlmand Backslashes were added by CMake when converting the relative path into absolute in `target_include_directories`. Do you see a way to prevent this? Should we document that the EDK on Windows needs all project paths to be absolute and `cmake_path`'ed?

tejlmand, 2025-02-18:
> [@pillo79](https://github.com/pillo79) I include the entire `include` directory by adding `target_include_directories(app PRIVATE include)` in my `CMakeLists.txt`.

@ofirshe thanks for the details.
Actually, a `target_include_directories(app PRIVATE include)` will not result in this, because the generator expression is using `INTERFACE_INCLUDE_DIRECTORIES` property from `zephyr_interface`, but what you're using according to the CMake trace is `zephyr_include_directories()` which populates `zephyr_interface` which is seen in those calls:
```
C:/Users/ofirs/bsp/src/drivers/e1_spi/CMakeLists.txt(5):  zephyr_include_directories(C:\Users\ofirs\bsp/include/utils )
C:/Users/ofirs/bsp/src/drivers/e1_spi/CMakeLists.txt(6):  zephyr_include_directories(C:\Users\ofirs\bsp/include/drivers/e1_spi )
```
and it seems that it's your own code which has expanded into `\`'es in the path and not the genex.

I wonder how exactly you're prepending the current path, because `${CMAKE_CURRENT_SOURCE_DIR}` should be using forward slashes.

Can you provide an exact copy of line  5 and 6 from `C:/Users/ofirs/bsp/src/drivers/e1_spi/CMakeLists.txt` ?

ofirshe, 2025-02-18:
Hey @tejlmand,

I've developed several custom drivers that I use as Zephyr modules, and you're absolutely right. Inside those modules, I include header files using `zephyr_include_directories` instead of `target_include_directories`. There wasn't a specific reason for this - it copied from an out-of-tree module example.

tejlmand, 2025-02-18:
> There wasn't a specific reason for this - it copied from an out-of-tree module example.

@ofirshe There's nothing wrong in using `zephyr_include_directories()`, that part of the comment was just to inform that actually `target_include_directories()` was not where to investigate, actually it made me a bit puzzled at first as I couldn't see why this call should break things 😅 

The important part right now is to understand from exactly where the backslahes originates, which is why I would still like you to post the two CMake lines in question.

> Can you provide an exact copy of line 5 and 6 from C:/Users/ofirs/bsp/src/drivers/e1_spi/CMakeLists.txt ?

ofirshe, 2025-02-18:
@tejlmand 
```
zephyr_include_directories($ENV{PROJECT_PATH}/include/utils)
zephyr_include_directories($ENV{PROJECT_PATH}/include/drivers/e1_spi)
```

tejlmand, 2025-02-19:
Thanks @ofirshe .

Please find a fix here: https://github.com/zephyrproject-rtos/zephyr/pull/85994

Still making some extra tests internally to ensure we don't see any extra side-effects.

PR #85994 (2025-02-21): cmake: sanitize json input string
Fixes: #85774 

Sanitize the json input string given to `to_yaml()` functio function. This ensures that a string like `foo\bar` becomes `foo\\bar` during internal CMake json processing and when written to the file it becomes `foo\bar`.
closed 2025-02-21 by fabiobaltieri