zephyrproject-rtos/zephyr #85815: tests: crypto:  mcxw72_evk /  frdm_mcxw72: build failure

hakehuang, 2025-02-15:
**Describe the bug**

the problem you're facing:
 - What target platform are you using?
 - mcxw72_evk and frdm_mcxw72

**To Reproduce**

tests/subsys/jwt/
tests/subsys/secure_storage
tests/subsys/random/rng

```
west twister -p mcxw72_evk/mcxw727c/cpu0 -s tests/drivers/entropy/api/drivers.entropy
```
or with west:
```
west build -p -b mcxw72_evk/mcxw727c/cpu0 tests/drivers/entropy/api -T drivers.entropy
```

**Expected behavior**

Build pass

**Impact**

CI for mcxw72_evk/mcxw727c/cpu0 , frdm_mcxw72/mcxw727c/cpu0

**Logs and console output**

```
zephyr/drivers/entropy/CMakeFiles/drivers__entropy.dir/entropy_nxp_ele.c.obj -MF zephyr/drivers/entropy/CMakeFiles/drivers__entropy.dir/entropy_nxp_ele.c.obj.d -o zephyr/drivers/entropy/CMakeFiles/drivers__entropy.dir/entropy_nxp_ele.c.obj -c /home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/drivers/entropy/entropy_nxp_ele.c\n/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/drivers/entropy/entropy_nxp_ele.c:18:10: fatal error: sss_crypto.h: No such file or directory\n   18 | #include "sss_crypto.h"\n      |          ^~~~~~~~~~~~~~\ncompilation terminated.\nninja: build stopped: subcommand failed.\n', 'TEST_PATH': 'testing.ztest.base.verbose_1'}]]}
```

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: v4.0.0-5520-gc544e4d08b43

fabiobaltieri, 2025-02-18:
cc [andrei-menzopol](https://github.com/andrei-menzopol), this was introduced in https://github.com/zephyrproject-rtos/zephyr/pull/83195

dleach02, 2025-02-20:
reopening as the HAL side PR merging closed this... lets let the Zephyr side do that

closed 2025-02-21 by kartben