zephyrproject-rtos/zephyr #85832: PTP: daytime not set correctly, having constant offset

theadib, 2025-02-16:
**Describe the bug**

The PTP daytime retrieved via ptp_clock_get() has some constant large offset to the real daytime.
The PTP reports correct syncronisation.
my sample output is 2025-02-18 02:59:15 UTC where the real daytime is somewhat 11:50 UTC

Date is correctly set but the time is wrong.
When I doublecheck the given epoch using https://www.epochconverter.com/
I get 1739847592 to Tuesday, February 18, 2025 2:59:52 AM
So the gmtime() acts correctly.

Also has nothing to do with TAI vs UTC (leap seconds)  because the difference here is larger.

**To Reproduce**

I used as base the PTP example given, I added ptp_clock_get() calls every 10sec in the main loop.
I used the STM nucleo_h563zi board, reported as supported board.
I used as PTP source the  NTSPico3 set to 1588, UDP, multicast, E2E, synchronized via GPS.
https://www.elpromaelectronics.com/product/nts-pico3/

CONFIG_PTP_LOG_LEVEL_DBG=y

**Expected behavior**

I expect that the reported daytime of the ptp_clock_get() is exact UTC.

In the given logs (below) the PTP is locked and the board acts as PTP receiver.
The time does not exceed the given jump limit of 1sec, otherwise there would be additional warnings.

**Impact**

**Logs and console output**

Note: the real daytime is 11:50 UTC
```
TIME RECEIVER
### Current PTP time: 1739847587.925317369
### Current date and time: 2025-02-18 02:59:10 UTC
[00:32:49.486,000] <dbg> ptp_msg: ptp_msg_post_recv: Port 1 received Announce message
[00:32:49.491,000] <dbg> ptp_port: port_delay_req_msg_transmit: Port 1 sends Delay_Req message
[00:32:49.491,000] <dbg> ptp_port: port_delay_req_timestamp_cb: Port 1 registered timestamp for 981 Delay_Req
[00:32:49.492,000] <dbg> ptp_msg: ptp_msg_post_recv: Port 1 received Delay_Resp message
[00:32:49.492,000] <dbg> ptp_clock: ptp_clock_delay: Delay 9223372036842497880ns
[00:32:49.565,000] <dbg> ptp_msg: ptp_msg_post_recv: Port 1 received Sync message
[00:32:49.565,000] <dbg> ptp_msg: ptp_msg_post_recv: Port 1 received Follow_Up message
[00:32:49.565,000] <dbg> ptp_clock: ptp_clock_synchronize: Offset -333345867ns
[00:32:50.486,000] <dbg> ptp_msg: ptp_msg_post_recv: Port 1 received Announce message
[00:32:50.565,000] <dbg> ptp_msg: ptp_msg_post_recv: Port 1 received Sync message
[00:32:50.565,000] <dbg> ptp_msg: ptp_msg_post_recv: Port 1 received Follow_Up message
[00:32:50.565,000] <dbg> ptp_clock: ptp_clock_synchronize: Offset -333338605ns
[00:32:51.486,000] <dbg> ptp_msg: ptp_msg_post_recv: Port 1 received Announce message
[00:32:51.491,000] <dbg> ptp_port: port_delay_req_msg_transmit: Port 1 sends Delay_Req message
[00:32:51.491,000] <dbg> ptp_port: port_delay_req_timestamp_cb: Port 1 registered timestamp for 982 Delay_Req
[00:32:51.492,000] <dbg> ptp_msg: ptp_msg_post_recv: Port 1 received Delay_Resp message
[00:32:51.492,000] <dbg> ptp_clock: ptp_clock_delay: Delay 9223372036842466725ns
[00:32:51.565,000] <dbg> ptp_msg: ptp_msg_post_recv: Port 1 received Sync message
[00:32:51.565,000] <dbg> ptp_msg: ptp_msg_post_recv: Port 1 received Follow_Up message
[00:32:51.565,000] <dbg> ptp_clock: ptp_clock_synchronize: Offset -333343991ns
[00:32:52.487,000] <dbg> ptp_msg: ptp_msg_post_recv: Port 1 received Announce message
[00:32:52.565,000] <dbg> ptp_msg: ptp_msg_post_recv: Port 1 received Sync message
[00:32:52.565,000] <dbg> ptp_msg: ptp_msg_post_recv: Port 1 received Follow_Up message
[00:32:52.566,000] <dbg> ptp_clock: ptp_clock_synchronize: Offset -333342366ns
[00:32:53.487,000] <dbg> ptp_msg: ptp_msg_post_recv: Port 1 received Announce message
[00:32:53.491,000] <dbg> ptp_port: port_delay_req_msg_transmit: Port 1 sends Delay_Req message
[00:32:53.491,000] <dbg> ptp_port: port_delay_req_timestamp_cb: Port 1 registered timestamp for 983 Delay_Req
[00:32:53.492,000] <dbg> ptp_msg: ptp_msg_post_recv: Port 1 received Delay_Resp message
[00:32:53.492,000] <dbg> ptp_clock: ptp_clock_delay: Delay 9223372036842422582ns
[00:32:53.565,000] <dbg> ptp_msg: ptp_msg_post_recv: Port 1 received Sync message
[00:32:53.566,000] <dbg> ptp_msg: ptp_msg_post_recv: Port 1 received Follow_Up message
[00:32:53.566,000] <dbg> ptp_clock: ptp_clock_synchronize: Offset -333352664ns
TIME RECEIVER
### Current PTP time: 1739847592.925449522
### Current date and time: 2025-02-18 02:59:15 UTC

```

the modification of the given sample code is minimal:
```
int main(void){
	init_testing();

	while(1) {

		get_current_status();

		const struct device *clk = net_eth_get_ptp_clock(net_if_get_by_index(1));
		struct net_ptp_time tm;
		ptp_clock_get(clk, &tm);
		printk("### Current PTP time: %llu.%09u\n", tm.second, tm.nanosecond);

		struct tm *ptm;
		time_t tai_time = tm.second - 37; // Convert TAI to UTC by subtracting leap seconds
		ptm = gmtime(&tai_time);
		printk("### Current date and time: %04d-%02d-%02d %02d:%02d:%02d UTC\n",
			   ptm->tm_year + 1900, ptm->tm_mon + 1, ptm->tm_mday,
			   ptm->tm_hour, ptm->tm_min, ptm->tm_sec);

		k_sleep(K_SECONDS(5));
	}

	return 0;
}
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 0.17, arm-none-eabi
 - Zephyr 4.1.0-rc1

**Additional context**

kartben, 2025-02-18:
cc @erwango

awojasinski, 2025-02-19:
Hi @theadib thanks for reporting the bug. I don't have any sophisticated PTP source with GPS but I'll test that with my Linux machine that has HW timestamping. Before that can you confirm that your PTP source is reporting correct time? Could you check that using for instance `ptp4l` on your host machine?

theadib, 2025-02-20:
Hi @awojasinski would be good to confirm the changes in my PR :-)
It solves for me the issue with this board.
My PTP box has NTP and PTP server.
so I compared those results until they match.

closed 2025-02-21 by fabiobaltieri