zephyrproject-rtos/zephyr #85841: twister: `WARNING - TS_START: already STARTED` on expected test case reboot

JordanYates, 2025-02-17:
**Is your enhancement proposal related to a problem? Please describe.**

When using twister with testcases that are expected to reboot (i.e. testing fault handling), numerous warning are output to the terminal.
```
WARNING - TS_START: already STARTED 'infuse_reboot':{'count': 1, 'repeat': 0}
WARNING - TC_START: already STARTED case 'lib.reboot.a.infuse_reboot.reboot':{'count': 1}
WARNING - TS_START: already STARTED 'infuse_reboot':{'count': 2, 'repeat': 1}
WARNING - TC_START: already STARTED case 'lib.reboot.a.infuse_reboot.reboot':{'count': 2}
WARNING - TS_START: already STARTED 'infuse_reboot':{'count': 3, 'repeat': 2}
WARNING - TC_START: already STARTED case 'lib.reboot.a.infuse_reboot.reboot':{'count': 3}
WARNING - TS_START: already STARTED 'infuse_reboot':{'count': 1, 'repeat': 0}
WARNING - TC_START: already STARTED case 'lib.reboot.b.infuse_reboot.reboot':{'count': 1}
WARNING - TS_START: already STARTED 'infuse_reboot':{'count': 2, 'repeat': 1}
WARNING - TC_START: already STARTED case 'lib.reboot.b.infuse_reboot.reboot':{'count': 2}
WARNING - TS_START: already STARTED 'infuse_reboot':{'count': 3, 'repeat': 2}
WARNING - TC_START: already STARTED case 'lib.reboot.b.infuse_reboot.reboot':{'count': 3}
```

**Describe the solution you'd like**

A way to specify that the test is expected to reboot in order to supress the warnings

**Additional context**

Messages originally added in #80088.

JordanYates, 2025-02-17:
Initial investigation makes it seem like this will be tricky to implement, since the location of the warning has no context at all about the testcase that the test is a part of.
https://github.com/zephyrproject-rtos/zephyr/blob/8a814dd3552edd968665549184968ccf3be07971/scripts/pylib/twister/twisterlib/harness.py#L839

nashif, 2025-02-18:
@golowanow  can you please take a look?

JordanYates, 2025-02-18:
My initial feeling is that the easiest way to do this would be to define a `ZTEST_REBOOTS` macro to use instead of `ZTEST`.
The macro would save a flag a in the testcase struct, which in turn would be checked at the start of the test to print `START_REBOOT - ` instead of `START -`, so the python side knows that the reboots are normal.
The testsuite could iterate over the testcases it contains in order to discover if any of its constituents expect to reboot to similarly change the starting message.

golowanow, 2025-02-18:
> When using twister with testcases that are expected to reboot (i.e. testing fault handling), numerous warning are output to the terminal.

right, these warnings also happen with this particular kind of test cases and there is at least one more [known already](https://github.com/zephyrproject-rtos/zephyr/pull/80088#issuecomment-2464312404)  `tests/drivers/watchdog/wdt_basic_api`.

> My initial feeling is that the easiest way to do this would be to define a ZTEST_REBOOTS macro to use instead of ZTEST.

to me it also looks reasonable to declare such tests at Ztest and let harness act specifically.

@JordanYates would you like to raise a PR for that already ?  or I can do it later, perhaps this week.

PR #88183 (2025-04-14): scripts: twisterlib: extend testcase schema for reboots
Extend the testcase schema to provide a way to communicate that a given
test is expected to reboot during execution. The generic harness takes
that information and suppresses `already STARTED` warnings if the
restart is expected.

Fixes #85841