zephyrproject-rtos/zephyr #85854: ARC MWDT toolchain support is broken

tagunil, 2025-02-17:
Every attempt to build Zephyr with ARC MWDT toolchain ends up with failure while building one or another assembly file, like:
```
In file included from /home/tagunil/tests2/zephyr/arch/arc/core/fast_irq.S:18:
In file included from /home/tagunil/tests2/zephyr/include/zephyr/toolchain.h:44:
In file included from /home/tagunil/tests2/zephyr/include/zephyr/toolchain/mwdt.h:18:
/home/tagunil/tests2/zephyr/include/zephyr/toolchain/common.h:101:6: error: Architecture unsupported
  101 |     #error Architecture unsupported
```
I've bisected it down to 65471a6095fd46d537dd4b7ad0667b4f1018a1c2. I'm not sure why our toolchain is the only one affected, need to look into it further.

LoveKarlsson, 2025-02-17:
Probably because we separated C/C++/ASM COMPILE_LANGUAGE options, since IAR don't use the same flags for compiler and assembler like GCC, but I'm not sure

bjorniuppsala, 2025-02-17:
Autoconf.h is probably not being included due to the separation between compilation languages. 
Look at how it is done  for gcc (cmake/compiler/gcc/compiler_flags.cmake:235)

Adding something like this in cmake/compiler/arcmwdt/compiler_flags.cmake should fix the missing include:
```
# GCC compiler flags for imacros. The specific header must be appended by user.
set_property(TARGET asm PROPERTY imacros "-imacros")
```

fabiobaltieri, 2025-02-17:
cc @tejlmand

kartben, 2025-02-17:
@RobinKastberg

bjorniuppsala, 2025-02-17:
I took the liberty to suggest the fix in PR #85871

RobinKastberg, 2025-02-17:
I profusely apologize, I was under the mistaken impression that all toolchains apart from IAR inherited from `gcc/compiler_flags.cmake` and should have been more careful.

It would make me feel better if we could detect this condition somehow since all the files getting out of sync feels like it will become a more and more common problem.

tagunil, 2025-02-17:
> I profusely apologize, I was under the mistaken impression that all toolchains apart from IAR inherited from `gcc/compiler_flags.cmake` and should have been more careful.

Well, our current MWDT toolchain is Clang-based, but we have a lot of flags done differently for the sake of compatibility with our legacy toolchain. So maybe inheriting from GCC flags is a viable option, and maybe not. I hope I'll find some time to look into it someday.

RobinKastberg, 2025-02-17:
@tagunil do you have trustzone support? 
It will probably need to be fixed as well. See:
https://github.com/zephyrproject-rtos/zephyr/commit/e9a0d146cfa5aebe92d1c420b0ef5876fb67a731

tagunil, 2025-02-17:
> @tagunil do you have trustzone support? 
> It will probably need to be fixed as well. See:
> https://github.com/zephyrproject-rtos/zephyr/commit/e9a0d146cfa5aebe92d1c420b0ef5876fb67a731

No, because our toolchain supports only ARC and RISC-V architectures. No Arm, no TrustZone.

tejlmand, 2025-02-18:
> For these properties that are not optional to have, I think we could at least add a obviously broken default that screams in the build-log about not being set. 

@bjorniuppsala We can set a default property value in:
https://github.com/zephyrproject-rtos/zephyr/blob/2ce2794987049022e1f754f87d3c9b9bd098a3a8/cmake/compiler/compiler_flags_template.cmake#L129

something like:
```
set_property(TARGET asm PROPERTY imacros "imacros asm property undefined")
```

it won't avoid situations like #84800 from occurring as it still requires someone to test with the toolchain implementation that was broken, but it will make debugging of the issue easier for those hitting the problem.

PR #85871 (2025-02-19): cmake: mwdt assembler option for imacros
This is an untested example of something that might fix #85854 I do not have access to the toolchain, so I cant test it personally. 

It adds the -imacros property for assembler language to compensate for changes introduced in PR #84800.
closed 2025-02-19 by fabiobaltieri