zephyrproject-rtos/zephyr #85867: sysbuild: warning when building with mcuboot

toonst, 2025-02-17:
**Describe the bug**
Mcuboot sets Kconfig to use custom mbedTLS config file, which is not used in the end when building via sysbuild, causing a config warning.

**To Reproduce**
When building with mcuboot:
```
cd samples/hello-world
west build -p -b native_sim --sysbuild -- -DSB_CONFIG_BOOTLOADER_MCUBOOT=y
```

 I get the following warning:

> warning: MBEDTLS_CFG_FILE (defined at zephyrproject/bootloader/mcuboot/boot/zephyr/Kconfig:288, modules/mbedtls/Kconfig:52, modules/mbedtls/Kconfig:52) was assigned the value 'mcuboot-mbedtls-cfg.h' but got the value 'config-tls-generic.h'. See http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_MBEDTLS_CFG_FILE and/or look up MBEDTLS_CFG_FILE in the menuconfig/guiconfig interface.

**Expected behavior**
No warnings when building with sysbuild & mcuboot.

**Impact**
 annoyance

nordicjm, 2025-02-17:
No relation to sysbuild, is an warning emitted by MCUboot. And is caused because of using `native_sim`, this is not supported for MCUboot. Not a bug

closed 2025-02-17 by nordicjm

toonst, 2025-02-17:
@nordicjm This problem is not tied to `native_sim`, it can be reproduced with any board. I used `native_sim` erroneously for demonstrating purposes.
I agree the solution might be a change in mcuboot, but it might also be fixed by passing other options via sysbuild.

nordicjm, 2025-02-17:
Will pass message on but original point stands, that is a warning for MCUboot, it has no relation to sysbuild. MCUboot issues get reported on https://github.com/mcu-tools/mcuboot