zephyrproject-rtos/zephyr #85916: RFC: Introduce Motor API

eHammarstrom, 2025-02-18:
## Introduction

I don't find that there is any simple motor interface for my motor drivers to implement.
We are currently maintaining our own and have implemented a few brushed driver ICs. 

### Problem description
Increase portability of motor code and introduce the ability to share the driver code.

### Proposed change
Introduce `struct motor_api` with a few simple functions to implement. Underlying drivers may still use e.g. the already existing Motor Stepper API, or not.

## Detailed RFC and Proposed Change
Introduce the following API,

```
struct motor_config {
  uint8_t speed; /**< Motor speed percentage [0, 100] */
};

struct motor_api {
  int (*forward)(const struct device *dev); /**< Run motor forward at the configured speed */
  int (*reverse)(const struct device *dev); /**< Run motor in reverse at the configured speed */
  int (*stop)(const struct device *dev); /**< Stop the motor if currently running */
  int (*set_config)(const struct device *dev, struct motor_config *cfg); /**< Set the configuration of the motor */
};
```

To enable and disable the device, if applicable, implement the Device PM API.

To read the load of your motor, if applicable, implement the sensor API and support channel `SENSOR_CHAN_CURRENT`.

## Alternatives
If mixing API interfaces you implemented is best practice then the above suffice, otherwise we would add the following to the vtable,

```
struct motor_api {
  int (*enable)(const struct device *dev);
  int (*disable)(const struct device *dev);
  int (*read_load)(const struct device *dev, int32_t *load);
  // Same as the above follows
};
```

rruuaanng, 2025-02-19:
I’ve finally been waiting for you, and I’ve been considering putting forward this proposal. I believe it would be even better if we could add a QEI API.

faxe1008, 2025-02-21:
A few thoughts:

- forward and reverse can be condensed into one function with a direction parameter
- what does the motor_config look like?

phb98, 2025-02-24:
i dont think the use of word "forward" and "reverse" is good choice. For example, if the motor is used in the tank robot, to make the robot move forward, 1 motor have to rotate forward, while the other one have to rotate "reverse". 

We should anticipate the case of linear motor, where it moves instead of rotate, so using "CCW" and "CW" is not good enough.

I suggest we have a single function `move`, with input is a boolean direction: Negative and Positive.

jbehrensnx, 2025-02-24:
The stepper motor api uses a direction enum with positive and negative, so replicating this might be a good idea.

`enable/disable` (or just enable with boolean flag like in the stepper motor api) should definitely be part of the api.

That said, my experience is only with stepper motors and we will need feedback from experts concerning other motor types.

As discussed in the RFC for the introduction of the [stepper api](https://github.com/zephyrproject-rtos/zephyr/issues/68547), the currently preferred solution is to introduce smaller, more specific motor apis and using them as the basis to develop a broader motion subsystem/motor api later.

eHammarstrom, 2025-02-24:
@faxe1008 The struct is in the original post, I have only suggested one configuration option thus far.

@phb98 I would be fine with one `move` function taking the direction arguments `MOTOR_DIRECTION_POSITIVE`, `MOTOR_DIRECTION_NEGATIVE`.

@jbehrensnx The drivers I have used usually have a sleep pin, and if you use a separate regulator for the motor you may have an enable pin for that regulator. I don't mind having smaller and more specific motor APIs for more complex motor driving, but this API is  suitable for motor driver chipsets where you just drive 2 PWM signals to run the motor. Furthermore, it can be placed on top of more specific driver APIs.

jbehrensnx, 2025-02-24:
And stepper motor drivers work quite differently, thus you cant simply set your api on top of it as a general motor api. Developing a general motor api is a very complex undertaking requiring input from many different directions, because there are so many different motor types that behave quite differently. There is also the possibility of a motion subsystem to consider. Something that will need to happen. All of this came up in the linked RFC.
Thus developing smaller apis as a start. DC motor control is important for Zephyr and a corresponding api should be created, but we are still far away from a general motor api. For that, the vendors that have the expertise need to get involved.

jilaypandya, 2025-04-05:
DC Motors can/should have their specific api. Have been playing around with one bldc driver myself https://github.com/jilaypandya/zephyr/tree/introduce/bldc-api.

KozhinovAlexander, 2025-04-05:
A motor may get an error (over-heating/-current/-voltage protections may hit any time). Thus such errors may be covered by additional call back with type of error and an integer value of the measured problematic parameter.
Also additional callback from encoder for crossing of zero point may be useful. But I have now idea how to use such one if rotation speeds are high? Would the system be able to handle such fast movements with a callback?

rruuaanng, 2025-04-06:
We need to isolate it from the stepper motor subsystem. I believe this RFC can serve as the foundational API for DC motor control. If we plan to extend support to AC motor control, we can start by introducing a motor control module that provides core functionality such as Clarke and Park transformations, vector modulation, and basic signal processing. For driver implementations that aim to encapsulate speed control for AC motors, developers can leverage the functions provided in this module to implement advanced control strategies such as field-oriented control and torque control.

Users can build upon the speed control functionality provided by the driver to implement more advanced use cases. In other words, my expectation for this API is that it should offer a way to drive motor rotation, rather than serve as a complete end-to-end solution.

For other use cases, I believe we can refer back to the QEI I mentioned earlier. We could provide a function called read_encoder within the motor subsystem to read values from the motor encoder. This would be helpful for implementing the advanced control strategies mentioned above. I envision the function prototype as follows:

```

enum encoder_type {
    HALL_ENCODER = 0,
    QUAD_ENCODER
};

(Depends on the dsp subsystem, we need to do some fixed point conversion to floating point angles.)

q31_t (*read_encoder) (const struct device *dev, enum encoder_type type);
```

The directional information provided by the QE can be implicitly encoded in the sign of the return value, with positive values indicating forward rotation and negative values indicating reverse rotation. In contrast, Hall sensors would return only positive values.

jilaypandya, 2025-04-06:
> For other use cases, I believe we can refer back to the QEI I mentioned earlier. We could provide a function called read_encoder within the motor subsystem to read values from the motor encoder.

Wouldn't this https://docs.zephyrproject.org/latest/samples/sensor/qdec/README.html suffice?

KozhinovAlexander, 2025-04-06:
> > For other use cases, I believe we can refer back to the QEI I mentioned earlier. We could provide a function called read_encoder within the motor subsystem to read values from the motor encoder.
> 
> Wouldn't this https://docs.zephyrproject.org/latest/samples/sensor/qdec/README.html suffice?

Maybe leaving all the encoder stuff in the first approach would be easier?

rexut, 2025-04-06:
> We need to isolate it from the stepper motor subsystem.

@rruuaanng: What do you mean by “motor subsystem”? That doesn't exist, or am I wrong? So far there is a stepper motor **API** and a rudimentary control of **servo motors** via the PWM **API**. I am not aware of a *motor subsystem* or a *motion (control) subsystem*.

I am no longer able to follow the discussion here exactly. If a new subsystem is being thought up / invented in this RFC, then please divert the ideas into a new RFC and bundle the origins there, e.g. link back into this RFC with the heading “Motor API needs …”.

KozhinovAlexander, 2025-04-06:
> @rruuaanng: What do you mean by “motor subsystem”? That doesn't exist, or am I wrong? So far there is a stepper motor **API** and a rudimentary control of **servo motors** via the PWM **API**. I am [...]

Maybe someone takes the responsibility to write RFC or actively reject doing so? Let's vote.

rruuaanng, 2025-04-06:
> Wouldn't this https://docs.zephyrproject.org/latest/samples/sensor/qdec/README.html suffice?

Oh, it appears to be present, please disregard my mention of the QEI.

> What do you mean by “motor subsystem”?

The motor subsystem is currently a proposed concept in the RFC. I believe the RFC only needs to implement basic rotational functionality for the motor; at the very least, no one is expected to attempt using a DC motor as a substitute for a servo system. I think we should evaluate whether this RFC is worthwhile, specifically, as a general-purpose API for motor speed control, without considering motion control or any other functionalities at this stage.

For the RFC, +1 (This would be helpful for implementing small fans and cooling products.)

jbehrensnx, 2025-04-07:
I am definitely against using this RFC as a basis for a general purpose motor api.

On the DC motor front I am not knowledgeable enough to evaluate if this is a good dc-motor api.

thenguyenyf, 2025-06-17:
Hello @eHammarstrom . I just find out this RFC. We are working on supporting the [MCK-RA8T1 kit](https://www.renesas.com/en/design-resources/boards-kits/mck-ra8t1?srsltid=AfmBOorwnwpwkOjZkl5yfOaylCg6FoKuHGiCTzLG8O7F5YUaMPA-v3-c#overview) which comes with a motor control solution - 3 phase BLDC motor.

We start working on first step to introduce our 3 phase PWM interface on Zephyr that may have the similar functional like [r_gpt_three_phase](https://renesas.github.io/fsp/group___g_p_t___t_h_r_e_e___p_h_a_s_e.html?pos=791,3,901,5,1204,3,1297,3,1356,3,3534,3,3612,3,3647,3,3725,3,3760,3,3838,3,3946,3,5726,3,5818,3,5899,3,6019,3,6107,3,6286,3,7189,3,7265,3,12133,3,13238,3). We will then add support for the inverter/motor controller that comes with the MCK-RA8T1, which is expected to reference support from the vendor's SDK [Motor Interface](https://renesas.github.io/fsp/group___m_o_t_o_r___a_p_i.html?pos=36,5,67,5,92,5,3136,6,3154,5,3364,6,3381,5,3594,5,4750,5,4867,5,6952,5,7662,5,7736,5,8043,5)

It would be ideal if the motor API can adapt with the 3 phase BLDC (with the PWM 3 phase controller backend) beside the main target that mentioned in this RFC. So, if you interested in cooperation, please don't hesitate to let me know to share our situation.

onyx22574, 2025-08-15:
I'm a new to Zephyr, is there any motor-control stack (whatever development stage) available for Zephyr?

teburd, 2026-10-22:
Robotics WG: We'd like to see a motor control API as well.

Some thoughts...

Depending on the context of the motor different kinds of control APIs/command and feedback may be desirable.

ROS: all joints are controlled with commands, torque, position, speed, etc. The joint API is universal but the joint itself and the hardware may change what is available. State information is available on the joint depending on the kind of joint and hardware controlling it.

For example having an encoder for positional information and/or current monitor for getting effort/torque being applied.

kpochwala, 2026-10-22:
I would like to maybe somewhat promote the proposed motor API to be a generic "Joint API". I think a higher-level one similar in concepts to [ros2_control Joint Kinematics](https://control.ros.org/iron/doc/ros2_control/hardware_interface/doc/joints_userdoc.html) would be nice, and pretty intuitive for folks coming from the ROS2 world. Let's take the [Zenbedded ](https://github.com/ros-controls/zenbedded)(official ROS2 controls integration with Zephyr) inverted pendulum demo:

[image]

and corresponding snippet from URDF (Universal Robot Description Format)

```xml
        <joint name="motor_joint">
            <state_interface name="position"/>
            <state_interface name="velocity"/>
            <command_interface name="acceleration"/>
        </joint>
        <joint name="pendulum_joint">
            <state_interface name="position"/>
            <state_interface name="velocity"/>
        </joint>
```

here the `motor_joint` has 2 state interfaces (data that can be read from the joint) - `position` and `velocity`, one command interface (data that can be "written into" the joint) - `acceleration`. There is also free-wheeling joint, `pendulum_joint` from which the data can be read, but it has no ability to move on its own. Something similar could be available in the devicetree:

motor_joint (blue):
- the motor driver that allows for acceleration control (be it stepper, BLDC or plain DC)
- the driver(s) that allow for reading the position and velocity data from the motor axis - for stepper that could be the same driver as used for the acceleration control, for DC it would be probably some encoder

pendulum_joint (red):
- the driver(s) that allow for reading the position and velocity data from the pendulum axis

would map to something along the lines:

```c
/ {
    joints {
        compatible = "zephyr,joints";

        motor_joint: motor_joint {
            compatible = "zephyr,joint";
            joint-type = <JOINT_TYPE_REVOLUTE>;

            command-devices = <&tmc5160_ctrl>;
            command-interface-names = "acceleration";

            state-devices = <&tmc5160_ctrl &tmc5160_ctrl>;
            state-sensor-channels = <SENSOR_CHAN_ROTATION SENSOR_CHAN_RPM>; // the stepper_ctrl would need to support sensor API? 
            state-interface-names = "position", "velocity";
        };

        pendulum_joint: pendulum_joint {
            compatible = "zephyr,joint";
            joint-type = <JOINT_TYPE_REVOLUTE>;

            state-devices = <&pendulum_encoder &pendulum_encoder>;
            state-sensor-channels = <SENSOR_CHAN_ROTATION SENSOR_CHAN_RPM>;
            state-interface-names = "position", "velocity";
            
        };
    };
};
```

Of course the limits and stuff could (should) be also specified there. AFAIK URDF lately got some built-in tags for specifying just that. Maybe it would be even possible to somehow translate the URDF file joint stuff into the generated `.dtsi` with the interface names and limits, and that file could be included in the overlay, joints referenced and bound with the devices.