zephyrproject-rtos/zephyr #85922: samples: samples/basic/minimal: mimxrt1180_evk/mimxrt1189/cm33: error: Build failure - error: fsl_rgpio.h: No such file or directory

hakehuang, 2025-02-18:
**Describe the bug**
below case are error

test is error on v4.1.0-rc1-49-g3e274ef91ba5 on mimxrt1180_evk/mimxrt1189/cm33
testcase path is samples/basic/minimal

see logs for details

**To Reproduce**
1. 
```
scripts/twister -p mimxrt1180_evk/mimxrt1189/cm33  -T samples/basic/minimal --sub-test samples/basic/minimal
```
or
```
# west twister -p mimxrt1180_evk/mimxrt1189/cm33 -T samples/basic/minimal --sub-test samples/basic/minimal
```
2. See error

```
Build failure - error: fsl_rgpio.h: No such file or directory
```

**Expected behavior**
test pass

**Impact**

**Logs and console output**

```

    
```

```

```

**Environment (please complete the following information):**
 - OS: (e.g. Linux ): posix
 - Toolchain (e.g Zephyr SDK): zephyr
 - Commit SHA or Version used: v4.1.0-rc1-49-g3e274ef91ba5

kartben, 2025-02-18:
can't reproduce, not a Zephyr upstream commit(?)

hakehuang, 2025-02-19:
> can't reproduce, not a Zephyr upstream commit(?)

@kartben , sorry, this is found by my pr https://github.com/zephyrproject-rtos/zephyr/pull/85446, which I found the one issue and root cause is below. it is not related to Ci or release scope.
```
CONFIG_GPIO=n but there is still dependency on fgpio.h, as the the board.h take the fgpio.h as default. but for zephyr this file is not needed. and in evkmimxrt1180_flexspi_nor_config.h, we can see below

#if defined(BOARD_FLASH_SIZE)
#define FLASH_SIZE BOARD_FLASH_SIZE
#else
#error "Please define macro BOARD_FLASH_SIZE"
#endif
this is not used in zephyr at all

test is error on v4.1.0-rc1-49-g3e274ef91ba5 on mimxrt1180_evk/mimxrt1189/cm33
testcase path is samples/basic/minimal

see logs for details
```

closed 2025-03-04 by hakehuang