zephyrproject-rtos/zephyr #85923: samples: samples/basic/minimal: mcxw72_evk/mcxw727c/cpu0: error: Build failure - error: fsl_port.h: No such file or directory

hakehuang, 2025-02-18:
**Describe the bug**
below case are error

test is error on v4.1.0-rc1-49-g3e274ef91ba5 on mcxw72_evk/mcxw727c/cpu0
testcase path is samples/basic/minimal

see logs for details

**To Reproduce**
1. 
```
scripts/twister -p mcxw72_evk/mcxw727c/cpu0  -T samples/basic/minimal --sub-test samples/basic/minimal
```
or
```
# west twister -p mcxw72_evk/mcxw727c/cpu0 -T samples/basic/minimal --sub-test samples/basic/minimal
```
2. See error

```
Build failure - error: fsl_port.h: No such file or directory
```

**Expected behavior**
test pass

**Impact**

**Logs and console output**

```

    
```

```
m=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=/home/shared/disk/zephyr_project/zephyr_test/zephyr/samples/basic/minimal=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/shared/disk/zephyr_project/zephyr_test/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/shared/disk/zephyr_project/zephyr_test=WEST_TOPDIR -ffunction-sections -fdata-sections -mcmse -specs=picolibc.specs -std=c99 -fno-lto -MD -MT zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj -MF zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj.d -o zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj -c /home/shared/disk/zephyr_project/zephyr_test/zephyr/arch/arm/core/offsets/offsets.c
In file included from /home/shared/disk/zephyr_project/zephyr_test/zephyr/modules/cmsis/./cmsis_core_m.h:24,
                 from /home/shared/disk/zephyr_project/zephyr_test/zephyr/modules/cmsis/./cmsis_core.h:10,
                 from /home/shared/disk/zephyr_project/zephyr_test/zephyr/include/zephyr/arch/arm/asm_inline_gcc.h:24,
                 from /home/shared/disk/zephyr_project/zephyr_test/zephyr/include/zephyr/arch/arm/asm_inline.h:18,
                 from /home/shared/disk/zephyr_project/zephyr_test/zephyr/include/zephyr/arch/arm/arch.h:30,
                 from /home/shared/disk/zephyr_project/zephyr_test/zephyr/include/zephyr/arch/cpu.h:19,
                 from /home/shared/disk/zephyr_project/zephyr_test/zephyr/include/zephyr/kernel_includes.h:36,
                 from /home/shared/disk/zephyr_project/zephyr_test/zephyr/include/zephyr/kernel.h:17,
                 from /home/shared/disk/zephyr_project/zephyr_test/zephyr/arch/arm/core/offsets/offsets_aarch32.c:28,
                 from /home/shared/disk/zephyr_project/zephyr_test/zephyr/arch/arm/core/offsets/offsets.c:9:
/home/shared/disk/zephyr_project/zephyr_test/zephyr/soc/nxp/mcx/mcxw/./soc.h:9:10: fatal error: fsl_port.h: No such file or directory
    9 | #include <fsl_port.h>
      |          ^~~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.

```

**Environment (please complete the following information):**
 - OS: (e.g. Linux ): posix
 - Toolchain (e.g Zephyr SDK): zephyr
 - Commit SHA or Version used: v4.1.0-rc1-49-g3e274ef91ba5

kartben, 2025-02-18:
can't reproduce (neither on macOS or Linux, with SDK 0.17.0, both on main@50c21f159143e18c8a05cbb946a5e7610033355f as well as official rc1)
`Commit SHA or Version used: v4.1.0-rc1-49-g3e274ef91ba5` <-- this isn't an upstream Zephyr version, please check your local environment and re-open as needed.

closed 2025-02-18 by kartben

hakehuang, 2025-02-19:
> can't reproduce (neither on macOS or Linux, with SDK 0.17.0, both on main@50c21f159143e18c8a05cbb946a5e7610033355f as well as official rc1)
> `Commit SHA or Version used: v4.1.0-rc1-49-g3e274ef91ba5` <-- this isn't an upstream Zephyr version, please check your local environment and re-open as needed.

@kartben , this is a enhancement issue, so it is based on my PR https://github.com/zephyrproject-rtos/zephyr/pull/85446