zephyrproject-rtos/zephyr #85936: samples: samples/basic/minimal: ms32z2xxdc2@D/s32z270/rtu0: error: Build failure - undefined reference to `sys_clock_cycle_get_32'

hakehuang, 2025-02-18:
**Describe the bug**

CONFIG_TIMER=n but there is still dependency on timer call.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 - ms32z2xxdc2@D/s32z270/rtu0
 - What have you tried to diagnose or workaround this issue?
 - N/A
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?

**To Reproduce**

```
scripts/twister -p ms32z2xxdc2@D/s32z270/rtu0  -T samples/basic/minimal
```

**Expected behavior**

Build PASS

**Impact**

board configs

**Logs and console output**

```
zephyr_project/zephyr_test/zephyr/twister-out/s32z2xxdc2@D_s32z270_rtu0/zephyr/samples/basic/minimal/sample.minimal.mt-no-preempt-no-timers/zephyr && /snap/cmake/1441/bin/cmake -E true\n/home/ubuntu/zephyr-sdk/zephyr-sdk-0.17.0-rc1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: modules/hal_nxp/hal_nxp/lib..__modules__hal__nxp.a(OsIf_Timer_System.c.obj): in function `arch_k_cycle_get_32':\n/home/shared/disk/zephyr_project/zephyr_test/zephyr/include/zephyr/arch/arm/misc.h:26: undefined reference to `sys_clock_cycle_get_32'\n/home/ubuntu/zephyr-sdk/zephyr-sdk-0.17.0-rc1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: /home/shared/disk/zephyr_project/zephyr_test/zephyr/include/zephyr/arch/arm/misc.h:26: undefined reference to `sys_clock_cycle_get_32'\ncollect2: error: ld returned 1 exit status\nninja: build stopped: subcommand failed.\n"
```

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: v4.1.0-rc1-49-g3e274ef91ba5

**Additional context**

mmahadevan108, 2025-02-18:
@manuargue @Dat-NguyenDuy , can you help take a look at this build failure.

kartben, 2025-02-18:
The board mentioned in this bug report doesn't exist in Zephyr.

Please use the right internal bug tracker or whatever and try not to overwhelm the release management team and project at large with unnecessarry bug reports as we're trying to stabilize 4.1.

kartben, 2025-02-18:
I apologize, it looks like there is a typo in the board name. It apparently should be `s32z2xxdc2@D/s32z270/rtu0`
However, like other recent bug reports, this is referencing a commit SHA that is NOT a Zephyr one, and that issue cannot be reproduced locally.

manuargue, 2025-02-18:
> However, like other recent bug reports, this is referencing a commit SHA that is NOT a Zephyr one, and that issue cannot be reproduced locally.

I agree with this comment. The steps to reproduce it don't build anything actually, with current main HEAD:
```
scripts/twister -p s32z2xxdc2@D/s32z270/rtu0 -T samples/basic/minimal
INFO    - Zephyr version: v4.1.0-rc1-57-g50c21f159143
...
INFO    - 12 test scenarios (12 configurations) selected, 12 configurations filtered (12 by static filter, 0 at runtime).
INFO    - 0 of 0 executed test configurations passed (0.00%), 0 built (not run), 0 failed, 0 errored, with no warnings in 3.56 seconds.
INFO    - 0 of 0 executed test cases passed (0.00%) on 0 out of total 948 platforms (0.00%).
INFO    - 0 test configurations executed on platforms, 0 test configurations were only built.
```
It can be reproduced with `west build -b s32z2xxdc2@D/s32z270/rtu0 samples/basic/minimal/ -T sample.minimal.no-mt-no-timers.arm`. But because this SoC is not in the `platform_allow` list and twister is not picking this up, I don't think this is affecting the CI or release.

hakehuang, 2025-02-19:
> It can be reproduced with `west build -b s32z2xxdc2@D/s32z270/rtu0 samples/basic/minimal/ -T sample.minimal.no-mt-no-timers.arm`. But because this SoC is not in the `platform_allow` list and twister is not picking this up, I don't think this is affecting the CI or release.

This is found when I trying to make minial config available to all NXP platforms, and I find there are actually problem in the hal lavel, so I report. @manuargue , please looks at my analysis below
```
CONFIG_TIMER=n but there is still dependency on timer call. It is Because the hal uncondtional use the timer

e.g. in
../modules/hal/nxp/s32/drivers/s32ze/BaseNXP/src/OsIf_Timer_System.c

function OsIf_Timer_System_GetCounter

#elif defined(USING_OS_ZEPHYR)
        /* ZephyrOS */
        (void)CoreId;
        Counter = Trusted_k_cycle_get_32();
and

#if (STD_ON == OSIF_ENABLE_USER_MODE_SUPPORT)
    #define Trusted_OsIf_Timer_System_Internal_Init(Freq)               OsIf_Trusted_Call1param(OsIf_Timer_System_Internal_Init, (Freq))
    #define Trusted_OsIf_Timer_System_Internal_GetCounter()             OsIf_Trusted_Call_Return(OsIf_Timer_System_Internal_GetCounter)
    #define Trusted_OsIf_Timer_System_Internal_GetElapsed(CurrentRef)   OsIf_Trusted_Call_Return1param(OsIf_Timer_System_Internal_GetElapsed, (CurrentRef))
    #define Trusted_k_cycle_get_32()                                    OsIf_Trusted_Call_Return(k_cycle_get_32)
#else
    #define Trusted_OsIf_Timer_System_Internal_Init(Freq)               OsIf_Timer_System_Internal_Init(Freq)
    #define Trusted_OsIf_Timer_System_Internal_GetCounter()             OsIf_Timer_System_Internal_GetCounter()
    #define Trusted_OsIf_Timer_System_Internal_GetElapsed(CurrentRef)   OsIf_Timer_System_Internal_GetElapsed(CurrentRef)
    #define Trusted_k_cycle_get_32()                                    k_cycle_get_32()
#endif

```

manuargue, 2025-02-19:
> This is found when I trying to make minial config available to all NXP platforms, and I find there are actually problem in the hal lavel, so I report. [@manuargue](https://github.com/manuargue) , please looks at my analysis below

After discussing with @hakehuang now I understand the context of this issue is https://github.com/zephyrproject-rtos/zephyr/pull/85446. Because the `CONFIG_TIMER=n` use-case is currently not supported for NXP S32 platforms due to HAL design constraints, the minimal sample selecting this configuration shouldn't be enabled.

I'm reopening this issue and keep it as an enhacement rather than a bug, beacuse is not reproducible with twister on main branch.

manuargue, 2025-04-28:
The `CONFIG_TIMER=n` use-case won't be supported for NXP S32 platforms due to HAL design constraints, the minimal sample selecting this configuration shouldn't be enabled on main branch. Closing this issue.

closed 2025-04-28 by manuargue