zephyrproject-rtos/zephyr #85942: Sample Bluetooth:central didn't work with ESP32S3

trungkiena6, 2025-02-18:
**Describe the bug**
I am running a simple central function test with the ESP32S3. However, it encounters an issue on creating the connection. It always return -13 (EACCES)

```
Create conn to 55:18:2A:7F:8A:52 (random) failed (-13)
Scanning successfully started
Device found: 54:AF:97:A3:92:72 (public) (RSSI -76)
Device found: 40:CA:63:CB:16:7F (public) (RSSI -90)
Device found: 57:4C:62:80:31:5F (public) (RSSI -53)
Device found: 4F:E3:E5:27:99:DA (random) (RSSI -80)
Device found: 70:D6:AF:C0:E2:8E (random) (RSSI -79)
Device found: 52:34:FF:9F:00:FE (random) (RSSI -74)
Device found: F1:51:1E:3A:60:19 (random) (RSSI -81)
Device found: 98:CD:AC:4C:C1:6A (public) (RSSI -85)
Device found: 61:F4:77:20:58:28 (random) (RSSI -90)
Device found: 6F:79:2A:9F:CD:74 (random) (RSSI -79)
Device found: 55:18:2A:7F:8A:52 (random) (RSSI -27)
Create conn to 55:18:2A:7F:8A:52 (random) failed (-13)
Scanning successfully started
Device found: 57:4C:62:80:31:5F (public) (RSSI -53)
Device found: 54:AF:97:A3:92:72 (public) (RSSI -68)
Device found: 98:CD:AC:4C:C1:6A (public) (RSSI -86)
Device found: 34:84:12:8D:7F:EA (random) (RSSI -68)
Device found: 52:34:FF:9F:00:FE (random) (RSSI -74)
Device found: 40:CA:63:CB:16:7F (public) (RSSI -93)
Device found: 71:A1:E1:AD:F1:BB (random) (RSSI -88)
Device found: 4F:E3:E5:27:99:DA (random) (RSSI -71)
Device found: 6F:79:2A:9F:CD:74 (random) (RSSI -79)
Device found: 55:18:2A:7F:8A:52 (random) (RSSI -27)
Create conn to 55:18:2A:7F:8A:52 (random) failed (-13)
Scanning successfully started
Device found: 57:4C:62:80:31:5F (public) (RSSI -53)
Device found: 52:34:FF:9F:00:FE (random) (RSSI -89)
Device found: 98:CD:AC:4C:C1:6A (public) (RSSI -86)
Device found: 40:CA:63:CB:16:7F (public) (RSSI -91)
Device found: 80:EC:CC:CD:FD:C1 (public) (RSSI -72)
```

**To Reproduce**
Simply build the settings example and run it:
```
west build -b esp32s3_devkitm -p -d zephyr/samples/bluetooth/central
```

**Expected behavior**
```
Scanning successfully started
Device found: 57:4C:62:80:31:5F (public) (RSSI -53)
Device found: 52:34:FF:9F:00:FE (random) (RSSI -89)
Device found: 98:CD:AC:4C:C1:6A (public) (RSSI -86)
Device found: 40:CA:63:CB:16:7F (public) (RSSI -91)
Device found: 80:EC:CC:CD:FD:C1 (public) (RSSI -72)
Connected: CF:D1:1C:97:CB:DC (random)
```

**Impact**
With version lower v4.0.0, it works normally. When upgrading to version v4.1.0-rc1, it encounters this issue, and I think it significantly affects programs that need to use the central mode.

**Environment:**

- OS: MacOS, Linux or Windows
- Toolchain: Zephyr SDK v0.16.8
- Zephyr v4.1.0-rc1

milrt, 2025-04-30:
Hello @trungkiena6 , have you found a solution or workaround for this issue? 
I'm experiencing the same problem, and it makes central mode unusable.

devemouse, 2025-05-23:
Same here. Any updates?

closed 2025-06-26 by dkalowsk