zephyrproject-rtos/zephyr #85952: samples: arch: mpu: mpu_test:  regression on some STM32 boards with Cortex M0+ and M33 processor

djiatsaf-st, 2025-02-18:
**Describe the bug**

The changes made in this PR https://github.com/zephyrproject-rtos/zephyr/pull/85033 introduce a regression in the 
Memory Protection Unit archi sample for some STM32 boards with Cortex M0+ and M33 (b_u585i_iot02a, nucleo_c071rb, nucleo_g071rb, nucleo_l073rz, nucleo_wba55cg).

In sample.yaml, the keyboard harness property is changed to shell.

**To Reproduce**

1. run ./scripts/twister --device-testing --device-serial /dev/ttyACM0 -p nucleo_g071rb -T  samples/arch/mpu/mpu_test/
 
   or
   - west build -p -b nucleo_g071rb samples/arch/mpu/mpu_test/ -T sample.mpu.mpu_test
   -  west flash
   -  send "mpu mtest 1"

2. See logs in path/to/twister_harness.log / handler.log or uart console

```
-------------------------------- live log call --------------------------------- 

INFO: send command: mpu mtest 1 

DEBUG: #: uart:~$ mpu mtest 1 

[00:00:00.628,000] <err> os: ***** HARD FAULT ***** 

[00:00:00.628,000] <err> os: r0/a1:  0x00000001  r1/a2:  0x0000000f  r2/a3:  0x00000000 

[00:00:00.628,000] <err> os: r3/a4:  0x00000001 r12/ip:  0x00000020 r14/lr:  0x0800018f 

[00:00:00.628,000] <err> os:  xpsr:  0x61000000 

[00:00:00.628,000] <err> os: Faulting instruction address (r15/pc): 0x080006ac 

[00:00:00.628,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0 

[00:00:00.628,000] <err> os: Current thread: 0x20000660 (shell_uart) 

[00:00:00.685,000] <err> os: Halting system 

ERROR: Read from device timeout occurred 

FAILED
```
**Expected behavior**

Valid console output : 

`The value is: 0x...`

**Logs and console output**

```
*** Booting Zephyr OS build v4.1.0-rc1-45-g2ce279498704 ***
uart:~$ mpu mtest 1

[00:01:42.190,000] <err> os: ***** HARD FAULT *****
[00:01:42.190,000] <err> os: r0/a1:  0x00000001  r1/a2:  0x0000000f  r2/a3:  0x00000000
[00:01:42.190,000] <err> os: r3/a4:  0x00000001 r12/ip:  0x00000020 r14/lr:  0x0800018f
[00:01:42.190,000] <err> os:  xpsr:  0x61000000
[00:01:42.190,000] <err> os: Faulting instruction address (r15/pc): 0x080006ac
[00:01:42.191,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:01:42.191,000] <err> os: Current thread: 0x20000660 (shell_uart)
[00:01:42.247,000] <err> os: Halting system

```

**Environment (please complete the following information):**

 - OS:  Ubuntu 22.04  LTS
 - Toolchain Zephyr SDK 0.17.0

kartben, 2025-02-18:
impacting quite a few stm32 targets -- will triage as medium, feel free to re-label as you see fit @erwango

nashif, 2025-02-18:
this is not a regression really, this sample was never run on any hardware before.