zephyrproject-rtos/zephyr #85957: RT1061 Does Not Send Gratuitous ARP After Power-Up and Network Interface Activation

ofirshe, 2025-02-18:
**Describe the bug**  
To send Gratuitous ARP when the interface goes operational up, the ARP module registers for `NET_EVENT_IF_UP` event,  
as seen [here](https://github.com/zephyrproject-rtos/zephyr/blob/c387fcae17eb8cef274e782acd817bdaa5fe2613/subsys/net/l2/ethernet/arp.c#L1006).  
However, it can be observed that after power-up, the first packet is only sent after `CONFIG_NET_ARP_GRATUITOUS_INTERVAL` seconds  
(by default, it's 60, but in this case, I set it to 30). This occurs because the ARP packet is triggered by the  
`arp_gratuitous_work_handler` delayed worker. 

**To Reproduce**  
I ran the `zperf` sample with Gratuitous ARP configurations enabled:  
```
CONFIG_NET_MGMT_EVENT=y  
CONFIG_NET_MGMT_EVENT_INFO=y  
CONFIG_NET_ARP_GRATUITOUS=y  
CONFIG_NET_ARP_GRATUITOUS_TRANSMISSION=y  
CONFIG_NET_ARP_GRATUITOUS_INTERVAL=30  
```
Additionally, I modified `CONFIG_NET_CONFIG_MY_IPV4_ADDR="172.18.133.10"` to observe outgoing packets in my lab.  

**Expected behavior**  
A Gratuitous ARP packet should be sent immediately after the network interface is up.  

**Impact**  
If a client tries to connect to the board without an updated MAC address for the RT1061  
(by default, it uses a random MAC address on each boot), the client will send multiple ARP requests  
to resolve the IP using the previous MAC in its cache. Only after multiple retries will the client  
broadcast an ARP request, which the RT1061 then responds to with the correct MAC address.  
This unnecessary delay and network traffic could be avoided if the RT1061 sent a Gratuitous ARP  
packet immediately after reset.  

**Environment**  
- **OS**: Windows  
- **Toolchain**: Zephyr SDK  
- **Commit SHA or Version used**:  

**Additional context**  
This does not appear to be a widespread board-specific issue. @rublos tested this behavior on three different platforms  
(native_sim, qemu_x86, nucleo_h723zg) and confirmed that Gratuitous ARP is sent on boot, as discussed [here](https://github.com/zephyrproject-rtos/zephyr/discussions/85532?notification_referrer_id=NT_kwDOAjyuWbQxNDcyNDc0ODgwNzozNzUzMTIyNQ#discussioncomment-12187368).

jacob-wienecke-nxp, 2025-02-20:
@ofirshe 
I have not been able to find SHA 66ff9380f82b.

Zephyr Version: tag v4.1.0-rc1.
Board: rt1060_evkb
Sample: Zperf (with the listed modifications to the prj.conf  file)

Similar to the report from @rlubos, I am seeing the ARP on boot.

[Capture.zip](https://github.com/user-attachments/files/18895825/Capture.zip)

ofirshe, 2025-02-21:
Thanks for your response, @jacob-wienecke-nxp. Since I'm out of office and don't have a board at home, I'll check `v4.1.0-rc1` again on Sunday.

ofirshe, 2025-02-23:
Hi @jacob-wienecke-nxp,  

I've tested the latest [main](https://github.com/zephyrproject-rtos/zephyr/commit/9b2fdace1d4c9686d43fbd8fad87b9620f576671), and I'm still encountering the issue.
I'm running on the `RT1060-EVK`. Did you test it on this board as well?
Additionally, I tested without any debug connection—I simply flashed the image and monitored the UART output.
[Screen Recording 2025-02-23 143919.zip](https://github.com/user-attachments/files/18930472/Screen.Recording.2025-02-23.143919.zip)

jacob-wienecke-nxp, 2025-02-24:
Tested today on RT1060-EVK using SHA: 9b2fdace1d4c9686d43fbd8fad87b9620f576671

My result is the same as before. The behavior seems to be the same whether I have a debugger connected or not.

Since the [Zperf](https://docs.zephyrproject.org/latest/connectivity/networking/api/zperf.html) sample is flashed, are you able to send packets using your PC and the EVK?

I am testing using the host machine as server:
CMD: iperf -s

and the Zephyr Device as client:
Zephyr Console: zperf tcp upload 10.81.17.65 5001 10 1K 1M

That seems to work fine as well.

ofirshe, 2025-02-24:
Hi @jacob-wienecke-nxp,

I tested today on the latest [main](https://github.com/zephyrproject-rtos/zephyr/commit/ee8266797c176b67867c1012d441141885bc9e4c), and the issue persists. I see the first gratuitous ARP packet going out of my board only after 30 seconds. You can also run my binary below. Here are the changes I made:

- Set the console port to LPUART3 (J22 pins 1 and 2)
- Modified the `zperf` config to fit my environment (set the IPv4 accordingly).

This is my full zperf configuration file:
```
CONFIG_NETWORKING=y
CONFIG_NET_ZPERF=y
CONFIG_NET_LOG=y
CONFIG_NET_IPV6=y
CONFIG_NET_IPV4=y
CONFIG_NET_DHCPV4=n
CONFIG_NET_UDP=y
CONFIG_NET_TCP=y
CONFIG_NET_TCP_IPV6_ND_REACHABILITY_HINT=y
CONFIG_NET_STATISTICS=y

CONFIG_NET_PKT_RX_COUNT=40
CONFIG_NET_PKT_TX_COUNT=40
CONFIG_NET_BUF_RX_COUNT=160
CONFIG_NET_BUF_TX_COUNT=160
CONFIG_NET_BUF_DATA_SIZE=256
CONFIG_NET_IF_UNICAST_IPV6_ADDR_COUNT=4
CONFIG_NET_IF_MCAST_IPV6_ADDR_COUNT=5
CONFIG_NET_IF_UNICAST_IPV4_ADDR_COUNT=1
CONFIG_NET_MAX_CONTEXTS=5
CONFIG_NET_TC_TX_COUNT=1
CONFIG_NET_SOCKETS=y
CONFIG_NET_SOCKETS_SERVICE_THREAD_PRIO=-1
CONFIG_ZVFS_POLL_MAX=9
CONFIG_ZVFS_OPEN_MAX=12
CONFIG_POSIX_API=y

CONFIG_INIT_STACKS=y
CONFIG_TEST_RANDOM_GENERATOR=y

CONFIG_NET_L2_ETHERNET=y
CONFIG_NET_SHELL=y

CONFIG_NET_CONFIG_SETTINGS=y
CONFIG_NET_CONFIG_MY_IPV6_ADDR="2001:db8::1"
CONFIG_NET_CONFIG_PEER_IPV6_ADDR="2001:db8::2"
CONFIG_NET_CONFIG_MY_IPV4_ADDR="172.18.133.10"
CONFIG_NET_CONFIG_PEER_IPV4_ADDR="172.18.133.20"
CONFIG_NET_CONFIG_MY_IPV4_NETMASK="255.255.240.0"
CONFIG_NET_CONFIG_MY_IPV4_GW="172.18.143.254"

CONFIG_LOG=y
CONFIG_SHELL_CMDS_RESIZE=n

CONFIG_CACHE_MANAGEMENT=y
CONFIG_SPEED_OPTIMIZATIONS=y

CONFIG_NET_MGMT_EVENT=y  
CONFIG_NET_MGMT_EVENT_INFO=y  
CONFIG_NET_ARP_GRATUITOUS=y  
CONFIG_NET_ARP_GRATUITOUS_TRANSMISSION=y  
CONFIG_NET_ARP_GRATUITOUS_INTERVAL=30  
```

I also monitored the incoming packets from my board to the switch and saw the counter increasing only after 30 seconds (when the first gratuitous ARP packet is sent).

[zephyr.zip](https://github.com/user-attachments/files/18947552/zephyr.zip)

jacob-wienecke-nxp, 2025-02-24:
@ofirshe I am testing the binary directly, and I modified my PC to use the PEER_IPV4_ADDR.

I still am getting the ARP on boot.

Is it possible that your switch port is not coming up in time? Maybe there is a possibility that the device sends out the initial ARP but it is missed. It could be worth trying a direct connection between your host PC and the Zephyr Device.

Perhaps there are other possible causes within the network. As far as your configs, changes, binary, I am not seeing any issues.

jacob-wienecke-nxp, 2025-02-28:
Hi @ofirshe,

Some recommendations for confirming that the first gratuitous ARP is being sent.

1) Using an Oscilloscope to monitor the PHY.
The waveform of the retransmitted gratuitous ARP packet should look nearly identical to the initial one. 

2) Using a debugger to monitor the PHY registers

3) Using a debugger to monitor the point in the code where gratuitous ARP is being sent.

closed 2025-03-17 by ofirshe